pascallanger/DIY-Multiprotocol-TX-Module #778: In DSM FwdPrg_05_Color.lua Unable to access Flight Mode 1

charbaugh1, 2022-12-08:
TX16S MK2 tx
EdgeTX 2.8.0
AR630 rx

During first time setup of AS3X, I assigned the channel for the flight modes to a 3-position switch that was preconfigured in model setup.  In the AS3X setup, when changing the state of the switch to adjust parameters, it only changes from FM2 to FM3.  When moving the switch to the position for FM1, the display shows FM2.  I have tried reassigning to other available switches to confirm it is not a hardware issue.

frankiearzu, 2022-12-18:
Tried to duplicate the issue but could not reproduce.
Only if i set my servo output channel to less than -100,  i only get FM1,FM2.   if i have (-100,0,+100) for the switch, all 3 flight mode works.  Check your output settings.

charbaugh1, 2022-12-18:
I’ve definitely checked and double checked the output values, and in the output monitor it confirms -100, 0, +100.  Strange is that I get FM2 and FM3, but I cannot get FM1.  Otherwise, it works perfectly.

 

Clark Harbaugh

frankiearzu, 2022-12-18:
Clark,
We can check what the RX are sending to the tool to see if it is a tool problem of something else.
the /LOGS/dsm_log.txt contains the last conversation between the tool and the RX.

For example, if you go to forward programming, and go to "Gyro Settings->F Mode setup" you can use the switch to switch the 3 positions. then Exit the script (pressing the RETR button). 
The last part of the log will have the actions when you flipped the switch.

NOTE the  "Val=0", "Val=1","Var=2" etc. those are the raw values coming from the RX in the Flight mode line. The display adds +1, so Var=0 is "Flight Mode 1", Var=1 is "Flight Mode 2", etc..  My TX switch and output gets "-100,0,+100).

This is how my log looks (end of the file) after going to that page and moving the switch the 3 different positions: 
```
19.170 WAIT_CMD: RESPONSE MenuValue: UPDATED: L[#0 T=V_nc VId=0x1000 Text="Flight Mode"[0x8001] Val=0 [0->10,0] MId=0x1021 ]
19.310 MENU_VALUES: SEND DSM_getNextMenuValue(MenuId=0x1021, LastValueId=0x1000) Extra: Text="Flight Mode"
21.190 MENU_VALUES: RESPONSE MenuValue: UPDATED: L[#0 T=V_nc VId=0x1000 Text="Flight Mode"[0x8001] Val=1 [0->10,0] MId=0x1021 ]
21.320 MENU_VALUES: SEND DSM_getNextMenuValue(MenuId=0x1021, LastValueId=0x1000) Extra: Text="Flight Mode"
22.220 MENU_VALUES: RESPONSE MenuValue: UPDATED: L[#0 T=V_nc VId=0x1000 Text="Flight Mode"[0x8001] Val=2 [0->10,0] MId=0x1021 ]
22.340 MENU_VALUES: SEND DSM_getNextMenuValue(MenuId=0x1021, LastValueId=0x1000) Extra: Text="Flight Mode"
```

charbaugh1, 2022-12-19:
My log looks different than yours.  I tried the test and got the following.  I highlighted the FM info.  Physically, when I went into forward programming, the switch was in the FM1 position.  I moved it to FM2, FM3, FM2, FM1, FM2.  You can see, it never sees the move to FM1.  I ran a couple more tests, changing the minimum value of the output to different values, but saw no change in behavior.  For reference, I have included screenshots of the output screen with the switch in all 3 positions.  I have also tried moving to another physical switch and a different output channel to test with no change.  The last thing I tried was resetting the receiver to factory default and reprogramming from scratch.

 

Radio Info:    tx16s EdgeTX  2.8.0

Date      :    2022-12-19   13:9

DsmLib Version :    0.5

DSM_StartConnection()

0.000 RX_VERSION: SEND DSM_getRxVersion()

0.100 RX_VERSION: RESPONSE Receiver=AR630 Version 2.38.5

0.150 MENU_TITLE: SEND DSM_getMainMenu()

0.220 MENU_TITLE: RESPONSE Menu: M[Id=0x1000 P=0x0 N=0x0 B=0x0 Text="Main Menu"[0x4B]]

0.270 MENU_LINES: SEND DSM_getFirstMenuLine(MenuId=0x1000)

0.360 MENU_LINES: RESPONSE ReqTXInfo(LineNum=0 DataType=0x1  DATA=RX: 09 05 00 01 00 00 00 07 00 00 00 00 00 00 00 00)

0.490 MENU_REQ_TX_INFO: CALL DSM_SendTxInfo_20(#0 DATA=TX: 0x20 0x06 00 00 00 40)

0.550 MENU_REQ_TX_INFO: RESPONSE ReqTXInfo(LineNum=1 DataType=0x1  DATA=RX: 09 05 01 01 00 00 00 07 00 00 00 00 00 00 00 00)

0.590 MENU_REQ_TX_INFO: CALL DSM_SendTxInfo_20(#1 DATA=TX: 0x20 0x06 01 01 00 01)

0.700 MENU_REQ_TX_INFO: RESPONSE ReqTXInfo(LineNum=2 DataType=0x1  DATA=RX: 09 05 02 01 00 00 00 07 00 00 00 00 00 00 00 00)

0.750 MENU_REQ_TX_INFO: CALL DSM_SendTxInfo_20(#2 DATA=TX: 0x20 0x06 02 02 00 02)

0.850 MENU_REQ_TX_INFO: RESPONSE ReqTXInfo(LineNum=3 DataType=0x1  DATA=RX: 09 05 03 01 00 00 00 07 00 00 00 00 00 00 00 00)

0.890 MENU_REQ_TX_INFO: CALL DSM_SendTxInfo_20(#3 DATA=TX: 0x20 0x06 03 03 00 04)

0.940 MENU_REQ_TX_INFO: RESPONSE ReqTXInfo(LineNum=4 DataType=0x1  DATA=RX: 09 05 04 01 00 00 00 07 00 00 00 00 00 00 00 00)

1.000 MENU_REQ_TX_INFO: CALL DSM_SendTxInfo_20(#4 DATA=TX: 0x20 0x06 04 04 00 00)

1.100 MENU_REQ_TX_INFO: RESPONSE ReqTXInfo(LineNum=5 DataType=0x1  DATA=RX: 09 05 05 01 00 00 00 07 00 00 00 00 00 00 00 00)

1.150 MENU_REQ_TX_INFO: CALL DSM_SendTxInfo_20(#5 DATA=TX: 0x20 0x06 05 05 00 00)

1.210 MENU_REQ_TX_INFO: RESPONSE MenuLine: L[#0 T=M VId=0x1010 Text="Gyro settings"[0xF9]   MId=0x1000 ]

1.250 MENU_LINES: SEND DSM_getNextLine(MenuId=0x1000,LastLine=0)

1.360 MENU_LINES: RESPONSE MenuLine: L[#1 T=M VId=0x105E Text="Other settings"[0x227]   MId=0x1000 ]

1.400 MENU_LINES: SEND DSM_getNextLine(MenuId=0x1000,LastLine=1)

MENU_LINES: EVT_TOUCH_TAP 263,71

MENU_LINES: EVT_TOUCH_TAP 263,71

MENU_LINES: EVT_VIRTUAL_ENTER,  SelLine=0

2.610 MENU_LINES: DSM_GotoMenu(0x1010,LastSelectedLine=0)

2.620 MENU_TITLE: SEND DSM_getMenu(MenuId=0x1010 LastSelectedLine=0)

2.720 MENU_TITLE: RESPONSE Menu: M[Id=0x1010 P=0x0 N=0x0 B=0x1000 Text="Gyro settings"[0xF9]]

2.760 MENU_LINES: SEND DSM_getFirstMenuLine(MenuId=0x1010)

2.810 MENU_LINES: RESPONSE MenuLine: L[#0 T=M VId=0x1011 Text="AS3X Settings"[0x1DD]   MId=0x1010 ]

[... 106 lines cut ...]
15.930 MENU_VALUES: SEND DSM_getNextMenuValue(MenuId=0x1021, LastValueId=0x1000) Extra: Text="Flight Mode"

16.120 MENU_VALUES: RESPONSE MenuValue: UPDATED: L[#3 T=LM VId=0x1003 Text="Safe Mode"[0x1F8] Val=176|"Self-Level/Angle Dem" NL=(0->244,3,S=0) [0->244,3] MId=0x1021 ]

16.150 MENU_VALUES: SEND DSM_getNextMenuValue(MenuId=0x1021, LastValueId=0x1003) Extra: Text="Safe Mode"

18.190 MENU_VALUES: INACTIVITY TIMEOUT

 

 

 

 

 

 

 

 

Clark Harbaugh

frankiearzu, 2022-12-19:
Charles,
The important lines are the ones with UPDATED value for the Flight Mode line.. I filter only those lines below:
```
6.970 MENU_LINES: RESPONSE MenuValue: UPDATED: L[#0 T=V_nc VId=0x1000 Text="Flight Mode"[0x8001] Val=1 [0->10,0] MId=0x1021 ]
13.310 WAIT_CMD: RESPONSE MenuValue: UPDATED: L[#0 T=V_nc VId=0x1000 Text="Flight Mode"[0x8001] Val=2 [0->10,0] MId=0x1021 ]
15.890 WAIT_CMD: RESPONSE MenuValue: UPDATED: L[#0 T=V_nc VId=0x1000 Text="Flight Mode"[0x8001] Val=1 [0->10,0] MId=0x1021 ]
15.890 WAIT_CMD: RESPONSE MenuValue: UPDATED: L[#0 T=V_nc VId=0x1000 Text="Flight Mode"[0x8001] Val=1 [0->10,0] MId=0x1021 ]
```
That means that the RX is only using 2 flight modes.  (Val=1:Flight Mode 2, and Val=2:Flight Mode 3, what is missing is the Val=0: Flight Mode 1).

This is information that the RX is sending the TX script to display.. so the problem is outside the script.
the Flight Mode 1 (Val=0) is usually done when your switch is in the -100 range  (or +100 if reversed).

Did a little experiment, change the ch to transmit the THR for the flight mode.
what i saw:   (-100 -90) Flight Mode 1, (-89 -6) Flight Mode 3,  (-5 +5) Flight Mode 2, (+6 +89) Flight Mode 3, (+90 +100) Flight Mode 3.

What i see is that unless the switch output is within 10% of the intended end, it defaults to FM3 (Val=2).
I think one of your ends (either +100, or +100) is not reaching the intended output expected by the RX.

Check that your output for the switch channel is sending 988us for -100, 1500us for 0, and 2012us for +100. (or close to it);
You can go to model setup->output page, edit the channel for your output switch, and you should see something like this the next image.
At the end, the RX behaves bases on the duration of the pulse that is receiving, the value with "us" (micro seconds). 

[image]

charbaugh1, 2022-12-20:
I appreciate all of your help on this.  I’ll continue to chase it on my own, since it appears to be either a TX or RX hardware issue.  I’m betting it’s an RX issue, since all of my settings and parameters on the TX match yours (confirmed the 988, 1500, and 2012us for each position). By the fact there are two “Val=1” entries in a row shows where I physically switched to FM1, but the receiver didn’t receive/respond appropriately.    

 

And thank you so much for doing the work and programming of the script!

 

Clark Harbaugh

 

 <https://user-images.githubusercontent.com/32604366/208542150-797dc950-60be-4eed-9bec-52c9d7be00d1.png>

closed 2022-12-20 by charbaugh1