pascallanger/DIY-Multiprotocol-TX-Module #793: ZC-Z50 Cessna

filipkotoucek, 2022-12-26:
Hello. I just received ZC-Z50 Cessna without the main motor in front (don't know if that matters). I had touble to binding to it using irangex x4plus module with FW 1.3.3.20. TX is Taranix qx7 with edgeTX 2.7.1. I tried both subprotocols for GD00X even with fine tuning over the full spectrum with 40 increments.

This led me to thought if the protocol is not somehow different. I tried to dump the communication. Here is the log:

Only TX is on:
```
Protocol selected: 63, sub proto 3, rxnum 0, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=6, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7

Packet detected: bitrate=Using NRF
250K C=7 S=Y A= E7 E7 E7 E7 E7 P(4)= AA 67 64 01
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7
RX on channel: 7, Time:     0us P: AA 67 64 01
RX on channel: 7, Time:  2994us P: AA 67 64 01
RX on channel: 7, Time:  2994us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2991us P: AA 67 64 01
RX on channel: 7, Time:  2992us P: AA 67 64 01
RX on channel: 7, Time:  3000us P: AA 67 64 01
RX on channel: 7, Time:  2986us P: AA 67 64 01
RX on channel: 7, Time:  2991us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2988us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2986us P: AA 67 64 01
RX on channel: 7, Time:  2996us P: AA 67 64 01
RX on channel: 7, Time:  2993us P: AA 67 64 01
RX on channel: 7, Time:  2995us P: AA 67 64 01
RX on channel: 7, Time:  2993us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2987us P: AA 67 64 01
RX on channel: 7, Time:  2994us P: AA 67 64 01

1 RF channels identified: 7
--------------------------------
Identifying RF channels order.
Time between CH:7 and CH:51

Channel order:
7:     0us

--------------------------------
Identifying Sticks and features.
```

The com between model and TX happens on ch 48 and 51. On identifying sticks I moved:
1. rudder stick (first byte changed)
2. trims (second byte changed 10 -> 11 etc)
3. RATE? ( second byte chanched 10 -> D0). This is quite interesting because the change does not have any impact on plane behaviour. There are 2 buttons on the TX front side. 
   - Left button which changes the value in dump beeps once everytime I click it but does not have impact on planes behaviour.
   - Right button does not change the values in dump. First click beep twice, second click beeps once, third click beeps twice. And there is audible plane behavior change between those two modes, so I assume this would be the RATE button.

```
Protocol selected: 63, sub proto 3, rxnum 0, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=6, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,461

Packet detected: bitrate=Using NRF
250K C=51 S=Y A= 0a0 AA 67 64 01 P(2)= 80 10
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,468
RX on channel: 48, Time:     0us P: 80 10
RX on channel: 48, Time:  5995us P: 80 10
RX on channel: 48, Time:  5983us P: 80 10
RX on channel: 48, Time:  5984us P: 80 10
RX on channel: 48, Time:  5991us P: 80 10
RX on channel: 48, Time:  5982us P: 80 10
RX on channel: 48, Time:  5983us P: 80 10
RX on channel: 48, Time:  5990us P: 80 10
RX on channel: 48, Time:  5983us P: 80 10
RX on channel: 48, Time:  5982us P: 80 10
RX on channel: 48, Time:  5986us P: 80 10
RX on channel: 48, Time:  5982us P: 80 10
RX on channel: 48, Time:  5985us P: 80 10
RX on channel: 48, Time:  5986us P: 80 10
RX on channel: 48, Time:  5989us P: 80 10
RX on channel: 48, Time:  5980us P: 80 10
RX on channel: 48, Time:  5990us P: 80 10
RX on channel: 48, Time:  5984us P: 80 10
RX on channel: 48, Time:  5984us P: 80 10
RX on channel: 48, Time:  5990us P: 80 10
RX on channel: 48, Time:  5983us P: 80 10
Trying RF channel: ,49,50,51
RX on channel: 51, Time:     0us P: 80 10
RX on channel: 51, Time:  5984us P: 80 10
RX on channel: 51, Time:  5992us P: 80 10
RX on channel: 51, Time:  5981us P: 80 10
RX on channel: 51, Time:  5993us P: 80 10
RX on channel: 51, Time:  5980us P: 80 10
RX on channel: 51, Time:  5988us P: 80 10
RX on channel: 51, Time:  5988us P: 80 10
RX on channel: 51, Time:  5981us P: 80 10
RX on channel: 51, Time:  5980us P: 80 10
RX on channel: 51, Time:  5998us P: 80 10
RX on channel: 51, Time:  5984us P: 80 10
RX on channel: 51, Time:  5980us P: 80 10
RX on channel: 51, Time:  5979us P: 80 10
RX on channel: 51, Time:  5993us P: 80 10
RX on channel: 51, Time:  5980us P: 80 10
RX on channel: 51, Time:  5983us P: 80 10
RX on channel: 51, Time:  5982us P: 80 10
RX on channel: 51, Time:  5989us P: 80 10
RX on channel: 51, Time:  5990us P: 80 10
RX on channel: 51, Time:  5986us P: 80 10
Trying RF channel: ,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,7

2 RF channels identified: 48 51
--------------------------------
Identifying RF channels order.
Time between CH:48 and CH:51
Time:  2996us
Time:  2996us
Time:  3001us
Time:  2996us
Time:  3008us
Time:  2984us
Time:  2993us
Time:  2994us
Time:  2988us
Time:  2991us
Time:  2996us
Time:  2996us

Channel order:
48:     0us
51:  2984us

--------------------------------
Identifying Sticks and features.
P: 80 10
P: 81 10
P: 86 10
P: 8B 10
P: 90 10
P: 95 10
P: 99 10
P: 9B 10
P: 9C 10
P: 9E 10
P: 9F 10
P: 9D 10
P: 97 10
P: 90 10
P: 87 10
P: 80 10
P: 7C 10
P: 77 10
P: 72 10
P: 6B 10
P: 67 10
P: 61 10
P: 5E 10
P: 5A 10
P: 57 10
P: 53 10
P: 52 10
P: 5C 10
P: 69 10
P: 76 10
P: 80 10
P: 83 10
P: 89 10
P: 90 10
P: 97 10
P: 9D 10
P: A2 10
P: A5 10
P: A7 10
P: AA 10
P: AB 10
P: AC 10
P: AD 10
P: AE 10
P: AF 10
P: AE 10
P: AD 10
P: AB 10
P: A7 10
P: 9F 10
P: 95 10
P: 89 10
P: 80 10
P: 7F 10
P: 7C 10
P: 7A 10
P: 77 10
P: 74 10
P: 73 10
P: 72 10
P: 70 10
P: 6E 10
P: 6C 10
P: 6B 10
P: 6A 10
P: 69 10
P: 6A 10
P: 6B 10
P: 6C 10
P: 71 10
P: 73 10
P: 77 10
P: 7B 10
P: 7C 10
P: 7F 10
P: 80 10
P: 7E 10
P: 7B 10
P: 77 10
P: 74 10
P: 72 10
P: 71 10
P: 70 10
P: 6F 10
P: 70 10
P: 75 10
P: 7F 10
P: 80 10
P: 81 10
P: 82 10
P: 83 10
P: 84 10
P: 85 10
P: 86 10
P: 85 10
P: 84 10
P: 83 10
P: 82 10
P: 81 10
P: 80 10
P: 80 11
P: 80 12
P: 80 13
P: 80 12
P: 80 11
P: 80 10
P: 80 0F
P: 80 0E
P: 80 0D
P: 80 0E
P: 80 0F
P: 80 10
P: 80 D0
P: 80 10
```

Sadly I don't know how to catch the bind in the logs.

Is there anybody who would know how to help me find where is the bug or what am I doing wrong?

filipkotoucek, 2022-12-26:
I should add that I red the previous related issues
https://github.com/goebish/nrf24_multipro/issues/58
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/646

pascallanger, 2022-12-27:
I'm not in front of my PC to check if this is close to an existing protocol but I would say that the address of the normal packets is only 4 bytes long, please change this (from 5 to 4) and do a new dump. This is probably why you do not mention anything about throttle and also that you don't see a change when you swap the rate.

filipkotoucek, 2022-12-27:
You are definitely correct that normal communication uses 4b address. In the next log throttle is 1st byte, rudder is 2nd, trim and rate 3rd.
[addr_4b.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10307960/addr_4b.txt)

filipkotoucek, 2022-12-27:
To my poor understanding, it seems similar to KF606.

pascallanger, 2022-12-27:
Yeah I'll add it under the KF606 but the normal packets are different. I don't think I'll have the time to look into it today. I haven't looked at the data provided yet but if you could write down which byte takes what value with min, mid, max values for the channels and trim as well as what happens when you press the switches multiple times.

filipkotoucek, 2022-12-27:
I'm done for today. With no luck yet. https://github.com/filipkotoucek/DIY-Multiprotocol-TX-Module/commit/c8755c1c3281b95fbd41c869e5bcfe7ae8dec718

What would be helpful is a second module to scan the communication.

pascallanger, 2022-12-28:
@filipkotoucek Looks good but you've done a couple of mistakes in your implementation.

pascallanger, 2022-12-28:
@filipkotoucek I've added all my comments to your commit https://github.com/filipkotoucek/DIY-Multiprotocol-TX-Module/commit/c8755c1c3281b95fbd41c869e5bcfe7ae8dec718

filipkotoucek, 2022-12-29:
Thanks for the comments. I will fix it today evening and try one more time.

filipkotoucek, 2022-12-29:
Still no bind. https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/commit/31f6eeae3b4ff57d360a4f91284235148a9d53d7

Is the data somehow encrypted or encoded? I have few NRF24 modules laying around. I was wondering if I can just simply sniff on the channels 7, 48 and 51 for raw data how it looks with original TX and multiprotocol module and compare the differences.

pascallanger, 2022-12-29:
Ok I'll check your latest code, I might have missed something in my first review.

pascallanger, 2022-12-29:
I think I've found the issue. You've declared and initialized len at the wrong place.
Move the line [uint8_t len = KF606_PAYLOAD_SIZE;](https://github.com/filipkotoucek/DIY-Multiprotocol-TX-Module/blob/31f6eeae3b4ff57d360a4f91284235148a9d53d7/Multiprotocol/KF606_ccnrf.ino#L49) at the beginning of the function before this line https://github.com/filipkotoucek/DIY-Multiprotocol-TX-Module/blob/31f6eeae3b4ff57d360a4f91284235148a9d53d7/Multiprotocol/KF606_ccnrf.ino#L34

pascallanger, 2022-12-29:
In fact there is more... You have messed up the packet length management all together.
I've added all my comments to your commit https://github.com/filipkotoucek/DIY-Multiprotocol-TX-Module/commit/c8755c1c3281b95fbd41c869e5bcfe7ae8dec718

filipkotoucek, 2022-12-30:
Many thanks for your help. Now the plane is working as it should. https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/pull/797

I made one more change to use the IDs from my dump for this subprotocol by default. I'm not sure if every pair (TX/plane) has unique IDs or what.

pascallanger, 2022-12-30:
Please test the proposed changes that I've added in your pull request to enable multiple IDs.

filipkotoucek, 2023-01-01:
Yes I will test it in few days.

filipkotoucek, 2023-01-11:
So I tested it with code in updated PR. Everything works. If you are fine with it, you can merge.

closed 2023-01-12 by pascallanger