pascallanger/DIY-Multiprotocol-TX-Module #795: Microzone Protocol

richardclli, 2022-12-29:
Microzone protocol
Radio: MC10
Receiver: MC9002
Chip used: A7105

Will provide SPI dumps in this issue.

richardclli, 2022-12-29:
Dump for controller switch on without receiver, chip init sequences.

[Switch-On.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10317785/Switch-On.zip)

richardclli, 2022-12-29:
The Microzone supports 6 different sub-protocols, and now only the first 4 (M1,M2,M3,M4) is used, M5 and M6 is reserved.

I have summarized the A7105 commands from M1 to M6 in the attached Excel file.

M1 is quite different from others.
M2 and M3 are quite similar.
M4-M6 uses a different payload size.

[Switch-On-SubProtocols.xlsx](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10318078/Switch-On-SubProtocols.xlsx)

richardclli, 2022-12-29:
Dump for binding a receiver (M2 subprotocol)
1. Power on receiver
2. Press binding button and let receiver goes in binding mode
3. Start SPI capture
4. Power on radio
5. Bind done

[Bind.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10318084/Bind.zip)

richardclli, 2022-12-30:
For channel encodings, M2 sub-protocol can support max 8 channels:

Payload length: 11 bytes
Bits per channel: 10 bits
Ch1 = packet[0] + (packet[1] & 0x03) << 8
Ch2 = packet[1]  >> 2 + (packet[2] & 0x0F) << 8
Ch3 = packet[2] >> 4 + (packet[3] & 0x3F) << 8
......

Final byte is always 0, not sure whether it has other purpose.

richardclli, 2023-01-03:
@pascallanger Interested in working out this protocol together?

richardclli, 2023-01-04:
Freq Hoping seems follows a fixed sequence:
0x11, 0x61, 0x4D, 0x9D, 0x25, 0x75, 0x07, 0x57, 0x1B, 0x6B, 00x39, 0x89, 0x2F, 00x7F, 0x43, 0x93

pascallanger, 2023-01-08:
Sure we can have a look. I've been quite busy lately so don't expect prompt responses.

richardclli, 2023-01-13:
> Sure we can have a look. I've been quite busy lately so don't expect prompt responses.

No problem, I will work out more details first.

richardclli, 2023-02-07:
@pascallanger Could you suggest which protocol source code is best for me I start the coding?

pascallanger, 2023-02-07:
I think you need to create a new protocol.

richardclli, 2023-02-09:
> I think you need to create a new protocol.

Yes, any suggestion on which protocol source should I copy and code for this new protocol.

MRC3742, 2023-02-11:
Not sure if this helps but I have an older Airfield branded Microzone (MC6DR) receiver that binds and operates on the Kyosho - Hype sub protocol.

This was out of an Airfield Corsair 1450mm RTF plane and the [MC6S TX FCC ID](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiTmvrH0Y39AhUBlWoFHeOqBgcQFnoECAkQAQ&url=https%3A%2F%2Ffccid.io%2FZMKMC4DFMCD6DF%2FLabel%2FLabel-Info-1478026&usg=AOvVaw2eU_v_7fvtq_hvMXgVXBkb) comes up as a Microzone Product. 

This protocol may be the same as the M1 of your MC10 ?

Rich

richardclli, 2023-02-12:
Thank you for your information.  Let me verify this.

> Not sure if this helps but I have an older Airfield branded Microzone (MC6DR) receiver that binds and operates on the Kyosho - Hype sub protocol.
>
> This was out of an Airfield Corsair 1450mm RTF [...]

richardclli, 2023-06-11:
@pascallanger Not sure if you may have some time to take a look on what is captured.  I may need some guide in doing the programming.

pascallanger, 2023-10-30:
Have you ever finished this protocol ?

richardclli, 2023-10-31:
Not yet, better to get some help from you.

pascallanger, 2023-10-31:
Sure, can you explain me what you've done so far? (Sorry just trying to catch up with things lately)

richardclli, 2023-10-31:
I just captured the SPI, then try to find the samples of M1 to M6 receivers, but can only got a M2.  Well, we can just work out M2 first, anyway.

pascallanger, 2023-10-31:
Ok so let's go for M2

richardclli, 2023-11-01:
> Ok so let's go for M2

Yeah!

richardclli, 2023-11-07:
@pascallanger So we can start with the spi dumps attached with this issue.