pascallanger/DIY-Multiprotocol-TX-Module #801: New XN297 Based Protocol, used in J20 plane.

richardclli, 2023-01-08:
I have captured the first log, may need some help to move forward.  This is my first time ti use XN297Dump.

[J20.log](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10367910/J20.log)

pascallanger, 2023-01-08:
@richardclli
The log indicates that XN297 formatted packets have been received with the following parameters: Rate=1Mb, RF Channel=19, Scrambled=Y, 5 bytes of address= 00 CE 31 9B 73.
There are plenty of RF frequencies detected but my guess is that some of them are just ghost, move away or change the orientation of the TX to decrease the number of false positive. There is probably only 3 real RF frequencies in use since the payload appears every 24462µs and the time between the channels seem to be 8154µs, I would say that the channels are 19, 26 and 30.
7 bytes of payload= 80 81 80 20 00 00 5E 
There is a checksum at the end of the packet on byte 6 (5E above). byte 6=(sum of the payload bytes 0..5) xor 0xFF
I can't do anything more without more information: bind packets, sticks, features...

pascallanger, 2023-01-08:
If this is this plane: https://fr.aliexpress.com/item/1005003521725715.html
Then it might be close to this protocol FX_nrf24l01.ino since the original radios have FX written on them.

richardclli, 2023-01-10:
> If this is this plane: https://fr.aliexpress.com/item/1005003521725715.html Then it might be close to this protocol FX_nrf24l01.ino since the original radios have FX written on them.

The shape is similar but not the same.  I will try to work out more detail according to your comments.

richardclli, 2023-01-13:
> @richardclli The log indicates that XN297 formatted packets have been received with the following parameters: Rate=1Mb, RF Channel=19, Scrambled=Y, 5 bytes of address= 00 CE 31 9B 73. There are [...]

I have done a second dump by putting the radio farther away, same as what you guess, 3 channels are detected: 19, 26, 56

[J20-2.log](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10409148/J20-2.log)

richardclli, 2023-01-13:
For the payload:

P[0] = Ail: left = 00, right = FF
P[1] = Ele: up = 00, down = FF
P[2] = Rud: left = 00, right = FF
P[3] = 0x20, not changing except pressing trim buttons, may assume it is always 0x20, trims can be done in mixer
P[4:0] = toggle switch
P[4:1-2] = 00 mode A, 01 mode B, 10, mode C
P[5] = 00 always
P[6] = throttle, seems scrambled, will change when moving others and throttle (maybe checksum related)

richardclli, 2023-01-13:
@pascallanger Can you advice how can I capture the binding information?

pascallanger, 2023-01-13:
I don't really understand:
P[6] = Thr and checksum: sum(P[0-6]) & 0xff = Thr: up = 00, down = FF
It needs a xor to calculate the check which I can't see. How can it be that the throttle is mixed with the checksum? The checksum is therefore useless... I'm wondering if the address is instead on 4 bytes and throttle is the first byte... At least it doesn't make any sense for me right now.

For the bind, the bind method must be indicated in the documentation. If it is short, you can force the rate to 1Mb (instead of auto that you are using) and scan the rf channels manually (or auto scan by setting the rf channel to -1).

richardclli, 2023-01-13:
> I don't really understand: P[6] = Thr and checksum: sum(P[0-6]) & 0xff = Thr: up = 00, down = FF It needs a xor to calculate the check which I can't see. How can it be that the throttle is mixed [...]

Well, I think you may consider that the thr is just calculated using sum(P[0-6]) & 0xff and no checksum in payload.

For binding, I think it is very short, and required the RX to be powered up first.  And I found that the address are constant for different RX.  Could the binding done in RX and it just remember the TX address?

pascallanger, 2023-01-13:
P: 81 81 80 20 00 00 5D
P: 80 81 80 20 00 00 5E
P: 80 00 80 20 00 00 DF
P: 81 00 80 20 00 00 DE
Strange since all the lines of J20.log matches P[6]=sum(P[0..5]) xor 0xff.

For the bind, the RX just needs to learn the TX address and which RF channels to use in normal mode that's all.  So what you need is just to capture one bind packet, usually the normal address is in clear text, the rf channels could appear or are calculated from the normal address.

richardclli, 2023-01-13:
Yes, because I did not move the stick and the stick is in down position (0xff).  The last byte will change with all stick direction changes.  May need more test to see if any other bytes need to be opt out, because there are 2 in the capture.  Here is the log that I move only the throttle: Down -Up-Down-Up....

[J20-5.log](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10410121/J20-5.log)

> Strange since all the lines of J20.log matches P[6]=sum(P[0..5]) xor 0xff: P: 81 81 80 20 00 00 5D P: 80 81 80 20 00 00 5E P: 80 00 80 20 00 00 DF P: 81 00 80 20 00 00 DE
> 
> For the bind, the RX just needs to learn the TX address and which RF channels to use in normal mode that's all. So what you need is just to capture one bind packet, usually the normal address is in clear text, the rf channels could appear or are calculated from the normal address.

richardclli, 2023-01-13:
Let me see if I can capture the bind packet, by fix capture of the 3 channels.

richardclli, 2023-01-13:
Checked the final byte will change with all moves: any stick dir, mode abc and the toggle switch

pascallanger, 2023-01-13:
> Checked the final byte will change with all moves: any stick dir, mode abc and the toggle switch. And this confirms the calculation is
> 
> thr = sum(P[0-6]) & 0xff

This is wrong... Look at your dump, the address is changing it means that the address is 4 bytes long only. If you change the address to 4 bytes then I expect the 1st byte of the payload will be throttle going from 0 to ff.
RX: 24426us C=19 S=Y A= **00** CE 31 9B 73 P(7)= 80 81 80 20 00 00 5E
RX: 24420us C=19 S=Y A= **50** CE 31 9B 73 P(7)= 80 81 80 20 00 00 54
RX: 24427us C=19 S=Y A= **98** CE 31 9B 73 P(7)= 81 81 80 20 00 00 44
RX: 24425us C=19 S=Y A= **14** CE 31 9B 73 P(7)= 81 81 80 20 00 00 35
RX: 24425us C=19 S=Y A= **AC** CE 31 9B 73 P(7)= 80 81 80 20 00 00 29
...

richardclli, 2023-01-13:
I found the config, it is the receiver no.

Attached is the log with the bind packet.  You are right, if using 4 address, then the first byte is the throttle:
P[0] = thr: down = 00, up = FF
P[1] = Ail: left = 00, right = FF
P[2] = Ele: up = 00, down = FF
P[3] = Rud: left = 00, right = FF
P[4] = 0x20, not changing except pressing trim buttons, may assume it is always 0x20, trims can be done in mixer
P[5:0] = toggle switch
P[5:1-2] = 00 mode A, 01 mode B, 10, mode C
P[6] = 00 always
P[7] = checksum

[J20-bind2.log](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10412817/J20-bind2.log)

pascallanger, 2023-01-13:
Ok we are getting close to be able to code this protocol.
Looking at the bind dump, I don't think this is the correct rf channel since only 1 packet has been received. I would expect far more. So I think it's just a ghost packet like you had at the beginning if you remember. Try to scan through the rf channels while binding is in progress (the plane/RX does not need to be turned on).
What the toggle switch is for? and mode ABC?
Do you have a link to that plane?

richardclli, 2023-01-14:
Since the binding only stay for 2 secs.  I did a manual scan (J20-bind-scan.log).  Finally figured out the channel 51 and 55 can receive a lot of binding packets, then I take the radio farther to test again to see which one is real channel and finally got channel 51 (J20-bind3.log).

[J20-bind-scan.log](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10416719/J20-bind-scan.log)
[J20-bind3.log](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10416720/J20-bind3.log)

richardclli, 2023-01-14:
The toggle switch is D/R, 0 smaller throw, 1 larger throw
A, B, C mode: A = 6G normal, B = 6G larger ele throw, C = 3D

richardclli, 2023-01-14:
I will try to see if I can do the coding, knowing you are quite busy recently.

richardclli, 2023-01-14:
FYI, I got the confirmation from the manufacturer, this plane is FX9630 and is only at pre-production stage, not selling worldwide yet.

pascallanger, 2023-01-14:
Mary be they can send me one 😉

richardclli, 2023-01-16:
For binding packet:
P[0-3] = TX RX address
P[4] = hoping_frequency[1]
P[5] = hoping_frequency[2]
P[6] = checksum (same method as normal payload)
P[7] = always 0x55

hoping_frequency[0] is always 0x13 (19)

commit 0fed2486 (2023-01-16): FX9630 protocol support, resolves #801 (#805)

New protocol FX9630