pascallanger/DIY-Multiprotocol-TX-Module #810: DSM FwdPrg_05_BW throws low memory on Zorro

BuddhaSource, 2023-01-18:
Hey guys, looking forward to scrip to be working.

I tried the old default DSM Fwd script which shows up as blank screen on Zorro. Then I tried the new one "DSM FwDPrg_05_BW" on my Radiomaster Zorro but I keep getting this low memory error.

I really doubt its low memory because I am able to run more graphical scripts on Zorro.
Let me know how can I debug for you.

pascallanger, 2023-01-19:
If I'm not wrong this is a known issue.
@frankiearzu can you confirm ?

BuddhaSource, 2023-01-20:
I saw ver 53 has a note for fixing low memory.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/Lua_scripts/DSMLIB#changes-and-fixes

Also, can I directly call the safe bind programming function without loading everything?
I can try isolating that function so that on app start, it executes just that.

Cheers.

frankiearzu, 2023-01-20:
Buddha,
Currently, it will not run in some radios due to memory limiations.   Have been only able to test in in Companion (PC software) simulating the smaller screen (Zorro), but the simulator don't impose any memory constrains.
A friend tested in a RadioMaster Boxter (newer processor) and worked.  Older radios are more limited in memory.

I know that i can improve a bit the memory use, but unfortunatly, don't have access to a zorro to test.
I can send you a message when i do some more improvements, and we can test again to see if worked. 

Frankie

BuddhaSource, 2023-01-20:
Thank you @frankiearzu 

If in any way I can debug, that would be great. I can run them here and paste some output.
Alternatively, do you know which functions I can execute to simply do DSM initializing for SAFE? If I can modify the code to just run that on the load it will solve my current problem for now.

Also, any script to check memory usage?

frankiearzu, 2023-04-05:
@BuddhaSource  have been working on gettting a low memory version that will work on the Zorro and other low memory radios. Have a limited version where you can do the majority of Editing, but you cannot setup a complete new Plane.. will work on that later.   You can download a zip file with all the files ready to put on your radio: 
https://github.com/frankiearzu/DSMTools