pascallanger/DIY-Multiprotocol-TX-Module #814: Frsky Archer R10Plus

r41065, 2023-01-20:
New FrSky Archer R10+ does not bind with 4in1.

Works fine with original XJT ( X9D) or ISRM (X9Lite) in ACCST D16 v2 ( LBT,  not testet if FCCv2 also works, FCCv1 did not)

Tried all  D16 Modes ( FCC /LBT in v1 and v2 ) also with different Values of Finetuning.

In Cloned Modus Receiver also works, but no Chance to bind to a different ID...

kochpet, 2023-02-23:
@pascallanger Hi, would it be helpful if I send you a Saleae Logic trace of the ACCST D16 binding process with the Archer R10Plus?

jogybaer0815, 2023-03-16:
it is a pity that these important receiver types are not supported. No idea how the developers can be supported to work here...
I own Archer R10 and can donate one.

ShahzaibNayyar, 2023-04-07:
Hi,
I was wondering if anyone has successfully paired the Archer SR10+ Receiver with the RadioMaster TX16s. The Archer SR10+ Receiver's documentation indicates that it can operate with both the ACCST D16 V2 and ACCESS protocols, and the TX16s also supports ACCST D16 V2. So, what seems to be the problem?

mick-74, 2023-04-07:
Hi also for me Radiomaster TX16s with latest edgetx and latest 4-in-1 module doesn't work.

Jetiman, 2023-04-08:
Here also. TX16s and Archer SR10+ will not work.

r41065, 2023-04-18:
FrSky introduces a complete Serie of Archer Plus Receiver
( RS Mini, RS+, R6+, GR6+, R8+, GR8+, SR8+, SR10+, R10+ )

hopefully there will be a solution

MikeBland, 2023-06-13:
I very recently acquired an Archer Plus R10+, so I've found a solution that I've tested working on two, different, MPM modules.
File FrSkyX_cc2500.ino:
Change line 54 from:
memcpy(&packet[7],"\x00\x18\x0A\x00\x00\xE0\x02\x0B\x01\xD3\x08\x00\x00\x4C\xFE\x87\xC7",17);
to
memcpy(&packet[7],"\x00\xCC\x00\x00\x00\x70\x14\x15\x00\xD3\x08\x00\x00\xCE\xE2\x85\xC7\x00\x00\x00\x00",21);
Change line 56 from:
memcpy(&packet[7],"\x27\xAD\x02\x00\x00\x64\xC8\x46\x00\x64\x00\x00\x00\xFB\xF6\x87\xC7",17);
to
memcpy(&packet[7],"\x27\xFB\x00\x00\x00\xBC\xEF\x19\x00\x26\x07\x00\x00\xB7\xED\x85\xC7\xA7\xA7\xA7\xA7",21);
Change liness 66 & 67 from:
packet[20] ^= 0x0E ^ rx_tx_addr[3];	// Update the ID
packet[21] ^= 0x1C ^ rx_tx_addr[2];	// Update the ID
to
packet[20] ^= rx_tx_addr[3];	// Update the ID
packet[21] ^= rx_tx_addr[2];	// Update the ID
and add a new line 68 as:
packet[22] ^= rx_tx_addr[1] & 0x3F;				// Update the ID

It appears the Archer Plus receivers may be more selective, so the MPM only operates over a reduced tuning range, the first one I got working had a tuning range of -20 to +20, and the second from -25 to +15, so you may need to try several different tuning values to obtain a bind.

mick-74, 2023-06-13:
Hi Thanks for the sharing, but sorry, can you help me to test it with module 4in1 installed on radiomaster tx16s pls?

MikeBland, 2023-06-13:
You need to be able to compile the firmware yourself until the changes are added to the repository.

pascallanger, 2023-06-13:
I'll include it in the next build.

mick-74, 2023-06-13:
thanks guys.  do you know when it will be available?  I don't know how to compile the firmware so I need to wait you mate.

jogybaer0815, 2023-06-16:
It is easy to compile with arduino - @pascallanger described it well how to do it..

Great. Works absolutely perfectly.
Big thanks to @MikeBland

mick-74, 2023-06-16:
Hi guys,
some one can share version for Radiomaster TX16S?  I appreciate a lot :D

jogybaer0815, 2023-06-16:
what channel order?

mick-74, 2023-06-16:
AETR

jogybaer0815, 2023-06-16:
please... (cut .csv)

[multi-stm-1.3.3.27_AETR.bin.csv](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/11769804/multi-stm-1.3.3.27_AETR.bin.csv)

mick-74, 2023-06-16:
Really a BIG THANKS!  I will try next week when I come to home.

mick-74, 2023-06-22:
Hi all, again sorry.
I am trying to use LUA from FRSKY that for them is for opentx but I am not able to make lua working fine.  wizard doesn't work fine, Image missing and so on.  Anyone has found a lua script to set gyro?  thx

mick-74, 2023-06-28:
rename with same name, removing cav extension.

dbock300, 2023-06-28:
When I try to rename it the .csv extension doesn't appear. So it's not there to delete. Windows 10

mick-74, 2023-06-28:
set known extension visible.

dbock300, 2023-06-28:
Got it thanks,  hidden file extensions.
Receiver now bound.
Next for Frsky LUA script to work with Edge

BladeScraper-Designs, 2023-06-30:
> please... (cut .csv) Mode Multi FrSkyX2 D16 - my optimum Freq. fine tune: +25
> 
> [multi-stm-1.3.3.27_AETR.bin.csv](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/11769804/multi-stm-1.3.3.27_AETR.bin.csv)

Confirmed this .bin is working well.  I have an iRangeX IRX4 and it flashed just fine and now my Archer Plus RS Mini binds successfully.
[image]

UltraFly, 2023-07-01:
I was encountering the same binding issue between my TX16S and SR10+. I updated the firmware on the SR10+ using the FrSky cable (no joy), and then flashed the Multi-Module with the "multi-stm-1.3.3.27_AETR.bin" firmware you provided.
The receiver now binds correctly but testing with servos plugged into it reveals that they move correctly but in a very jerky manner. I wonder if this firmware really fixes the issue completely or if there is still a timing issue or something.

MikeBland, 2023-07-01:
The change only affects the binding. I tested it using a X9LiteS with a URUAV (Lite) module to a R10+ and everything works smoothly.

mick-74, 2023-07-01:
Hi, on my side with tx16s 4in1 all seems ok, the issue that I have now is related to the LUA that doesn't work also following frsky manual that suggest 1.0.2 firmware and specific lua.  no way to calibrate or set receiver.  also STK doesn't work well.. :(

UltraFly, 2023-07-02:
Maybe something went wrong when I updated to the v1.0.5 firmware for the SR10+.
That firmware is apparently a pretty substantial change, they change the channels used for stabilisation and S-Port behavior so i guess any testing should be done on this one

Ralf-W, 2023-08-18:
Today I have bound my TX16S with internal 4in1 MPM successfully to an Archer Plus GR6 (GR6+)  using V1.3.3.27. Binding was immediately successful in a range from -2 to +51. Thus I have to set frequency tuning to +26 for my TX16S with internal 4in1 MPM.
Servos run smoothly with this setting. Voltage telemetry worked directly after binding. The internal variometer came alife after the first power cycle.

Thanks to everybody, making this work!

Then I flashed mm-stm-serial-aetr-v1.3.3.28.bin - and no binding was possible within a frequency range from -80 to +80. The green LED starts flickering while the TX is in bind mode within a frequency tuning range from about -5 to +50.

MRC3742, 2023-08-19:
> Then I flashed mm-stm-serial-aetr-v1.3.3.28.bin - and no binding was possible within a frequency range from -80 to +80. The green LED starts flickering while the TX is in bind mode within a frequency tuning range from about -5 to +50.

That's because the code changes and additions listed above, by MikeBland, have not yet been added to the master branch.
I've attached a zip file with the new self compiled code. (with the updates listed above added for the new Archer Plus Receivers)
[mm-stm-1.3.3.28_AETR_PlusFix.bin.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/12384968/mm-stm-1.3.3.28_AETR_PlusFix.bin.zip)

The lua scripts should now also work using this firmware file. 
This is one of the fixes that was added to the master branch for 1.3.3.28

Rich

Ralf-W, 2023-08-19:
Thanks Rich!
I just tested your extended version mm-stm-1.3.3.28_AETR_PlusFix.bin: As expected it binds successfully to my Archer Plus GR6 (GR6+)

Is this version 1.3.3.28, or 1.3.3.27 safe to fly an airplane?
How large is the risk that their improvements damage proven features of the stable version 1.3.3.20?
Where can I find  a changelog between 1.3.3.20 and 1.3.3.28?
Or where/how can I compare the source codes?

MRC3742, 2023-08-19:
Review the commits to the master branch since 1.3.3.20 was released (Oct 12, 2022)
[https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/commits/master](url)

I would consider this a test firmware! These changes appear to work but you should range test first and fly it in a safe area with an inexpensive foamie before trusting it.

Rich

Ralf-W, 2023-08-19:
OK! Thamks again!

dbock300, 2023-08-20:
I used the new firmware 1.3.3.28 fix.
Receiver bound no problem flew with no issues. Used the new lua script for setup worked fine. But the calibration lua script executes but does not display the x,y,z numbers or the receiver orientation as the old lua script did.

ahmad-kassem, 2023-08-24:
I flashed V1.3.3.28_AETR_PlusFix.bin to my TX16S, and ACCST V1.0.5 to my Archer Plus RS, I can bind the receiver but not all telemetry sensors are being discovered and therefore Betaflight lua is not working.
Any Advice?

Ralf-W, 2023-08-24:
Did you power cycle the Rx after binding? Then, delete all sensors and search sensors again. In my tests it took some time before all available telemetry sensors were found. Finally stop sensor search. 
Hope this helps. I my case it did the trick.

dbock300, 2023-08-24:
Can anyone help with new Lua script ?
The calibration does not show the orientation or the x,y,z values.

ahmad-kassem, 2023-08-24:
> Did you power cycle the Rx after binding? Then, delete all sensors and search sensors again. In my tests it took some time before all available telemetry sensors were found. Finally stop sensor search. Hope this helps. I my case it did the trick.

I did power cycle several times, and i waited for 3 minutes for the telemetry sensors scan, but still i am not finding all the sensors.
what else could I try?

pascallanger, 2023-08-25:
Available in v1.3.3.29

RudiKleinSA, 2023-08-27:
Thank you for above.

Where do I find v1.3.3.29?

Ralf-W, 2023-08-27:
v.1.3.3.29 is here:
https://downloads.multi-module.org/latest-test/

RudiKleinSA, 2023-08-27:
Thank You very much.

I have the Archer plus RS8

I updated to my TX16S Multimodule to v1.3v3v29 but cannot do Frsky S8R / S6R RX calibration.

I am on v2.9.0-rc2 Providence

The calibration goes to the first screen saying to put the receiver in the following position eg Flat

X=Green
Y=Green
Z=RED

I also turned the receiver in all directions - the stabilisation seems to work, but I cannot do the calibration and plane setup.

dbock300, 2023-08-27:
Similar problem with SR10 plus. Doesn't show orientation or values. 3rd report no reply

pascallanger, 2023-08-27:
I don't have any of these RXs so no idea what's going wrong... If someone ship or sponsor a RX then I can have a look.

MRC3742, 2023-08-27:
I don't have one of these to test either but reviewing the lua script I see that the following code is listed for the X,Y,Z values.

local calibrationFields = {
  {"X:", VALUE, 0x9E, 0, -100, 100, "%"},
  {"Y:", VALUE, 0x9F, 0, -100, 100, "%"},
  {"Z:", VALUE, 0xA0, 0, -100, 100, "%"}
}

Has anyone tested that this script is working on a genuine FrSky TX running OTX?

Rich

dbock300, 2023-08-27:
I'm running Edge tx which might be the issue.

RudiKleinSA, 2023-08-29:
[image]
[image]
[image]
[image]

BladeScraper-Designs, 2023-08-29:
This is not a Multiprotocol issue.  You need to download the OpenTX Lua script from the receiver product page.  The new receivers have a different system and thus have diff configuration requirements.

Keith

[image]<https://user-images.githubusercontent.com/93178003/264102822-803e7f47-3312-4e93-b50b-77d2c8067c9d.png>
[image]<https://user-images.githubusercontent.com/93178003/264103175-a931f459-f23d-4592-b265-72ef8c033e35.png>
[image]<https://user-images.githubusercontent.com/93178003/264103650-d8767888-2dde-4d4f-94af-d1fb6631a906.png>
[image]<https://user-images.githubusercontent.com/93178003/264103388-735fcafa-cc6d-4dd1-991c-078170922d2f.png>

dbock300, 2023-08-29:
Keith
I have downloaded the new Lua script.

[image]<https://user-images.githubusercontent.com/93178003/264102822-803e7f47-3312-4e93-b50b-77d2c8067c9d.png>
[image]<https://user-images.githubusercontent.com/93178003/264103175-a931f459-f23d-4592-b265-72ef8c033e35.png>
[image]<https://user-images.githubusercontent.com/93178003/264103650-d8767888-2dde-4d4f-94af-d1fb6631a906.png>
[image]<https://user-images.githubusercontent.com/93178003/264103388-735fcafa-cc6d-4dd1-991c-078170922d2f.png>

RudiKleinSA, 2023-08-29:
Hi Keith,

Thank you very much.

Could you please just point me in the right direction.

mick-74, 2023-08-29:
Hi all,
first of all thanks for all changes, I have edgetx+ tx16s radiomaster+ MTM .30.

I use frsky archer+ sr10+ and I am able to bind it and from new FW and lua script I can get all gyro value, but the Issue is related to calibration, I can open lua script for calibration, ask me to press enter  but script stop there,  no chance to exit from script and I have to restart radio.

I used all latest lua and FW from frsky site.

Thanks in advance for any suggestion.

RudiKleinSA, 2023-08-30:
Ok, 

I think i found the right LUA SCRIPT at this link below for the **"NEW" FLYSKY ARCHER PLUS SR8**

https://www.frsky-rc.com/archer-plus-sr8/

[image]

[image]

I will test this later today and let you guys know.

RudiKleinSA, 2023-08-30:
Hi,

This is what happens when I run the **OPEN TX** LUA SCRIPTS for the **"NEW" FRSKY ARCHER PLUS SR8**

These are new Open TX scripts that do not work in Edge TX just by copying them into the scripts folder,

[image]
[image]
[image]
[image]
[image]
[image]

dbock300, 2023-08-30:
That's the same thing I'm getting

[image]<https://user-images.githubusercontent.com/93178003/264409888-77baa228-151d-4a59-b3d0-e38df6ad90d9.png>
[image]<https://user-images.githubusercontent.com/93178003/264410053-2d636c8a-0bc0-470f-9eb8-541c173df51f.png>
[image]<https://user-images.githubusercontent.com/93178003/264410239-438e1041-2c9d-4253-a134-c295bf651321.png>
[image]<https://user-images.githubusercontent.com/93178003/264410380-9b24ae7b-e37c-488b-a935-a6af42cc852e.png>
[image]<https://user-images.githubusercontent.com/93178003/264410535-1cff51d1-99f9-4097-9b5e-3c5f86c17e30.png>
[image]<https://user-images.githubusercontent.com/93178003/264410696-6b8bd818-045c-47ae-a179-e7a46b2092f9.png>

mick-74, 2023-08-30:
I had same issue because are missing icons, check lua image path.  But after that calibration lua doesn't  work

[image]<https://user-images.githubusercontent.com/93178003/264409888-77baa228-151d-4a59-b3d0-e38df6ad90d9.png>
[image]<https://user-images.githubusercontent.com/93178003/264410053-2d636c8a-0bc0-470f-9eb8-541c173df51f.png>
[image]<https://user-images.githubusercontent.com/93178003/264410239-438e1041-2c9d-4253-a134-c295bf651321.png>
[image]<https://user-images.githubusercontent.com/93178003/264410380-9b24ae7b-e37c-488b-a935-a6af42cc852e.png>
[image]<https://user-images.githubusercontent.com/93178003/264410535-1cff51d1-99f9-4097-9b5e-3c5f86c17e30.png>
[image]<https://user-images.githubusercontent.com/93178003/264410696-6b8bd818-045c-47ae-a179-e7a46b2092f9.png>

dbock300, 2023-08-30:
How to check lua script image path ?

[image]<https://user-images.githubusercontent.com/93178003/264409888-77baa228-151d-4a59-b3d0-e38df6ad90d9.png>
[image]<https://user-images.githubusercontent.com/93178003/264410053-2d636c8a-0bc0-470f-9eb8-541c173df51f.png>
[image]<https://user-images.githubusercontent.com/93178003/264410239-438e1041-2c9d-4253-a134-c295bf651321.png>
[image]<https://user-images.githubusercontent.com/93178003/264410380-9b24ae7b-e37c-488b-a935-a6af42cc852e.png>
[image]<https://user-images.githubusercontent.com/93178003/264410535-1cff51d1-99f9-4097-9b5e-3c5f86c17e30.png>
[image]<https://user-images.githubusercontent.com/93178003/264410696-6b8bd818-045c-47ae-a179-e7a46b2092f9.png>

RudiKleinSA, 2023-08-30:
Must this ISSUE  of not being able to calibrate the "NEW" FRSKY ARCHER PLUS SR8

be reported to EDGE TX 
https://github.com/EdgeTX

as it might not have anything to do with
[image]

Ralf-W, 2023-08-30:
You already have the missing images in your folder /SCRIPTS/TOOLS/img. 
But the new lua script is searching them in /SCRIPTS/TOOLS/RB_img. Thus copy /SCRIPTS/TOOLS/img to a new folder /SCRIPTS/TOOLS/RB_img - or rename /SCRIPTS/TOOLS/img to /SCRIPTS/TOOLS/RB_img. 
Sorry, I don't have the Archer hardware to do any tests myself.

dbock300, 2023-08-30:
I'll give it a try.

mFreundt, 2023-10-05:
Hallo,
I am trying to bind a Archer Plus GR6 to a Radiomaster Pocket with CC2500 Chipset.
The solutions described for the 4 in 1 does not seem to work.

I tried the currente Firmware V1.3.3.20, but none of the settings worked...
> Question: I learned the D16 = ACCST V2 -> Which should fit to the Compatibility Definition!?
Any hint?

Thanks,
Martin