pascallanger/DIY-Multiprotocol-TX-Module #819: Bluefly protocol

dc59, 2023-02-12:
Can it be possible to add Bluefly HP100 protocol ? this protocol is supported by Deviation.
https://www.deviationtx.com/wiki/supported_models
Thanks！

dc59, 2023-02-12:
My fault, move to Discussions.

pascallanger, 2023-10-30:
I can give it a try if you are still interrested.

pascallanger, 2023-11-09:
Please test the new BlueFly protocol available in v1.3.3.36 located here https://downloads.multi-module.org/latest-test/

dc59, 2023-11-11:
Thanks Pascal for your work.
I tried it with RM TX12 / EdgeTX 2.8.1 / external 4 in 1 multi-module, it can bind to Bluefly HP100BL but only few second, the red led go solid when I bind, and then all servo go upper position or bottom position and led gose off (it seems lost bind), then I can't do anything!
if you need video , please let me know.

dc59, 2023-11-11:
I have tried freq. tune and it dosen't work.

pascallanger, 2023-11-11:
Please test the version 1.3.3.37 same location which should be available in the next 5 minutes (auto build in progress)

dc59, 2023-11-11:
I can't download test build due to error message as below.

"
Error 404 – Not Found
Artifact 'multi-test-build' not found for run #6831147439.

Check on GitHub: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/actions/runs/6831147439#artifacts

[Return to the home page](https://nightly.link/)
"

dc59, 2023-11-11:
Hi Pascal,
I downloaded source code and compiled it with Arduino IDE get this file "multi-stm-1.3.3.37.bin" , flash this FW into 4in1 module, it get some result as 1.3.3.36!

pascallanger, 2023-11-12:
There is only one place in all the files where the version is written so if the scripts has generated the file "multi-stm-1.3.3.37.bin" then if you upload it correctly to the radio it will show 1.3.3.37 . It just can't display anything else!

dc59, 2023-11-12:
Hi Pascal,
Thank for uploading new 1.3.3.37 FW, but I'm not lucky, it worked as same as 1.3.3.36, after bind and servo move, then lost binding.

pascallanger, 2023-11-12:
Can you control the servos or not at all ?

dc59, 2023-11-12:
Not at all, after servo movement LED goes off (looks like lost bind) , I can't control anything.

dc59, 2023-11-17:
Hi Pascal,
https://youtu.be/0HKsCVIssqE
New 1.3.3.37 got same result.

dc59, 2023-11-18:
Hi Pascal,
I found file date inside the multi-test-build.zip is all Nov. 12 2023, could it be problem that FW file is not really been updated?

benlye, 2023-11-18:
@dc59 I think you saw the same thing as me, an email received last night (17/11) saying 'Please test the version 1.3.3.37 same location which should be available in the next 5 minutes (auto build in progress)' but this looks like Github doing something strange and sending an email for a message Pascal posted a week ago.

There is no new build since November 12th.

dc59, 2023-11-18:
Thanks benlye, I did receive the mail as you mentioned,that’s why I downloaded test build again for test and found the file date problem!