pascallanger/DIY-Multiprotocol-TX-Module #824: Not Enought Memory with Boxer

michael88997, 2023-02-23:
I am trying to run the DSM forward programming and getting the not enough memory error on startup with my boxer.. updated to the latest edgetx

frankiearzu, 2023-03-18:
Im aware of it.. creating a new version that will use much less memory, but those radios are very short on RAM.  I have a FrSky QX7 that im using for that.. no much RAM to play.
Will post when i have something.

frankiearzu, 2023-04-09:
Try the v0.54 MIN version.  is a limited version just to change existing settings, but cannot setup a brand new plane with it.
You can download the zip from here:
https://github.com/frankiearzu/DSMTools/tree/main/DSM%20Forward%20Programming/Black_White_128x64

closed 2023-10-30 by pascallanger