pascallanger/DIY-Multiprotocol-TX-Module #830: DSM Forward Programming not correctly setting AS3X/Safe Gain Channel

silicongarage, 2023-03-13:
EdgeTX v2.81
MM v1.3.3.20
Spektrum v2.38.5
Receivers:  AR637TA (unlocked),  AR8360T

Issue:  
When trying to set an AS3X/Safe Gain Channel the entered channel is changed to a different value when clicking Enter.
I am trying to set the Safe Gain to CH(9) and the AS3X Gain to CH(10).
The AR657TA resets both to CH(6)  and the AR8360T resets both to CH(5).
Both receivers use CH(5) for Flight Mode select.

frankiearzu, 2023-03-18:
Will take a look.   The RX is validating the values, so reverting them back to a valid channel.  Have you try using/moving the switch to set the value??

UltraFly, 2023-08-22:
I have a similar problem on a Spektrum AR10360T reciever using a TX16S Mk2 radio.
The AR10360T has 10 PWM channels and recieves a total of 12 channels
I can put the flight mode selection on channel 11 but I can't put stability gain on channel 12, I can only go up to ch9
This is not good because all channels 1-10 are used by PWM servos, so only channel 12 is available for this use.

frankiearzu, 2023-08-22:
The fix for the Gains is fixed in V,055..   Have to update the files here.

You can get it here:
https://github.com/frankiearzu/DSMTools

pascallanger, 2023-08-22:
@frankiearzu Will you do a pull request to update your scripts on this repo?

closed 2023-10-30 by pascallanger