pascallanger/DIY-Multiprotocol-TX-Module #832: Zorro- While selecting Protocol Type, Unable to scroll in reverse from Config.

KC748, 2023-03-15:
Greetings.
I initially posted the following issue with EdgeTx, but 3djc suggested the issue may be on the Multi side. So please have a look.

In EdgeTx v2.8.1, Model Setup, it is possible to scroll past the last protocol, (XK currently) to the beginning of the list, Config. However, from Config, it is not possible to scroll in reverse, back to the last protocol.

Over shooting the last protocol is quite easy and it's a little annoying having to scroll all the way back to the end of the list from the beginning.

[image]

Would it be possible to allow scrolling in reverse from Config to the end of the protocol list?
Or. a hard stop at the end of the list. A haptic and/or beep at the stop would be nice.

Thanks.

pascallanger, 2023-03-15:
Hummm, I'm pretty sure I'm doing a roll over on both sides. I'll check tomorrow on opentx to check the behavior there.

pascallanger, 2023-10-30:
Old EdgeTX issue.

closed 2023-10-30 by pascallanger