pascallanger/DIY-Multiprotocol-TX-Module #833: XK Module Status: Protocol invalid

marcoose, 2023-03-16:
I am using a Radiomaster TX16S and running V1.3.3.20 AETR.

Everything seems to be working fine for every single protocol (I tried them all!) _except_ XK.  For all other protocols I get the V1.3.3.20 AETR string (and "Binding" briefly, as appropriate).  But whenever I switch to XK the Module Status shows "Protocol Invalid".

I tried re-flashing the firmware, but get the same results.  Nothing turned up in my searches, and I'm not really sure where to start troubleshooting.

Thanks!

KC748, 2023-03-16:
The XK protocol invalid seems to be a known issue.
https://github.com/EdgeTX/edgetx/issues/3264

marcoose, 2023-03-16:
Doh - so much for my search abilities.  Thanks!

closed 2023-03-16 by marcoose