pascallanger/DIY-Multiprotocol-TX-Module #838: Jumper T16 Pro wont upgrade internal multi module.

MickJamie, 2023-03-19:
My kit is:
Jumper T16 PRO modified to V2.
VERS opentx-t16-2.3.5
Internal module status V1.3.3.7 AETR
I am trying to upgrade the internal module to mm-stm-serial-aetr-v1.3.3.20 (I have loaded the LUA scripts under TOOLS that are on same download link)
The upgrade is failing with a message 'NEEDS FILE stm-opentx-noinv' 
From what I can find this upgrade should work, so can anyone help please.

MickJamie, 2023-03-22:
Was an OpenTX issue. Changed to EdgeTX and works fine. Teach me to be a dinosaur.

closed 2023-03-22 by MickJamie

Andre-Gielkop, 2024-12-01:
Hello, not an OpenTX issue. Just had the same problem on a TX16S, with the last version of EDGETX, when trying to update internal module, receice same message "'NEEDS FILE stm-opentx-noinv". Anything I can do?