pascallanger/DIY-Multiprotocol-TX-Module #840: Jumper 4in1-SE Module "Fail to init. device" in FlashMulti Verbose

CarryFlagMyth, 2023-03-20:
Hi,
Module inherited from a Jumper T12. Tried several things already to have it _at least_ read by Flash Multi: purchased a new micro USB cable, upgraded OpenTx radios' firmware (Taranis Q7 & Jumper T12). Green light when plugging USB & Device manager showing _Silabs CP210x USB to UART Bridge_ to port COM8. Major problem here is as described in title: _Read/Write_ buttons in Flash Multi software won't go past the first step. I have seen you reply once COM ports could be busied by programs running in the background; yet I don't see why, having not ran a configurator program for ages since I'm stuck with this one problem for now.
Am I missing something here? Thank you

benlye, 2023-03-21:
Can you post the output from FlashMulti?  Screenshots including the detailed log window.

CarryFlagMyth, 2023-03-21:
Yes, if you want
[image][image]
[image]

CarryFlagMyth, 2023-03-21:
Thanks for your reply. I've downloaded zip file afresh and ended with a FlashMulti 0.6.1 folder with 24 files, 4 folders. Is this what it's supposed to be? Also, I was prompted I was launching a program that couldn't be checked by Windows SmartScreen. Could program's subroutines have been hindered at a point?
[image]

benlye, 2023-03-21:
You just need to make sure you extract all the files that are in the archive, which should include a small folder structure. 24 files in 4 folders looks about right.  It should look like this:

[image]

Importantly, you should have stm32flash.exe in the tools folder. Depending on your system, you may have to right-click on stm32flash.exe and check 'Unblock' on the security tab.

CarryFlagMyth, 2023-03-21:
Mine folder is just the same, save for the config file, reading _flash-multi.exe.config_; and it hasn't a small gear icon.
[image]
Running FlashMulti once unblocked from Windows command line gives a likewise result, safe for the _stm32flash.exe does not exist_ argument:
_stm32flash 0.4

http://stm32flash.googlecode.com/

Interface serial_w32: 115200 8E1
Failed to init device._

benlye, 2023-03-21:
Can you try moving the flash-multi-0.6.1 folder somewhere that doesn't have accented characters in the path?  E.g., `C:\Temp\flash-multi-0.6.1`.

I think `téléchargements` might be causing the problem (as per https://github.com/benlye/flash-multi/issues/30)

CarryFlagMyth, 2023-03-21:
I've moved the Flash Multi folder to hard drive root C:\ but exact same things occur.

CarryFlagMyth, 2023-03-21:
Actually no my mistake; it now reads:

.\tools\stm32flash.exe -r "C:\Users\andre\AppData\Local\Temp\tmp7729.tmp" -S 0x8010000:32 -b 115200 COM8

stm32flash 0.4
Failed to init device.

http://stm32flash.googlecode.com/

Interface serial_w32: 115200 8E1

CarryFlagMyth, 2023-03-21:
And no longer stm32 doesn't exist - as windows popup error screen meant the same failure I wouldn't notice

benlye, 2023-03-22:
OK, now we're at least trying to talk to the module.  'Failed to init device' means that the flashing program couldn't find an STM32 to talk to on the given COM port.  This can be for many reasons, from a bad USB cable to a faulty MULTI-Module.  All I can suggest is trying different USB cables or ports, or even a different computer (if you can).

CarryFlagMyth, 2023-03-22:
I suppose USB cables I use cannot be blamed as one is brand new; and both are showing the Multi Protocol device as CPx210 etc. properly in Device Manager. As from another computer, I have just tried, stm32 file needs to be unlocked, and past this point, program did run but won't init. device.

CarryFlagMyth, 2023-03-22:
As for the module to be defective; at least not as far as its behavior once inserted inside the radio _turned on_ is concerned. Therefore, have we exhausted every known causes of update malfunctioning _via_ Flash Multi? Beside unreliable computer hardware, what could prevent this module from initializing, according to you?

benlye, 2023-03-22:
In order for the flashing tool to interact with the module, the module must be in 'BOOT0' mode.  This is achieved by connecting the BOOT0 pin on the MCU to VCC.  On the JP4in1 modules this should happen when the USB cable is plugged in.  We have seen at least one module failure where it did not.  There's a header available on the PCB where you can manually bridge VCC and BOOT0.

That said, the fact that your module has only the green LED lit when it's plugged in suggests that this is probably not the problem you have - if the module was in 'normal' mode the red LED would be active.

That leaves some other hardware issue - a bad CP2102 chip, a bent or broken data pin in the USB connector, a bad solder joint between the USB connector and the PCB, or a bad connection elsewhere.

The PCB in this module doesn't serial header pins so the options for flashing it via serial are limited.  I assume you've already tried flashing it from the radio, which is why you are trying to use the USB port?

One last thing to check - when using the USB port, the module should be removed from the radio, or if it's in the radio, the radio must be turned off.

CarryFlagMyth, 2023-03-22:
You're right, I first aimed at upgrading my module through the SD card. This is what puzzles me most since I've been trying out this Flash Multi software: the **[SYNC ERROR]** message when running *.bin file from SD card is reputed to be caused by no bootloader present, or more precisely, that Flash Multi was never used to flash the device at least once. In my case, "Flash Module Bootloader" in the _Advanced>Actions_ menu is never available. In Flash Multi Documentation Index:  _This menu item_ (as well as Reset Module to DFU Mode) _is only enabled when a MULTI-Module in USB Serial mode is attached, and the COM port corresponding to the module is selected_...
Is this not the case for my module? When are these menu items allowed? My guess is Jumper 4in1-SE modules are one of a kind, and that other modules with USB plugs are enabling to run these functions when connected. More to this, I have read there's a difference between 1st and last Gen. Jumper 4in1-SE modules. Isn't there an interplay between the series that could account for the malfunctioning we're struggling with?
As for the ultimate check you mention, a couple of days ago when I started working this out, I used a bad USB cable, and ever since, I have taken greater care to whether the module was radio off; otherwise device manager would rarely if ever recognize the device as a Virtual COM Port like it should. With the Jumper T12 cable and the one I bought last week-end, Flash Multi always gets a COM port no matter if the radio is lit. But now I know that's the way it ought to be, and am attempting to read and write considering this.

CarryFlagMyth, 2023-03-29:
I have tried other diverse methods, one being flashing firmware from SD, with a custom booloader allowing to run apps; it felt promising but didn't work either (See @LapinFou [Opent Tx Wiki](http://opentx-doc.fr/)). I didn't bridge Boot0 & VCC pins, for I gather it's the same than plugging USB on JP4in1 devices. I have read a post on Github of someone eventually having updated his module with success by running Flash Multi from other computers. I keep that in mind.
So one remaining method is using Arduino Ide, and I'm not too comfortable with the idea; although I did purchase a USB to serial adapter. It has an easy 3.3/5 volts switch, so here is not my concern. Am I to solder pin headers on the back of my module to proceed with this method? Could you hint me on how to set up the hardware?
I can do soldering if needed, and I do have tiny jumpers for bridging pin headers; as well as I also have male/female wire extenders matching pin headers. But I cannot figure out how much of soldering - _if any_ - is required. My guess is Boot0 & VCC pins might be needing it, as for the linking with the adapter, I was wondering if a wire extension with pin headers at its end could suffice instead of soldering some more.
Perhaps you'll dismiss most of this considering it is the likewise of what I have attempted so far; but I thought I'd ask ;)

benlye, 2023-03-30:
Unfortunately the JP4IN1 module does not expose headers or pads for the serial TX and RX connections on the PCB - you would need to solder TX and RX onto the (very small) pins of the STM32 or CP2102 chip directly, and use the BOOT0 and 3V3 header holes to bridge BOOT0.  TX and RX are PA9 and PA10 on the STM32.

[image]

[image]

CarryFlagMyth, 2023-03-31:
I won't solder on the processor's legs for sure - I know it's sometimes suggested for drones with missing UART. But I did solder pin headers onto the GND 3,3v Boot0 pins; yet I'm not sure I'm making an effective short between the two latter. Is the green light the only signal then? Shall I short when plugging USB then release? Lastly, I wonder if the five MPM's plugs that connect it with the transmitter cannot be used with USB/Serial adapter? I believe I have seen some of them labelled as Rx/Tx; and beside, since the module's presumably updatable _via_ the transmitter, that implies a serial link could work the same... This is profane speech, but it seems odd given MPMs haven't their Tx/Rx made available for connecting.

benlye, 2023-04-01:
Green LED only indicates that the module is in BOOT0 mode.  Unfortunately the RX/TX pins that connect to the radio are not the same ones that can be used to flash the module when it is in BOOT0 mode - they can only be used when the bootloader is running (and you don't have the bootloader on your module).

CarryFlagMyth, 2023-04-07:
I recently used a *.frk file, and when launching it from regular radio mode (not _bootloader_), it offers to either flash S.Port, Internal module or External module. Have you ever heard of a MP module that could be flashed in this fashion; or is this just the External RF files of the radio that are updated this way, regardless of whether an external module is involved in the process?

horjab, 2023-08-06:
Hi everyone.
I have the same problem. 
Did someone find the solution please?
Thank you

pascallanger, 2023-08-06:
Which problem ? What are the LEDs showing up at boot ?

benlye, 2023-08-06:
@horjab no need to post twice. We can track your issue at https://github.com/benlye/flash-multi/issues/54.

horjab, 2023-08-06:
Here I found later the same issue I have, but no conclusion.

benlye, 2023-08-06:
We don't know if it's the same problem. You need to post the verbose logs. I don't care if we troubleshhot your problem here or on the Flash Multi repo, but we're only doing it in one place, so if you want to do it here then I'll close the other one.

horjab, 2023-08-06:
Thank you 
We will continue here.
Please,  how I  post the verbose logs?

benlye, 2023-08-06:
Tick the 'Show verbose output' box, try to read or write the module again, and post the verbose logs.

horjab, 2023-08-06:
Is this box in the multiflash?

benlye, 2023-08-06:
> Is this box in the multiflash?

Yes

horjab, 2023-08-06:
bingo 👍 .\tools\stm32flash.exe -r "C:\Users\h.bucovanu\AppData\Local\Temp\tmp8177.tmp" -S 0x8010000:32 -b 115200 COM5

stm32flash 0.4
Failed to init device.

http://stm32flash.googlecode.com/

Interface serial_w32: 115200 8E1

benlye, 2023-08-06:
So now you need to try all the same things - different USB cables, different PC (if possible).  If none of that works then you probably also have a faulty module and unfortunately there's nothing further we can do.

horjab, 2023-08-06:
Waall ready done cables and 2 dells.
Will soldering to the processor with the FTDI witch  I all ready use for un internal module T16S work ?

benlye, 2023-08-06:
> Waall ready done cables and 2 dells. Will soldering to the processor with the FTDI witch I all ready use for un internal module T16S work ?

It's possible.  You need TX, RX, 3V3, and GND.  You can use the empty holes on the PCB to bridge BOOT0 to 3V3 to get the chip into flashing mode.

horjab, 2023-08-06:
OK ça marche. I will try it. 
thank you for your support and readiness

pascallanger, 2023-08-06:
Have you tried to flash from the radio itself ?

horjab, 2023-08-06:
No because the version is 1.2.1.25 not even 1.2.1.51 my gues is that is  no bootloader inside 
but I will try this also 
Thank you

closed 2023-10-30 by pascallanger