pascallanger/DIY-Multiprotocol-TX-Module #841: No 64K option for Zorro CC2500 at downloads.multi-module.org/.

KC748, 2023-03-24:
TX12 has 64K firmware option at downloads.multi, but Zorro internal CC2500 does not.

A [member on RCG](https://www.rcgroups.com/forums/showthread.php?4027391-RadioMaster-ZORRO-is-here&postid=50729743) reported only the 64K version worked on his Zorro CC2500.

> I've succeeded binding the Zorro to the Rx.
…Then I upgraded to 1.3.3.20 and found that only the 64KB version did work.
>

benlye, 2023-03-24:
64K option is available for the Zorro now.  

[image]

I've asked RadioMaster to confirm if all Zorro CC2500 radios are 64KB (I suspect that they are).  Once they let me know I'll remove the non-64KB option for the Zorro CC2500.

closed 2023-03-24 by benlye

KC748, 2023-03-24:
That's terrific! Thank you.