pascallanger/DIY-Multiprotocol-TX-Module #842: Forward Programming error on AR8020T Rx

ror1948, 2023-03-25:
At main menu
get error Unknown_F8

also Rx Unknown_1B v2.37.4

also there is no menu items displayed.

ror1948, 2023-03-26:
Ok using  DSM FwdPrg_05_Color on AR8020T I also get unknown_F8 but it still seems to work
[error.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/11070738/error.zip)

ror1948, 2023-03-26:
better release zipfile layout
[RR_DSM_FP_053_Jan15.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/11070749/RR_DSM_FP_053_Jan15.zip)
just unzip to route of sdcard

ror1948, 2023-03-26:
[image]

frankiearzu, 2023-04-09:
@ror1948:   Thanks for the scren shots.

That RX don't have a Gyro, so i think the only settings are Servo Frame rates.
We only need to add the proper message and RX name to the code/file:

Version 0.53. 
Look towers the end of the file where the RX name and Text messages are set.

RXName[0x001B] = "AR8020T"

Text[0x00F8] = "Settings" ???  We can check on a Spektrum TX how that message ID translates to English.

If you start using version 0.54, the messages can be added to the file /SCRIPTS/TOOLS/DSMDATA/msg_fwdp_en.txt"
Will be like this

T|0x00F8| Settings
RX|0x001B| AR8020T

I will add them to my files, so that they will come standard in the next release.

Like your Idea of the Zip file... Just copy the entire thing to the SDCARD in one shot.. will do that in the future.

frankiearzu, 2023-04-09:
You can download v0.54 zip files from here:
[image]

mawzthefinn, 2023-04-23:
You should also be able to set Failsafe and trigger bind from Forward Programmming on the AR8020T

Also I see a couple gaps in that table based on the receiver lineup and the umbering pattern

RX | 0x0017 | AR6610T  
RX | 0x001D | AR10100T

Both should have the same capabilities as the AR8020T.

pascallanger, 2023-10-30:
I'm guessing things have been addressed by @frankiearzu