pascallanger/DIY-Multiprotocol-TX-Module #847: Can MPM configured to use a single S.Port line for half duplex communication?

richardclli, 2023-04-19:


richardclli, 2023-04-28:
@pascallanger Any ideas?

pascallanger, 2023-04-28:
Yes it can.
Some modules at the beginning where not correctly manufactured using a nand instead of a xor but other than that it's just code based.

richardclli, 2023-04-29:
> Yes it can. Some modules at the beginning where not correctly manufactured using a nand instead of a xor but other than that it's just code based.

So need code modification?  Can you point out which file start with?

pascallanger, 2023-04-29:
It's in the main and telemetry.

richardclli, 2023-04-30:
Currently the serial protocol uses USART2 for receiving and USART3 for transmitting (with receiving disabled).  So try to configure USART3 for receiving instead of USART2 can make this works, right?  You use interrupt to handle receiving?

pascallanger, 2023-10-30:
Sorry Richard, Do you still need help ?

richardclli, 2023-10-31:
Abandoned this thought, no need anymore.

closed 2023-10-31 by pascallanger