pascallanger/DIY-Multiprotocol-TX-Module #849: FRSky SxR Receiver setup running firmware v1.3.3.20

cpwilson, 2023-04-22:
Setting up a FRSky S8R today and having some issues when trying to go into SxR setup Lua.

When I enter the SxR lua to make changes to the SxR setup I get ESSI warnings, RSSI drops to almost nothing and telemetry to TX fails.  As such cannot setup SxR.     Data seems to start from SxR but then RSSI drops to almost nothing and no more data received from RX.     If I power cycle the RX no change but when I power cycle the TX connection comes back and is normal until I enter SxR lua again.    Using latest SxR lua v2.02

Issue Present under the following conditions:
Radio is Radiomaster TX16s running EdgeTX 2.8.2
RX binds fine, EULBT RX running S8R_LBT_20190531 firmware
MPM running firmware v1.3.3.20

Issue NOT Present under the following conditions:
Radio is Radiomaster TX16s running EdgeTX 2.8.2
RX binds fine, EULBT RX running S8R_LBT_20190531 firmware
MPM running firmware v1.3.3.14

So reverting to MPM firmware v1.3.3.14 sees the issue go away and able to setup SxR no problems.

frankiearzu, 2023-05-02:
Just FYI. Tested with EdgeTX 2.8.2 and 2.8.3,  MM firmware v1.3.3.20, and S6R Receiver, RX firmware 2.1.2 (2021-07-08).
Maybe your RX firmware is a bit old??

cpwilson, 2023-05-02:
I have upgraded a spare RX I have and agree it now works with MM firmware v1.3.3.20.

So whilst that's a fix/workaround it still does not explain why the older RX firmware has issues with the newer MM firmware v1.3.3.20.     As soon as I roll back the MM  firmware back to v1.3.3.14 all is well again.

frankiearzu, 2023-05-03:
I can't see any changes in the code for the FrSky protocols from version 1.3.3.14 to 1.3.3.20..   I can see changes almost 2 years ago, but that is way older than 1.3.3.14.  Not an expert in this protocol, but got my attention since i just recently setup a S6R.

pascallanger, 2023-05-03:
S6R and S8R are known for a few bugs like they do not support every RX numbers, they mess up the frequencies... The latest fw is fixing most of these bugs as far as I know.

cpwilson, 2023-05-03:
As I have found, when I upgraded MM firmware to v1.3.3.20 I lost the ability to access/ SxR settings when it perfectly previously on v 1.3.3.14.

pascallanger, 2023-05-03:
If the link is unstable, you won't be able to configure the s6r/s8r.
I don't feel responsible for frisky bugs...

3djc, 2023-06-13:
you need to be not too close to the SxR (frsky known swamping issue) and also ensure NO telemetry sensor is connected to SxR during setup

pascallanger, 2023-10-30:
No more communication, considering the issue solved by upgrading the RX.

closed 2023-10-30 by pascallanger