pascallanger/DIY-Multiprotocol-TX-Module #851: Bug: it's not working for WFly, bind WFR09S, only work several seconds then disconnected

ganggangdede, 2023-05-03:


ganggangdede, 2023-05-07:
Jumper T-Pro Internal Multi-Module, firewire: mm-stm-serial-aetr-v1.3.3.20.bin
TX firewire: opentx-T-Pro-2.3.15.bin
RX: WFly WFR09S
I tried other receivers such as FlySky Std and FrSky D8, it works well

ganggangdede, 2023-05-12:
> Jumper T-Pro Internal Multi-Module, firewire: mm-stm-serial-aetr-v1.3.3.20.bin
> TX firewire: opentx-T-Pro-2.3.15.bin
> RX: WFly WFR09S
> I tried other receivers such as FlySky Std and FrSky D8, it works well

> Which module type? Which firmware version on the module? What's the global ID of the module? What's the RX number? Have you tried different RX numbers?

Jumper T-Pro Internal Multi-Module, firewire: mm-stm-serial-aetr-v1.3.3.20.bin
TX firewire: opentx-T-Pro-2.3.15.bin
RX: WFly WFR09S
I tried other receivers such as FlySky Std and FrSky D8, it works well

ganggangdede, 2023-07-19:
> I suspect it could be a frequency issue, but the firmwire mm-stm-serial-aetr-v1.3.3.20.bin is disable the frequency tuning by default, can you please do me a fever to build the firewire with the frequency tuning enabled, so I can do a test

pascallanger, 2023-07-19:
Can you try first different RX numbers before playing around with freq tuning?

pascallanger, 2023-10-30:
No answer to my request for details, closing...

closed 2023-10-30 by pascallanger