pascallanger/DIY-Multiprotocol-TX-Module #853: Bug, FRSKYX EU protocol fails to send SPort data

MikeBland, 2023-05-10:
When using FRSKYX (V1) protocol set to use EU LBT, attempts to send SPort data (to configure receiver options) cause the control link to become poor and SPort data responses don't get returned. Works fine in FCC mode and FRSKYX2 (V2) also works OK.

MikeBland, 2023-05-13:
Fix is change line 103 in FrSkyX_cc2500.ino
from:
	FrSkyX_seq_sport(21,packet_length-2);	//21=RX|TXseq, 22=bytes count, 23..packet_length-2=data
to:
	FrSkyX_seq_sport(21,packet_length-(protocol==PROTO_FRSKYX && (FrSkyFormat & 2 )) ? 4 : 2);	//21=RX|TXseq, 22=bytes count, 23..packet_length-2=data

pascallanger, 2023-06-04:
Thanks, it will be in the latest release.

closed 2023-06-04 by pascallanger

MikeBland, 2023-06-19:
This change should only affect FrSkyX EU, not FrSkyX2, and is needed to make the MPM operate with FrSky receivers correctly.

TheIsotopes, 2023-06-19:
@MikeBland yes, that was the mistake.
Now it runs with your changes without problems.

I'll will post a PR with the changes soon.