pascallanger/DIY-Multiprotocol-TX-Module #855: Spektrum surface compatibly

RobertoD91, 2023-05-14:
I think that it is not clear from the documentation whether the multi module is compatible with Spektrum surface receivers:
- surface DSM ?
- surface DSMX ?
- surface DSM2 (work-in-progress ? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/769)
- surface DSMR (seems ok? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/534)

I have a Jumper T16 with EdgeTX 2.8.3, with internal multi module 1.3.3.20 and I can't bind with either a [SR3300T](https://www.spektrumrc.com/product/sr3300t-dsm-3ch-rx-w--telemetry-surface/SPMSR3300T.html) or [SR3001](https://www.spektrumrc.com/product/sr3001-dsm-3-channel-pro-series-receiver-surface/SPM1205.html) (both are DSM only and they work with a spektrum dx3s radio).

Is it possible to update the documentation clearly writing if the multi module is compatible only with Spektrum air or also with some Spektrum surface?

pascallanger, 2023-05-15:
I don't own any surface TX/RX so I can't say what IS NOT supported.
What you do see in the doc is what IS supported. So when you see DSMR appearing, it means it is supported: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-dsmr_1f---5

The problem is the Spektrum naming. They kept the same DSM2 or DSMX for surface and air but these are 2 different protocols.

RobertoD91, 2023-05-15:
Understand, thanks for your work!
I don't think I can open the radios and use a logic analyzer (at least for now), but I have a hackrf, can it be useful to acquire the signal between a spektrum surface tx and rx?

pascallanger, 2023-10-30:
I will close this issue since the only missing protocol is already tracked in another issue: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/769

closed 2023-10-30 by pascallanger