pascallanger/DIY-Multiprotocol-TX-Module #866: betaflight-tx-lua-scripts do not work with v1.3.3.27

TheIsotopes, 2023-06-17:
Since v1.3.3.27 the betaflight-tx-lua-scripts don't work anymore. The message "Waiting for API version" appears and the script must be terminated by pressing rtn for a long time. back to v1.3.3.25 and it works again.

@pascallanger pls can you check whats wrong.

pascallanger, 2023-06-17:
Have you try to rebind your RX? Have you tried to redo the frequency tuning ?

TheIsotopes, 2023-06-17:
I forgot to write that this affects FrSkyX2 EU. Here some screenshot:

[image]

[image]

When flying there's no problems and telemetry also works.

> Have you try to rebind your RX? Have you tried to redo the frequency tuning ?

@pascallanger I can try and get back to you later.

TheIsotopes, 2023-06-17:
@pascallanger Newly bound and frequency tuned, unfortunately the problem still exists.

TheIsotopes, 2023-06-18:
@pascallanger found the problem, it's https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/commit/cefe69a692f07d4621a2cd8a2ed972f4ace501d6. if I undo this, betaflight-tx-lua-scripts works again.

MRC3742, 2023-06-20:
This problem also effects using any of the UNI-RX scripts.
All telemetry is working but will not populate any fields in the scripts.
MPM version 1.3.3.27 and Protocol is FrSkyX2 D16 FCC

The linked PR listed above cures the problem with these scripts also.

Rich

Talha909, 2023-07-05:
what is the link of this MPM version 1.3.3.27. i cant able to found this.

closed 2023-08-29 by TheIsotopes