pascallanger/DIY-Multiprotocol-TX-Module #868: ACT S3D support planned?

onki69, 2023-06-28:
Hi,

Is there a plan to implement the ACT S3D protocol.
AFAIK ACT S3D is quite similar to MPX MLink (uses MSB for telemetry) and also the Cypress chip.
Since we have a lot of former ACT users in our club I can try to get test hardware.

twinking92, 2024-08-25:
Hi, are there any news regarsing this? I also would be interested in implementing the Act protocol. I have receivers ans also Transmitter Modules. Where would i need to send them? Alternative I could try to do some work on my own. At least working with arduino and doing some programming is no Problem. But I would need some starting advices on how to Start.

twinking92, 2025-01-12:
> If you can get a TX and RX sent to me I can have a look.

Hi Pascal, 
How can I contact you? I wouldime to provided you the Transmitter Module and receiver. Would be realy happy it you could have a look at this protocol. Best regards
Fabian

Dietrich-L, 2025-06-15:
Anything new here? 
I happen to own some of these receivers and wuold gladly reuse them
Dietrich