pascallanger/DIY-Multiprotocol-TX-Module #870: Unable to select ASSAN protocol

jims1234, 2023-07-30:
I just recieved a new Radiomaster 4in1 Tx module with  V1.3.3.7 AETR installed in it so have not modified it in any way .. I'm using it in a new Radiomaster TX16S MK2 ELRS radio.  I can easily bind FrSky and RM D16 and D8 recievers in it and the 1st time I went to bind an ASSAN Rx it worked just fine,  but now I can't see the ASSAN protocol in the list any more to select it.. my supported protocol list is in alphabetical order and always starts with BAYANG not ASSAN, so I can't select it any longer.. I've looked thru all the options and ASSAN has disappeared.  Is this a s/w bug? If that is the case, I am considering installing an older version of the Milti Firmware into this Tx module to try to fixit.. If I do that tho, will I need to re-bind all my planes again?  Is this behavior repeatable in other types of 4in1 modules?

pascallanger, 2023-07-30:
It sounds more like an edgetx issue. You should contact them

jims1234, 2023-07-31:
Will do.. and Radiomaster Suppo.o. I have tried flashing the module with V1.3.3.7 and V1.3.3.20 and it's behaving the same with both..

pfeerick, 2023-10-19:
I believe this was a software bug in some of the earlier EdgeTX 2.8.x versions that were subsequently fixed in 2.8.5 (with the fix also present in 2.9.0 onwards), so this issue should be resolved now.

closed 2023-10-20 by pascallanger