pascallanger/DIY-Multiprotocol-TX-Module #872: Aux3 and Aux4 cannels do not work with IRX4 module on a Jr 9503 radio

rajaskhan, 2023-08-05:
I converted an old JR X9503 2.4 ghz radio with an IRX4 plus multimodule in PPM mode.  I can get channels 1-7 outputs driving servos (THRO to AUX2) on a JR R921 receiver but cannot get servos to work on channels 8-9 (AUX3 and AUX4).  The module is set to dial position '7' for DSM2 protocol.  
Interestingly, the module binds to FRSKYX protocol and my X8R receivers with all 8 channels (Dial set to position '5')

pascallanger, 2023-08-05:
How many channels have you set on the dsm2 protocol entry ?

rajaskhan, 2023-08-05:
Hello Pascal,

Thanks for the prompt response.  I am using the module in PPM mode and not
Serial ( the radio is the JR9503 and does not support Serial protocol) and
the dial is switched to Position 7 for DSM2. The radio binds ok and
everything works except channels 7-8 for Aux3 and Aux4.  All channels are
working as per the channel monitor on radio screen.

pascallanger, 2023-08-05:
I understand that you are using it in ppm mode. But you need to tell the module how many channels you want to send in DSM since some receivers are not happy if you send too many.
What I'm guessing is that you are using the module out of the box. In ppm mode, you MUST compile yourself the code to tell the module how to behave. Part of it is to configure which type of ppm radio you have, if you want banks and what protocol and associated parameters are assigned to each dial position.
So unless you compile the code yourself, you will continue to face this problem.
Bottom line: this is not a bug, the module just does what it has been configured for.

rajaskhan, 2023-08-05:
Pascal,

The module is an IRangeX from AliExpress and straight out of the box.  I
agree with your analysis and will need to learn how to compile the code. I
am familiar with basic Arduino coding but the IRangeX module that I have is
STM32 so will the Arduino programming GUI work?

Could you link me to a page on GitHub which allows me to read up, educate
myself, and compile a version of code with the number of
channels/assignments that I need?

Thank you again for your response.

Regards,

Raja

pascallanger, 2023-08-05:
Everything is explained on the main page of this github repo but here is the direct link : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md

closed 2023-08-05 by pascallanger