pascallanger/DIY-Multiprotocol-TX-Module #873: No output on Flysky FS-A8S Receiver

bishko, 2023-08-06:
Hi Guys, 

I'm having some trouble setting up the Flysky FS-A8S with my Radiomaster TX16S. 

When selecting the FlySky2A protocol and the "PPM,SBUS" from the sub-menu, the receiver binds, but does not produce any SBUS output. As per the manual I have ensured that the receiver is in SBUS mode via the blinks. I have tired to output both 8 and 16 channels from the TX16S in all other sub-menu options, but it still does not seem to work. I have also tried to run the receiver in IBUS mode as well and to then bind to the TX16S and none of the option produce an output for some reason. 

Could someone please let me know what I might be dowing wrong or just let me know what the correct way is to set this receiver up?

Thank you very much in advance! :)

geovinny1, 2023-08-22:
I have also observed a problem with Flysky and SBUS. My test as follows:
Radiomaster TX16S MkII, EdgeTx 2.7.1, Multimodule V1.3.3.20 AETR, Protocol FlSky2A PWM,SBUS
Flysky FS-iA6B receiver feeding into an Arduino for analysis.
The receiver outputs IBUS not SBUS in all cases (including other sub-protocols). I know that because I am decoding the output and it is IBUS. I removed all sensors (input telemetry) to ensure that wasn't confusing things. I rebound and cold booted.
In order to test MYSELF, I can decode a Frysky S8R receiver and it is SBUS. This originally cropped up because a commercial 16ch SBUS decoder as seen on aliexpress would not work with the FS-iA6B.

geovinny1, 2023-08-22:
I should also say it's not clear if the OP needs an inverter with FS-A8S and SBUS. I have inverters as necessary. (you don't get anything decipherable if the inversion is wrong)

pascallanger, 2023-08-22:
I don't have access to a genuine Flysky TX so it's going to be hard to troubleshoot... Do you know if this is supported by this receiver (at least on the manual)? Do a genuine Flysky TX supports the switch from IBUS to SBUS?
I think this was implemented for a 3rd party firmware.

geovinny1, 2023-08-22:
I have no documentation that the FS-iA6B (receiver) supports SBUS. With the OP's FS-A8S i-BUS/S.BUS is written on the case (I don't have an FS-A8S).
I know that the FS-iA6B itself works with sub-protocols marked "SBUS", but SBUS doesn't come out of it. I know that PWM/PPM options work OK on the CH1 connector.
I thought IBUS or SBUS was Tx-generated, I don't believe the receiver is doing some sort of conversion. But I have no means to check the protocol until it comes out of the receiver. For that matter, I can't check rx firmware version except that it's "new". It must be relatively easy to see if a TX16S MkII is transmitting SBUS protocol when SBUS is selected?

For my purposes I can live with iBUS. But the OP and others may surely have problems (e.g. my 16ch SBUS decoder is not useable with FS-iA6B, and I'd be nervous to buy an FS-A8S for SBUS)
Cheers

pascallanger, 2023-08-22:
"I thought IBUS or SBUS was Tx-generated," No, the RX is dealing with that. This is just a config flag sent to the RX to switch from one to the other same as PWM/PPM.

geovinny1, 2023-08-22:
Thank you for clearing that up. Both the OP and my receiver may be Rx firmware issues. Cheers

closed 2023-10-30 by pascallanger