pascallanger/DIY-Multiprotocol-TX-Module #875: Turbo Racing protocol

g2gcrazyy, 2023-08-18:
Would it be possible to add turbo racing to the multimodule? I can provide a turboracing rc car and transmitter if needed.

pascallanger, 2023-10-30:
I've already tried in the past without luck. The frequencies used seem completely random... But I might give it another try with the MT12 getting out.

fiechr, 2024-01-29:
If this is about these:
[image]

I have two of those including receivers, one labeled like the one in the picture and one labeled "Carson Reflex Wheel Start".
Don't really need them anymore. But I assume hardware access isn't really the issue?

Freddyb888, 2024-03-10:
I hope one day I can bind my turbo racing cars with my radiomaster mt12 4in1. Same brand, they shouldale it compatible.

pascallanger, 2024-03-10:
Don't put too much hope into it. This protocol is quite complex and using a RF chip that I've never seen before with all the features enabled like scramble and FEC.