pascallanger/DIY-Multiprotocol-TX-Module #880: Horus X12S with Radiomaster RM4in1

jimmy6616, 2023-09-02:
Running edge 2.8.3 on an X12S which i purchased second hand to upgrade from my aging Jumper

I purchased a Radiomaster RM4in1 Multi to use with my Frsky , Jumper and SFHSS Rx's

The module works fine im my Jumper, binds, shows RSSI and all telemetry.

In the X12S, different story. Seems to bind okay, No RSSI or any telemetry and the staus in info says 'No MULTI_TELEMETRY detected'
Maybe a Edge config issue or script on the X12 missing. When i connect to a Frsky X6R using the internal module everything is fine

Any help would be appreciated

pascallanger, 2023-09-02:
Solution in issue https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/621

closed 2023-09-02 by pascallanger

jimmy6616, 2023-09-02:
Try these photos if it help please
[image]
[image]

pascallanger, 2023-09-04:
Sorry these pictures are too fuzzy. Take sharp pictures so I can see the traces.

jimmy6616, 2023-09-04:
Are these better? Jim
[image]
[image]

Merga189, 2024-11-13:
Same issue here. RADIOMASTER RM 4IN1 NOT working with FrSky X10 S Express. But as Pascal closed this one here, I'll shift over to #621