pascallanger/DIY-Multiprotocol-TX-Module #885: can't bind pelikan scx24

bd4, 2023-09-11:
I am trying to bind Radiomaster Boxer with internal Multi module to an SCX-24 from Horizon Hobby, serial AXI00001T1. It has AE-6 ESC/Receiver combo. I have tried in low power mode and normal mode, close and far away, no luck. I start binding on the transmitter before turning on the scx24, which has a read light on the AE-6 that blinks three times than stays off, while the [Bnd] indicator in edgetx continues to bilnk.

Did something change with these models, so i.e. older scx-24 were using a different receiver with different protocol? Note that the transmitter that came with it does work, so the car and receiver are functioning normally. I did see that there is a V2 out now, I think from 2022, and what I have is a V1. The V2 has a blue ESC/receiver combo, HRZ0015.

bd4, 2023-09-11:
It's worth noting, that in the receiver manual, it says the Frequency is "2.4GHz FHSS". That is the only clue regarding protocol.

bd4, 2023-09-11:
Also the transmitter that came with it is an AX-4, which is listed as supported in the Protocol_details doc. So not sure what I am doing wrong here, or of something changed.

pascallanger, 2023-09-11:
Most if not all the time, bind works like this:
1. Power on the RX in bind mode. If there is a button then press it. If the RX has no button and do not need to be bound each time then you need to wait for the RX to go in bind mode after a timeout.  If the RX has no button and needs to be bound each time then you should send a bind not to long after power on.
2. Put the radio in bind mode

pascallanger, 2023-09-11:
Obviously you need to select the right protocol and subptotocol.

bd4, 2023-09-11:
Thanks for the reply! There is no bind button or jumper on the AE-6, the instructions say that it should immediately enter bind mode. I did try leaving it on for 5+ minutes, and it did not change to a different mode or succeed in binding. I did select Pelikan and SCX24 protocol, but I am not sure if this model of SCX24 uses the same protocol as what was used to create the original SCX24 support. It would be helpful to know what receiver was used for that. But the docs do say transmitter AX-4, which is the same as what I have, which would suggest it was the same protocol. So not sure what the problem is here.

pascallanger, 2023-09-11:
Can you try to increase the RX number by 1 and try again?

pascallanger, 2023-09-11:
Search the closed issues for scx24, may be you will find the RX type used at that time.
Do you have a logic analyzer to dump the a7105 SPI bus? Can you confirm the TX has an a7105 inside near the antenna?

bd4, 2023-09-11:
Tried several RX numbers, up to 06, no luck so far.

I don't have a logic analyzer currently, would one of the inexpensive usb ones from amazon work?

I'll see if I can open up the TX.

pascallanger, 2023-09-11:
Yes $10 saleale compatible 8 channels can do the trick. You'll have to solder small wires around the a7105.

bd4, 2023-09-11:
It does have an a7105. I am not good at soldering yet, but motivated and have an iron and practice board to use before trying the real thing. This is what it looks like: https://bda.space/images/ax-4-circuitboard.jpg. The antenna is on the right, the a7105 is on the left. There are some things that look like pads on the left, but not sure if/how are connected to a7105. Ordered an analyzer, should arrive in a day or two.

I want to use this with a flight type edgetx transmitter, to train pilots on the ground re basic controls and orientation before advancing to planes.

bd4, 2023-09-13:
Logic analyzer arrived. Any links to documentation on what pins to connect and how to dump SPI bus on the a7105?

bd4, 2023-09-14:
I found the rcgroups thread about adding SCX-24 support, in particular:
https://www.rcgroups.com/forums/showpost.php?p=46962099&postcount=21959

I have the C-10, so not sure why it is not working. I know that sometimes models have revisions and radios change, but often there is noise about that on the internet, e.g. the Smasher V2.

Here is the picture from rcgroups of the transmitter the user used to reverse, with your annotation: https://static.rcgroups.net/forums/attachments/1/1/5/4/3/7/a14854331-93-capture.png. Mine does look different. Mine does have 7 pads that would be easier to solder, but it is hard to see if/how they are connected to the ar7105. I think I will need a lot of practice before I am ready to solder directly to the tiny chip pins... I do have a microscope though :P.

pascallanger, 2023-09-14:
We are interested by 5 wires in total: GND (can be taken where you want Batt- if you want) and the 4 others noted on the picture below.
You don't have to solder on the A7105 pins which are too small. What I would do is to use the 4 tracks which goes between the A7105 and ESP as I highlighted on the picture, remove the green using a cutter to access the copper then solder your wire in place. There is no way to do shorts or break something if you use this method. Once your happy use a tape to secure all the wires so one won't get loose with time.

[image]

bd4, 2023-09-14:
Thanks that all sounds doable! I am going to go through an elenco solder practice board this weekend, then give the transmitter soldering a go.

pascallanger, 2023-09-14:
Also try to use the software Logic 1.2.14 for the captures. Latest versions have a bug with SPI decoding that they've never corrected...

bd4, 2023-10-30:
Got distracted with other projects. I am putting a new ESC and RX in my SCX24, which makes the stakes for soldering the stock TX lower, still planning on attempting, hopefully this weekend.

pascallanger, 2023-10-30:
You can send me the TX and RX then, I will add the proto for others.

bd4, 2023-10-31:
> You can send me the TX and RX then, I will add the proto for others.

ok I actually have three surface tx/rx that I pulled out, this v1 scx24, vanquish vt/vr-1, and an hbx 16889a pro (esc/rx combo). I think the HBX may be wltoys proto and already supported, but didn't try, wanted to use with a radiolink surface tx so needed to swap anyway. Could send all at once.

pascallanger, 2023-10-31:
If you are willing to send me everything at once, contact me by email pascal _ langer @ yahoo . fr (remove the spaces).

bd4, 2023-11-04:
Related to #900 , closing this.

closed 2023-11-04 by bd4