pascallanger/DIY-Multiprotocol-TX-Module #888: MPM issue with Edgetx 2.9 in Horus x10s Express

Barstock, 2023-09-19:
No issues operating MPM in Horus x10s Express with either Opentx or Edgetx 2.8.5 and earlier editions.
Updated to 2.9 and experienced the following:
1 Unable to maintain a link with Flysky IA10B
The system would link for 3 to 5 seconds and then ‘shut down’ the MPM
2 Unable to complete a new bind to the same receiver or Hitec Optima
3 Was able to maintain a link with a Hitec Optima and Graupner 24
Nb The internal module was off during those operations.
MPM IRangex 4 in 1 Module V1.3.3.20 AETR
Horus x10 s express ISRM-S-X10E Status 1.1.0/ 2.1.9 FCC

pascallanger, 2023-09-19:
Places open the case with edgetx

pfeerick, 2023-10-19:
Cross linking to EdgeTX issue : https://github.com/EdgeTX/edgetx/issues/4069

It seems to have been a gremlin that the OP is not unable to reproduce any more, nor have we been able to reproduce on our side.

closed 2023-10-20 by pascallanger