pascallanger/DIY-Multiprotocol-TX-Module #891: V761 Protocol - Battery Telemetry / Low Voltage Alarm - Feature Request

Narb1cus, 2023-09-24:
Hi,

Was wondering if it is possible to add battery telemetry to the V761 protocol for use with Open TX ?

The stock TX features a low voltage alarm but this is not currently supported in the 4in1 firmware. 

Thanks

pascallanger, 2023-09-24:
Which planes have telemetry?  For sure not the old models when I've reversed the protocol....
Is there a way to identify them from outside?

Narb1cus, 2023-09-24:
Hi, Thnaks for the reply. 

I'm really not sure tbh. Afaik they all support the low voltage alarm with the stock TX from the reviews i've watched. 

I know that the Volantex Spitfire and BF109 definitely do as my friend has them both and bought them within the last year. He's just bought a Jumper tlite with 4in1 and they wrok great with the latest frimware but low voltage alarm is the one feature he's missing.  

They both have this version of the flight board installed - https://uk.banggood.com/VolantexRC-Mini-Mustang-P-51D-or-Mini-F4U-or-Mini-P47-RC-Airplane-Fixed-Wing-Spare-Part-2_4GHz-4CH-Receiver-Board-With-On-Board-Servos-p-1997572.html?cur_warehouse=CN&rmmds=search

pascallanger, 2023-09-25:
I've posted on RCG. I know one of my friend purchased a "zero" not so long ago. May be it has this feature, I'll ask him to check the manual.

pascallanger, 2023-10-30:
Telemetry added in v1.3.3.33

closed 2023-10-30 by pascallanger