pascallanger/DIY-Multiprotocol-TX-Module #892: MPM update impossible: "not a valid firmware" message

Billy60, 2023-09-27:
Hello
I downloaded the proper MPM fw file (TX16S, Serial, version 1.3.3.20, I am running 1.3.3.16) and have stored it in the Firmware file of the Sdcard ok.
When trying to 'write firmware' I get the following error message: not a valid firmware.
What does it mean and what is the solution ?
Thanks !

pfeerick, 2023-10-19:
How exactly are you trying to update? Are you going into the SD browser (SYS/Radio Settings => SD Card), selecting the file, and choosing to flash the internal or external module? You *do not* update the MPM via the OpenTX/EdgeTX bootloader. 

If you are trying to update it via the SD card browser, I would try deleting the file on your SD card, re-downloading it from https://downloads.multi-module.org/, and making sure the file you copy to your SD card is approximately 115KB in size. Make sure you safely eject the card, and try updating it again.

pascallanger, 2023-10-30:
No update since a month, considering the issue solved.

closed 2023-10-30 by pascallanger