pascallanger/DIY-Multiprotocol-TX-Module #897: MultiLuaScripts.zip is missing the DSMLIB subfolder and files

alm-nl, 2023-10-28:
Just downloaded the v1.3.3.33 files from https://downloads.multi-module.org/ and I noticed the DSMLIB subfolder is missing in the zip-file, while these files are required when you want to use the DSM Forward Programming scripts.
Can these files be included in the zip-file as well? It would make it easier for users to use this functionality.

Thank you.

pascallanger, 2023-10-30:
@alm-nl Thanks for the notification
@benlye Hi Ben, Is this something you could look into? Thanks, Pascal

benlye, 2023-10-30:
I'm on it. Will have a fix for the action shortly. 

How should we fix the current release? Should I just replace the LUA script zip file with the new one? It's either that or a new release, which seems unnecessary.

pascallanger, 2023-10-30:
I would just replace the file in the current release.

benlye, 2023-10-30:
I've updated the zip file on the latest release.

```
C:\Temp>"C:\Program Files\7-Zip\7z.exe" l MultiLuaScripts.zip

7-Zip 22.01 (x64) : Copyright (c) 1999-2022 Igor Pavlov : 2022-07-15

Scanning the drive for archives:
1 file, 204445 bytes (200 KiB)

Listing archive: MultiLuaScripts.zip

--
Path = MultiLuaScripts.zip
Type = zip
Physical Size = 204445

   Date      Time    Attr         Size   Compressed  Name
------------------- ----- ------------ ------------  ------------------------
2023-10-30 10:58:55 .....        29338         7631  SCRIPTS\TOOLS\DSM FwdPrg_05_Color.lua
2023-10-30 10:58:55 .....        32369         8896  SCRIPTS\TOOLS\DSM FwdPrg_05_MIN.lua
2023-10-30 10:58:55 D....            0            0  SCRIPTS\TOOLS\DSMLIB
2023-10-30 10:58:55 .....        27360         7194  SCRIPTS\TOOLS\DSMLIB\DsmMenuLib.lua
2023-10-30 10:58:55 D....            0            0  SCRIPTS\TOOLS\DSMLIB\img
2023-10-30 10:58:55 .....         2735         2669  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_21.png
2023-10-30 10:58:55 .....          817          642  SCRIPTS\TOOLS\DSMLIB\img\wt_2ail_2flp.png
2023-10-30 10:58:55 .....          880          806  SCRIPTS\TOOLS\DSMLIB\img\h_rx_pos_6.png
2023-10-30 10:58:55 .....         1361         1361  SCRIPTS\TOOLS\DSMLIB\img\h_swp_3_120.png
2023-10-30 10:58:55 .....          862          776  SCRIPTS\TOOLS\DSMLIB\img\h_rx_pos_2.png
2023-10-30 10:58:55 .....         1210         1056  SCRIPTS\TOOLS\DSMLIB\img\tt_taileron.png
2023-10-30 10:58:55 .....         2776         2763  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_16.png
2023-10-30 10:58:55 .....         2992         2936  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_3.png
2023-10-30 10:58:55 .....         1408         1313  SCRIPTS\TOOLS\DSMLIB\img\h_rx_pos_1.png
2023-10-30 10:58:55 .....         2805         2738  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_10.png
2023-10-30 10:58:55 .....         1361         1361  SCRIPTS\TOOLS\DSMLIB\img\h_swp_3_135inv.png
2023-10-30 10:58:55 .....         1272         1272  SCRIPTS\TOOLS\DSMLIB\img\h_swp_3_90inv.png
2023-10-30 10:58:55 .....          721          544  SCRIPTS\TOOLS\DSMLIB\img\wt_1ail.png
2023-10-30 10:58:55 .....          679          498  SCRIPTS\TOOLS\DSMLIB\img\wt_flaperon.png
2023-10-30 10:58:55 .....         2811         2770  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_14.png
2023-10-30 10:58:55 .....         1210         1056  SCRIPTS\TOOLS\DSMLIB\img\tt_traileron.png
2023-10-30 10:58:55 .....         2743         2670  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_23.png
2023-10-30 10:58:55 .....          679          496  SCRIPTS\TOOLS\DSMLIB\img\wt_2ail.png
2023-10-30 10:58:55 .....         2540         2483  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_4.png
2023-10-30 10:58:55 .....         1213         1116  SCRIPTS\TOOLS\DSMLIB\img\h_rx_pos_7.png
2023-10-30 10:58:55 .....         2995         2962  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_7.png
2023-10-30 10:58:55 .....         2548         2501  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_6.png
2023-10-30 10:58:55 .....         2834         2777  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_9.png
2023-10-30 10:58:55 .....         1263         1220  SCRIPTS\TOOLS\DSMLIB\img\tt_2rud_1ele.png
2023-10-30 10:58:55 .....         2721         2669  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_24.png
2023-10-30 10:58:55 .....         2698         2630  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_22.png
2023-10-30 10:58:55 .....         1293         1227  SCRIPTS\TOOLS\DSMLIB\img\tt_vtail.png
2023-10-30 10:58:55 .....         1162         1080  SCRIPTS\TOOLS\DSMLIB\img\h_rx_pos_3.png
2023-10-30 10:58:55 .....         2696         2640  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_25.png
2023-10-30 10:58:55 .....         2547         2480  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_8.png
2023-10-30 10:58:55 .....         1239         1075  SCRIPTS\TOOLS\DSMLIB\img\tt_2rud_2ele.png
2023-10-30 10:58:55 .....          866          785  SCRIPTS\TOOLS\DSMLIB\img\h_rx_pos_4.png
2023-10-30 10:58:55 .....         1055         1029  SCRIPTS\TOOLS\DSMLIB\img\wt_elevon.png
2023-10-30 10:58:55 .....         2619         2544  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_18.png
2023-10-30 10:58:55 .....         1254         1254  SCRIPTS\TOOLS\DSMLIB\img\h_swp_norm.png
2023-10-30 10:58:55 .....         2622         2557  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_19.png
2023-10-30 10:58:55 .....         2536         2502  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_2.png
2023-10-30 10:58:55 .....         1374         1374  SCRIPTS\TOOLS\DSMLIB\img\h_swp_3_120inv.png
2023-10-30 10:58:55 .....          728          600  SCRIPTS\TOOLS\DSMLIB\img\tt_1rud.png
2023-10-30 10:58:55 .....         2781         2762  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_12.png
2023-10-30 10:58:55 .....         2725         2693  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_1.png
2023-10-30 10:58:55 .....         2575         2494  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_5.png
2023-10-30 10:58:55 .....         2629         2566  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_17.png
2023-10-30 10:58:55 .....         1332         1332  SCRIPTS\TOOLS\DSMLIB\img\h_swp_3_140.png
2023-10-30 10:58:55 .....         1341         1341  SCRIPTS\TOOLS\DSMLIB\img\h_swp_3_140inv.png
2023-10-30 10:58:55 .....         1464         1381  SCRIPTS\TOOLS\DSMLIB\img\h_rx_pos_5.png
[... 6 lines cut ...]
2023-10-30 10:58:55 .....          755          582  SCRIPTS\TOOLS\DSMLIB\img\wt_2ail_1flp.png
2023-10-30 10:58:55 .....         1229         1187  SCRIPTS\TOOLS\DSMLIB\img\tt_1rud_1ele.png
2023-10-30 10:58:55 .....         2674         2623  SCRIPTS\TOOLS\DSMLIB\img\rx_pos_20.png
[... cut ...]
2023-10-30 10:58:55 .....        10376         2759  SCRIPTS\TOOLS\MultiChan.txt
2023-10-30 10:58:55 .....        10230         2804  SCRIPTS\TOOLS\MultiChannelsUpdater.lua
2023-10-30 10:58:55 .....        16905         3798  SCRIPTS\TOOLS\MultiConfig.lua
2023-10-30 10:58:55 .....         6369         1935  SCRIPTS\TOOLS\MultiLOLI.lua
2023-10-30 10:58:55 .....         3481         1459  SCRIPTS\TOOLS\pidDsm.lua
------------------- ----- ------------ ------------  ------------------------
2023-10-30 10:58:55             491294       188411  77 files, 2 folders
```

pascallanger, 2023-10-30:
Perfect thanks !

closed 2023-10-30 by pascallanger