pascallanger/DIY-Multiprotocol-TX-Module #900: can't bind Kyosho KT-531P and Mini-Z AWD FHSS

bd4, 2023-11-02:
I hold bind button down while turning the Mini-Z on as described in manual to enter bind mode. Configured model in EdgeTX to use MPM, Kyosho sub FHSS. Click bind. Nothing happens. Have tried different rx number and low power just to see, so far nothing works. Is there anything else I can try?

This TX is listed as supported in the docs, but maybe there are multiple subprotocols it supports. This TX has serial KT531P242181 and lists 202305 as the date.

pascallanger, 2023-11-02:
Strange none of the surface protocols seem to work for you... I don't own any of these models but I'm sure they were tested.

bd4, 2023-11-02:
The RadioLink Surface protocol is working for me. I wonder if there was some change in common code that affected/broke multiple surface protocols? Or it could be related to manufactures making changes to electronics without clearly changing model numbers, so the ones I have are different from what was tested.

pascallanger, 2023-11-02:
Strangely all the issues you report are using the A7105 RF chip... Which radio do you have ? Is there an A7105 protocol working for you? RadioLink uses the CC2500 RF chip.

pascallanger, 2023-11-02:
Are you still willing to send me the TXs and RXs you have? I could had more IDs so more people can play together with that hw.

pascallanger, 2023-11-02:
Spektrum = CYRF6936, V761 = NRF24L01
For example, A7105 = Flysky which is well known : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#a7105-rf-module

pascallanger, 2023-11-02:
> Definitely happy to send TX/RXs I am no longer using, and you can keep for continued testing.

Please contact me on my personal email pascal _ langer @ yahoo . fr (without spaces)

bd4, 2023-11-02:
Email sent, let me know if you get it. Maybe yahoo did not like my custom domain, will try sending via gmail also.

bd4, 2023-11-04:
Ok I just successfully bound my Mini-Z with Zorro 4-in-1 using Kyosho sub FHSS. It has v1.3.3.7 AETR, while my Boxer has v1.3.3.20 AETR. So not sure yet if this is hardware related failure with A7105 on Boxer or a regression in newer firmware. I think I will try upgrading Zorro first, as that is less used and I don't want to break any working models by downgrading the Boxer.

Will also try SCX-24, I have not swapped out RX on it yet.

bd4, 2023-11-04:
SCX-24 also binds with Zorro Internal 4-in-1 on 1.3.3.7

bd4, 2023-11-04:
Another interesting piece of information I had forgotten about: I briefly had a taer mpm firmware flashed on Boxer, because when I first got it I was trying to use a taer Nano S3 heli file, before I became convinced aetr is way more common and it would be less work to stick with it. It is possible I messed something up with firmware, but weird it would fail in such a specific way. I will try different versions and see if any will bind with A7105 protos.

bd4, 2023-11-04:
I updated Boxer to 1.3.3.33 (flash and lua), and now bound successfully to SCX-24. I did not see any diff between the .20 lua scripts and what I had installed along with .20 which was not working, so not sure what was going on.

Given that the latest is working, not sure if it's worth trying to excavate exactly what happened. When I have some more time I will try reflashing .20 to see if I can reproduce, but still going to close this for now.

closed 2023-11-04 by bd4