pascallanger/DIY-Multiprotocol-TX-Module #902: Add support for GA-4H generic Ebay controller

davidrattvik, 2023-11-04:
Hi and thank you for the work that you guys have done!

I have a bunch of common receivers that i believe would be nice to add to the protocol list.
I have captured a data stream from each of the pins on the transmitter chip (XN297L) ( it is connected to the common nRF24L01.
The controller does not seem to be smart in any way as you can connect multiple receivers to the same controller. you can also bind a receiver long after you started the controller. i tested this by running a couple receivers at the same time. during this time i connected another receiver and pressed the bind button on the reciever which connected directly and started to copy whart the other receivers where doing. therefore i don´t believe that there is any 2 way communication and that the controller sends its adress out at every data burst. the first adress that the reciever sees will be the one that it takes its commands from.
also there is no bind button on the transmitter.

each stream was captured with the controller in a different position.
Start no bind
Start bind
Full left
Full right
full throttle
full break
ch3 toggle
ch4 toggle.
I hope that i have collected enough data for you to be able to make a protocol from this.

captured with logic2

[GA-4H.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13257784/GA-4H.zip)

[image]
[image]
[image]
[image]
[image]

davidrattvik, 2023-11-04:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

davidrattvik, 2023-11-05:
attached below is a protocol dump made via visual studio code.
it should simplify the process i hope.

[rf signal dump.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13259649/rf.signal.dump.txt)

pascallanger, 2023-11-09:
Thanks, I'll have a look soon and let you know

pascallanger, 2023-11-10:
There are multiple issues:
 - Your sampling rate is a bit too low so some clocks are not seen
 - You do not let it run long enough on the files "start" and "start and bind"
    - When you say "start and bind", I assume the difference is that you placed the RX in bind mode ?
 - For the XN297dump text file, you need to set the address to 4 bytes instead of 5 like you've done (RX number=4)
 - Also the address on the SPI dump is different from over the air. Are you using 2 different TXs?

Right now, I'm just unsure how it all works. I have the impression it is like the Turbo Racing with RF frequencies going all over the place without a specific pattern.

pascallanger, 2023-11-10:
Also if you could use logic v1 instead since they've never fixed the export bug in the SPI analyzer of the V2 despite many requests...

pascallanger, 2023-11-10:
Also I need one channel per file in which you should move slowly from one end to the other. For example throttle would give a slow move mid->full->mid->break->mid. So 4 files for the 4 channels.
And an extra long file from power on, may be 1 or 2 minutes.

pascallanger, 2023-11-10:
More I look at it more it looks like the Turbo racing protocol that I haven't been able to reverse...

pascallanger, 2023-11-10:
Packets:

F4|96|96|64|C8|00
--|--|--|--|--|--
Type|P1|P2|P3|P4|P5
Type F4 = channels

P1=STR,P2=THR,P3=CH3,P4=CH4: 0x64..0x96..0xC8 100..150..200 ->small amplitude...

F3 | 01 | 83 | 0B | E4 | 00
-- | -- | -- | -- | -- | --
Type|P1|P2|P3|P4|P5

Type F3 = ??
P1 = 0x01
P2,P3,P4 = ??
P5 = 0x00 or 0x14
At least in the small dumps

FC|4D|76|E0|8B|00
-- | -- | -- | -- | -- | --
Type|P1|P2|P3|P4|P5

Type FC = ??
P1,P2,P3 = ??
P4= 0x8B
P5 = 0x00
At least in the small dumps

pascallanger, 2023-11-10:
At this stage I only need a extra long dump from power on and a good dose of luck.

davidrattvik, 2023-11-11:
Thanks for the feedback. Tonight i'll try and get all the information that you require. I thought that a couple of data bursts would be enough since we are talking about 2.4 ghz.
So to sum it up, you need more data and a slight tuning of my measuring method.
I am learning as I do this so I have a bit of homework cut out for me.

davidrattvik, 2023-11-11:
Also yes. You are correct pascallanger. As I remember I used different transmitters.

pascallanger, 2023-11-11:
> So to sum it up, you need more data and a slight tuning of my measuring method.

Yes, faster sample time on your logic analyzer and let it run for a LONG time

> As I remember I used different transmitters.

Please dump all the transmitters you have over the air as indicated previously

davidrattvik, 2023-11-12:
i don´t know if i will find the time to do all that you asked of me this weekend but i managed to connect each of my 4 TX´s to the analyzer and make a start and bind.
i captured 1 minute of data for each controller at 8 MS/s (whatever MS/s means) whitch for some reason was the highest speed at whitch my analyzer was able to work in logic1. i spent last evening trying to figure out what you meant with the 5 channel to 4 channel change in over the air read. I was not able to figure out exactly what you meant. i tried to change the settings during the debug session but could not see that the feed displaying 5 channels changed. I appreciate the time you take out of your day to support the community and i am sorry that i could not deliver all that you asked for.

[Uploading logic1 analys start and bind.zip…]()

i have attached the new dumps made with LOGIC 1 this time.

davidrattvik, 2023-11-12:
[logic1 analys start and bind.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13327506/logic1.analys.start.and.bind.zip)

pascallanger, 2023-11-12:
I will check if 8 million samples per second are enough but the dumps look ok.

They give us some small clues:

TX1 address: 45 **45** 5A **18**
TX2 address: 45 **8B** 5A **00**
TX3 address: 45 **1A** 5A **3D**
TX4 address: 45 **9F** 5A **03**

Packets FC bind:

FC|4D|76|E0|8B|00
-- | -- | -- | -- | -- | --
Type|P1|P2|P3|P4|P5

Type FC = bind
P1,P2,P3 = values change every packets but they are the same on the 4 TXs: same bytes sequence and no loop over 120 packets.
P4= TX_ADDR[1]
P5 = TX_ADDR[3]

Packets F3

F3 | 01 | 83 | 0B | E4 | 00
-- | -- | -- | -- | -- | --
Type|P1|P2|P3|P4|P5

Type F3 = ??
P1 = 0x01
P2,P3,P4 = values change every packets but they are the same on the 4 TXs: same bytes sequence and no loop over 120 
P5 = flip between 0x00 and 0x14 every other packet

Packets F4 are common to all 4 TXs:

F4|96|96|64|C8|00
--|--|--|--|--|--
Type|P1|P2|P3|P4|P5
Type F4 = channels

P1=STR,P2=THR,P3=CH3,P4=CH4: 0x64..0x96..0xC8 100..150..200 ->small amplitude...
P5 = 0x00

========================
F4 are sent on the same RF channels with no loop on the 4 TXs
F3 are sent on two RF channels which are different on the 4 TXs. The 2 RF channels alternate every packet.
FC are sent on one RF channel: 0x4E=78

davidrattvik, 2023-11-12:
Great!
Do you still want me to perform the airdumps as well? If you could guide me towards a site where i can learn to change the 5byte to 4 byte.
The problem i am refering to:
 
"The dumps look to have been done from 2 different TXs. I would need a good over the air dump from the 2 TXs over a long period of time using a bitrate of 250K using a 4 bytes TX address on channels 0x0F=15d, 0x23=35d, 0x4E=78d and any other channels with a good flow of packets starting by 0xF4. This might give some clues to figure out F3 and FC types."

Or did i just circumvent that with the logic analyzer?

pascallanger, 2023-11-12:
I think I have enough with the logic analyzer dumps you provided. If you could do one extra on any of the 4 TXs but really long, the longest the software allows you to do.

davidrattvik, 2023-11-12:
Ok. I'll give it a go later tonight. I assume that it just needs to stand idle during the sampling.  Also something that might be interresting is that button 3 is toggle and Button 4 returns automatically. I assume though that the receivers channels 3 and 4 dont care about that and will modulate a ppm signal either way.

pascallanger, 2023-11-12:
Yes just let it still during the capture.
Ch3 and ch4 are normal channels from a origin perspective.
You have to understand that there is a change that we won't be able to reverse this protocol...

davidrattvik, 2023-11-12:
[logic analys 100s to 500s.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13329018/logic.analys.100s.to.500s.zip)

the program limits me to a maximum of  approximately 500 seconds. The analyzer is limited to this at 2MS/s. The program also limits me from lowering the sampling rate lower than 2MS/s.
so the largest capture will be 500 seconds at 2MS/s.
Yes i understand that it might not be possible to create a protocol from this.
At least we, the community might learn something from this.

if i manage to increase the sample time i will post another zip-file.

davidrattvik, 2023-11-12:
i managed to sample 1800 seconds. the file is too large to upload to this chat.

pascallanger, 2023-11-12:
F3 frequencies have something in common:
TX1: 45 45 5A 18 , F3:1A,2D -> 2D-1A=13
TX2: 45 8B 5A 00 , F3:0F,23 -> 23-0F=14 => only this diff is different, but also note that both RFs are odd, others are even/odd or odd/even
TX3: 45 1A 5A 3D , F3:14,27 -> 27-14=13
TX4: 45 9F 5A 03,  F3:27,3A -> 3A-27=13

pascallanger, 2023-11-12:
RF covers all the frequencies from 0x05 to 0x4E at the exception of 0x10, 0x20, 0x30, 0x40, 0x4D
0x4E is only used for FC, so we could assume that all other RFs are between 0x05 to 0x4C  at the exception of 0x10, 0x20, 0x30, 0x40. It could be a final calculation like RF=(value % 0x48) + 5 .
Types F3 and F4 can use the same RF channels

pascallanger, 2023-11-12:
F4 frequencies follows a pattern ABCD EFGH ABCD EFGH ABC ???? IJKL MNOP IJKL MNOP IJK ???? ...

pascallanger, 2023-11-13:
I confirm that this is nearly the same as the Turbo racing protocol (types F3,F4,FC) which is using a different RF chip but the payload looks the same. For sure the same manufacturer.

pascallanger, 2023-11-13:
Ok the F3 frequencies are calculated based on TX_ADDR[1] TX_ADDR[3]:
RF0=( ( ( TX_ADDR[1] + TX_ADDR[3] ) & 7F ) % 48 ) + 5
RF1=RF0 + 13
if RF0==10 or 20 or 30 or 40 then RF0--
if RF1==10 or 20 or 30 or 40 then RF1--

The calculation of RF0 can't be 100% sure with the data we have, it could be one of these 3 solutions:
1. RF0=( ( ( TX_ADDR[1] + TX_ADDR[3] ) & 7F ) % 48 ) + 5
2. RF0=( ( ( TX_ADDR[1] & 7F ) + ( TX_ADDR[3] ) & 7F ) % 48 ) + 5
3. RF0=( ( ( TX_ADDR[1] & 7F ) + TX_ADDR[3] ) % 48 ) + 5
I'm not sure if 1 and 2 are equivalent...

TX1: 45 **45** 5A **18** , F3:**1A**,**2D** -> RF0=**1A** , RF1=RF0+13=**2D**
TX2: 45 **8B** 5A **00** , F3:**0F**,**23** -> RF0=10 , RF1=RF0+13=**23** => RF0=10 is not allowed RF0=RF0-1=**0F**
TX3: 45 **1A** 5A **3D** , F3:14,27 -> RF0=**14** , RF1=RF0+13=**27**
TX4: 45 **9F** 5A **03**, F3:27,3A  -> RF0=**27** , RF1=RF0+13=**3A**

pascallanger, 2023-11-13:
At this stage:
 - F3 frequencies are known: the 3 unknown bytes could be a sync on how to calculate the next F4 RF frequencies
 - FC frequency is known: the 3 unknown bytes could be a sync on how to calculate the next F4 RF frequencies
 - F4 frequencies are unknown but follow a "pattern", all bytes are fully known

That's still a lot of unknowns...

pascallanger, 2023-11-13:
@davidrattvik  Can you see if you can connect your logic analyzer on the RX? If you can that would allow us to send stuff to it and see how it will react.

davidrattvik, 2023-11-13:
Absolutely! I will give it a try after work. I guess that we will send known data via the multi protocol module and analyse what the receiver spits out.

pascallanger, 2023-11-13:
Yep 👍

pascallanger, 2023-11-13:
Can you launch a dump of this receiver from power on, bind with a TX and let it run for some time?

pascallanger, 2023-11-14:
I've purchased a 5€ receiver on ebay but it might take a month or two to get here...

pascallanger, 2023-11-14:
The files you uploaded so far are good. I don't think we need larger dumps.
Take your time, we are not in a hurry.
Old modules are based on an Atmega328 (nano) but the modules are now using a STM32 for some time. There is a big chance that you have the STM32 version, correct?
Thanks to @benlye , we have our own dedicated Multi boards for the Arduino IDE. Follow these instructions to compile: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#option-3---compiling-and-updating-firmware

davidrattvik, 2023-11-14:
here are 4 more dumps. the amount of data that is saved in the idle unbound mode is very high. therefore i cannot sample more than a minute or so until the filesize gets too big.

[2 MHz, 3 B Samples rx start, bindmode.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13355437/2.MHz.3.B.Samples.rx.start.bindmode.zip)

[2 MHz, 3 B Samples. bind mode, tx start, bind.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13355449/2.MHz.3.B.Samples.bind.mode.tx.start.bind.zip)

[2 MHz, 3 B Samples rx start. searching for bound tx.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13355454/2.MHz.3.B.Samples.rx.start.searching.for.bound.tx.zip)

[2 MHz, 3 B Samples rx start, search, bindmode, tx start,.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13355505/2.MHz.3.B.Samples.rx.start.search.bindmode.tx.start.zip)

[image]

[image]

davidrattvik, 2023-11-14:
> @davidrattvik Can you launch a dump of this receiver from power on, bind with an "unknown" TX (not the one it was bound to), let it run for a couple of seconds and finally turn off the TX?

I missread this. I'll make another dump with binding to unknown and turning of tx.

davidrattvik, 2023-11-14:
here is the one requested. start, bind and tx turn off.

[Uploading 2 MHz, 3 B Samples. rx start, bindmode, tx start, bind to tx, idle, tx stop, searching for paired tx.zip…]()

pascallanger, 2023-11-15:
You have a STM32 module has you can see on the big chip.
You've never updated it as I understand, is it correct? If that's the case use flash-multi to install the drivers, install the latest bootloader and write an existing firmware first: https://github.com/benlye/flash-multi
Then if you could try to compile the latest version of the code in arduino. Here is my Arduino's config:
[image]
Debug will only be used if we want to get data out from the module.
From there, I can start to code the "GA-4H" protocol and you'll be able to play with it and dump the RX to see how it behaves.

pascallanger, 2023-11-15:
Just looked at the RX dumps. The data (=MOSI&MISO) pin is not connected to the logic analyzer... Basically you should see 3 signals on the dumps and there are only 2 right now.
Here is the pin causing issue:
[image]
Could you check that your connections and soldering are correct?
Or may be the Data pin is not routed on the other side as I thought...

davidrattvik, 2023-11-15:
There is a connection at that pin. I measured it with my multimeter. I'll hockey my oscilloscope to it to see where to connect

davidrattvik, 2023-11-15:
[2 MHz, 3 B Samples rx start, bindmode, tx start, bind to tx,idle, tx stop, searching for paired tx. 3 channels.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13367876/2.MHz.3.B.Samples.rx.start.bindmode.tx.start.bind.to.tx.idle.tx.stop.searching.for.paired.tx.3.channels.zip)

i must have had a bad connection. now there is 3 data streams.
i.ll try the firmware updates tomorrow evening i think. i believe that i did an update on both the bootloader and firmware since i managed to get the Serial debug function running.
we´ll see tomorrow how it goes.

pascallanger, 2023-11-16:
I'm puzzled by the RX dump.
When the RX starts, it looks for the F3 packet on the recorded ID which is ok.
When it switches to bind, it listens on the FC packet on channel 0x4E using the address 55 45 05 08 which is ok.
But it gets strange when it receives a unknown packet change the ID then reads the packet and get the FC which has the ID in it?!?. From there all the packets have an offset with what it is supposed to get on the RF channel it is looking for... For example it switches to the F3 RF channel, it gets a F4, it changes the RF channel to a F4 and gets the F3... Totally weird and totally puzzled...
When the RX loses connectivity, it continues to hop 30 times then only look for F3 packets.

I think we could do something which is not great but:
 - Send a FC to get it to bind, need to see if it needs to receive something to store the ID right away
 - Send a F3 with a known sync
 - Send F4s with the RF channels corresponding to the sync
 - Send the SAME F3 with the SAME sync
 - Send F4s with the RF channels corresponding to the sync

If the RX receives the F3, I guess it will start to hop using that sync. Where it goes bad is if it doesn't receive the F3, it will continue hop but we can catch it on the next F3 and resync it. In this case it could lose a maximum of 19 packets -> (19+1)*14ms=280ms. There might be a trick to prevent this loss which is to send 2 x F4s, 1 F4 using the sync packet hopping and another F4 like if it has not received the F3 and we do that back to back. This way we should catch it whatever it does and send it back where we want it to be.

davidrattvik, 2023-11-17:
ok so the module is ready to receive a custom protocol.
i did the updates to the bootloader and deselected the one i used to "sniff" the transmitter.
arduino stops the compilation because of too many protocols otherwise all seems ready to go.

bill-healey, 2023-12-08:
Any luck with this?  It would be cool to be able to emulate the turbo racing protocol, their cars are quickly gaining in popularity.

davidrattvik, 2023-12-08:
Pascal got pretty far as I can see and I believe that he is taking some well deserved time of until the receiver that he ordered arrives. He is way ahead of me in understanding what is happening, but as far as I can tell, most of the protocol is understood and there are still some channel-switching that needs to be solved. Also pascal: no stress! Take your time and let me know if there is anything we/i can do to help you along.

davidrattvik, 2024-03-04:
Something that just struck me regarding these receivers: if you start the transmitter first and then press bind on several receivers, all receivers will bind to the same transmitter equally . So their is no real "pairing" here. Just a receiver listening to the first signal it hears. What if we were to send ch1234 under the first ID. And then transmit ch5678 under another ID. If the transmitter switched between these 2 IDs every other data burst, it would then be possible to use the receivers together to form a 8ch receiver.

davidrattvik, 2024-03-05:
Yes I am definitely available. Would you like me to purchase a new receiver for you? 
If there is anything else that I can help you with, let me know!

gigozago, 2025-01-29:
Hello, apols for jumping into this conversation. I have an OpenTX transmitter with 4-in-1 module, and I'm considering to buy a few of these generic GA-4H receivers (for RC planes I build from scratch). I struggle a bit to follow the discussion so far...would you say these receivers would easily bind by setting protocol NX297 Dump and Subprotocol 1M, as I think to have understood here please? Thanks a lot for your kind feedback.

pascallanger, 2025-01-29:
No these RXs are not supported...
I've never received the one I purchased to see if I could figure out the protocol. So for now it's a bit dead.

gigozago, 2025-01-29:
Thanks @pascallanger 

(With the risk of being slightly off topic...) I have ordered a GoolRC 3CH (looks very similar to the FlySky 3CH I'm using on a plane with fixed rudder), which I'm going to test soon. Otherwise, Dumbo RC 6CH are my favourite, have many of them. But it's nice to try and experiment new ones