pascallanger/DIY-Multiprotocol-TX-Module #903: Kyosho FHSS Receivers Will not bind

RichW012, 2023-11-05:
Hi there, my Radiomaster TX16s will not bind to a Kyosho RC Car receiver. The list of protocols includes Kyosho FHSS, my receiver is the KR-331. The original TX is the KT_231P+ I have tried all settings combinations and I have the latest EdgeTX "Providence" v2.9.1. and the module firmware V1.3.3.33. Thanks

pascallanger, 2023-11-05:
Who told you it's a multi protocol issue?
As far as I know it works, so I would like some proof.

RichW012, 2023-11-05:
Ive been on the Radio Master Products User Group and that's what they said.  Im not an expert in this so I bow to the greater knowledge. Its just been a case of trying everything with a receiver that is known to work, with a radio that's got no problems, and this is whats left. I understand the MiniZ receivers bind OK, but its my RTR Car ( Neo inferno3 )that won't. How can I supply proof?

pascallanger, 2023-11-06:
Do you have any protocol working with the A7105 RF component?

RichW012, 2023-11-06:
No I don't have any other receivers that uses a protocol using that component, so guessing where you are going, I need to test it somehow and make sure its not just a faulty A7105 RF component in my TX, is that possible with a multi-meter? I do have a diagram of the chip.

pascallanger, 2023-11-06:
A multimeter won't help. Do you know anyone around you with a flysky receiver? This is quite a known brand.

pascallanger, 2023-11-06:
I've been able to borrow a mini-z with a kt-531 TX. I'll have it on Saturday so I can test.

pascallanger, 2023-11-07:
My friend came with his mini-z FHSS with KT-531 radio. I've been able to bind his mini-z right away and control it with my TX16S.
Bind procedure: On the mini-z FHSS push the bind button and power it on -> led flashing quickly. Then launch a bind on the TX16S using KYOSHO/FHSS. That's it, you have full control of the car including CH4 which is for the front LEDs.
Here is the video: https://youtu.be/vBmwb3wiUTg

pascallanger, 2023-11-08:
If it still doesn't work when doing a proper bind sequence then it means that the transmitter KT_231P+ and KR-331 are not using the same "FHSS" protocol as the KT-531p. If you want this protocol to be added then you need either to ship me the TX+RX to France or sponsor the purchase of a TX+RX set locally (around 60€ incl shipping).

pascallanger, 2023-11-08:
Ok I think here it is: https://www.kyosho.com/jpn/support/instructionmanual/digital_prcs/pdf/82146_KRG-331_IM_A4.pdf
It's not the Kyosho FHSS protocol but what they call the "Syncro" protocol. Syncro TX are KT-231P, KT-231P+, KT-331, KT-432PT, KT-431S, KT-631ST.

RichW012, 2023-11-09:
Thank you for all your help Pascal! We have got to the bottom of the mystery! I will consider sponsoring the purchase, however I am in the process of moving and don't have spare money right now. As RadioMaster have just released the MT12 Surface Radio others might want to help. Thanks again

WazabiQc, 2024-01-06:
I was having the same issue with a Kyosho Blizzard, KT-431S radio with a KR-632 receiver.. I thought it was Kyosho FHSS at first as well because it's written all over it.. Would that be the same issue then ?

Wandereradi, 2024-01-17:
@pascallanger Well i am game with whatever you need, I was out of country so couldn't reply back. 

So now let know if I could use a Logic Analyser and share the output from the TX to the A7105 chip.

or a Link from where i could ship you a TX RX. Mind you I am in India so there is no way to return the radio to me ( it would be your)

other option 
I am confident enough do the the Logic analyser stuff if you have a thread on how to. Pretty good with soldering and stuff, I cant code thats all but analyser will not need coding so i am pretty ok, with a little guidance.

Wandereradi, 2024-01-26:
@pascallanger  please send me your details as to where to ship a new KT331 with a receiver. I tried a couple dataloggers ( clone ) but was not successfull.. thanks in advance

pascallanger, 2024-01-26:
@Wandereradi you can contact me at pascal _ langer @ yahoo . fr

Wandereradi, 2024-01-29:
@pascallanger  I have recorded the Logic for as below on the TX side. using 4MS/s  CH 0 -> EN, CH1 -> CLK, CH2 -> SDIO.
Binded Receiver - TX Logic Powered on
Binded Receiver - TX Forward & Reverse
Binded Receiver - TX Left & Right
Binded Receiver - TX Channel 3 
Fresh Receiver - TX Bind 

If there is any issue with the data please let me with what Sample rate should i Capture.
[KT331 Dump with 4MS_s.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14080865/KT331.Dump.with.4MS_s.zip)

pascallanger, 2024-01-29:
I've started to look at the dumps. Just by curiosity, is it an A7105 or another RF chip? Strangely they access an unknown register on the A7105, or there is something else.
I had to rewrite part of my decoder since you are using a Logic 2 version on which the SPI decoder is buggy... Logic 1 is fine. Anyway I can see what's going on.
The payload is huge with 37 bytes!
The radio is sending bind packets at every power up then switch to normal mode. The RX does not send anything to the TX so you don't need to power the RX when doing dumps.
They are using the same ID as Flysky and other Kyosho TXs: 0x5475c52A
I've quickly checked the RF channels, a sequence is repeating but not more than that for now.
I've checked the throttle capture, they transmit this channel on 2 bytes (LSB first) 1000 low 2000 high which is directly the PWM value. So the same will be true for all the others.
More to come...

pascallanger, 2024-01-29:
I should be able to code this protocol with what I have understood. I need to check this proto against the existing kyosho protocols to see if it's a variant.

pascallanger, 2024-01-29:
It's nearly a copy of the kyosho FHSS protocol. I need to check exactly why you can't bind, there is only one byte difference in the payload and RF channels are the same. Normal mode has also a few bytes difference.
What I haven't looked at all is the init of the component, may be that's where the real difference is.

Wandereradi, 2024-01-29:
Hi @pascallanger here are the pics from the RF board on the KT331.
you mentioned above that Logic2 has a issue so you mean to see the application installed on the laptop right?
which version are you using so i will also capture as per that. just in case. 
Yes the receiver has blinking light for a few seconds afte the Tx has powered up - may be 2secondish or max 3.

[image]
[image]

pascallanger, 2024-01-29:
Interesting it's an A7105 but they are accessing a register which doesn't exist, a bug in their fw probably so I don't care...
What happens when you try to bind the receiver with the Kyosho/FHSS protocol? Do you see something on the RX LED?

Wandereradi, 2024-01-29:
its a A7106 not 5

pascallanger, 2024-01-29:
I should put my glasses on 😄 The A7106 have this additional register but I don't think it matters.
Ok the init of the A7105/6 is the same as the Kyosho FHSS. The ID is also the same. The bind packet has only 1 byte difference. I'm going to create a sub protocol Syncro and change this byte to see if that's what preventing to bind.

pascallanger, 2024-01-29:
I've pushed all the differences I've seen between FHSS and Syncro. Let's see if it works, you can download the test version v1.3.3.44 here: https://downloads.multi-module.org/latest-test/
I think most of the changes are useless, so if it works I would like to come back on some of them to lighter the code so I would need you to do some more tests.

Wandereradi, 2024-01-29:
> I've pushed all the differences I've seen between FHSS and Syncro. Let's see if it works, you can download the test version v1.3.3.44 here: https://downloads.multi-module.org/latest-test/ I think [...]

Hi Pascal, I forgot to inform I have Radiomaster MT12 4 in 1 tx, so i would need a build for that.

Also correct me - I would need to put this Bin file in the firmware folder and run firmware from firmware in the sys page

pascallanger, 2024-01-29:
The zip file on the test link contains all the builds. You most likely want to use this one: mm-stm-serial-aetr-v1.3.3.44.bin
Place it in the firmware folder of the SD card then in SYS navigate to the SD card -> firmware -> click on the above bin file and flash the internal module as proposed (if the bin file doesn't appear reduce the file name length).

Wandereradi, 2024-01-29:
> The zip file on the test link contains all the builds. You most likely want to use this one: mm-stm-serial-aetr-v1.3.3.44.bin Place it in the firmware folder of the SD card then on the radio in SYS navigate to the SD card -> firmware -> click on the above bin file and flash it the internal module (if the bin file doesn't appear reduce the file name length).

Uploading the firmware to the Tx , fingers crossed. 

Edit - when i scroll the protocols, and come to Kyosho, it shows Subtype FHSSSY for a second and than changes to FHSS and i do not see a "Syncro"

Wandereradi, 2024-01-29:
Update - If i use Kyosho FHSS, it does not pair.

pascallanger, 2024-01-29:
You should see Kyosho FHSS->Hype-Sync
If you don't see that it means that you are either using an old version of EdgeTX (most likely) or the module is not talking anymore to the radio (you should see the module version displayed under the model where you select Multi

pascallanger, 2024-01-29:
Ok I made a typo on the subproto which might cause what you describe. The system is currently building the version v1.3.3.45, download it from the same place.

Wandereradi, 2024-01-29:
> Ok I made a typo on the subproto which might cause what you describe (FHSSHy). Download the version v1.3.3.45 from the same place and try. I'm not at home and I don't have any hw to test my code against.

I uploaded the new 1.3.3.45 but i do not see Syncro under Kyosho and FHSS doe pair.

pascallanger, 2024-01-29:
Do you see the hype subprotocol? Do you see the firmware version under multi?

Wandereradi, 2024-01-29:
> Do you see the hype subprotocol (Kyosho FHSS->Hype->Sync)? Do you see the firmware version under multi?

I do see what you mentioned above but do not see Firmware # AETR

pascallanger, 2024-01-29:
Can you take pictures of your radio screen ?

pascallanger, 2024-01-29:
Update your version of edgetx

Wandereradi, 2024-01-29:
> Update your version of edgetx

[image]
[image]

Wandereradi, 2024-01-29:
> Update your version of edgetx

How do I update the Edge Tx on RadioMaster MT12 4-in-1

pascallanger, 2024-01-29:
"No telemetry" is the issue here

Wandereradi, 2024-01-29:
> > > Update your version of edgetx
> > 
> > 
> > How do I update the Edge Tx on RadioMaster MT12 4-in-1
> 
> Download companion for edgetx and follow the instructions. I'm sure you will find plenty of tutorials

the Edge Buddy does not support MT12 radio that i have, ie do not see it in the list. Also I see only FHSS and Hype, when Kyosho is selected and there is no "Syncro" - 1.3.3.45 

[image]

pascallanger, 2024-01-29:
They give you the link to the latest version on the screen above.
Until the message no telemetry is displayed on the screen it won't work...

Wandereradi, 2024-01-29:
> They give you the link to the latest version on the screen above. Until the message no telemetry is displayed on the screen it won't work...

I have re-installed the edge tx firmware seen on above screenshot using the bootloader method but it is the same version.

pascallanger, 2024-01-29:
If you create a new model and select multi with whatever protocol (not Kyosho), do you see the version or no telemetry ?

Wandereradi, 2024-01-29:
> If you create a new model and select multi with whatever protocol (not Kyosho), do you see the version or no telemetry ?

Hi Pascal,

Yes, when i am using the 1.3.3.33, and bind with Flysky 2A - I do see Version, and sometimes  - I sometimes 
see Version # Binding.

pascallanger, 2024-01-29:
No the fw for the Mt12 is not different. I bet that if you load that model with the new firmware it will work. I must have introduced a bug in the Kyosho protocol which is doing the main code to run and therefore the radio not recurving info from the module. I should be home tomorrow and I'll investigate further.

Wandereradi, 2024-01-30:
> No the fw for the Mt12 is not different. I bet that if you load that model with the new firmware it will work. I must have introduced a bug in the Kyosho protocol which is preventing the main code to run and therefore the radio is not receiving info from the module. I should be home tomorrow and I'll investigate further.

I have created a new model when I was testing the code, but the behavior was same. Unless I am missing something

pascallanger, 2024-01-30:
You are not doing what I was asking or talking about but forget anyway this was like I was thinking a crash when you select the protocol kyosho/fhss...
Please test v1.3.3.46

Wandereradi, 2024-01-30:
@pascallanger  Observations:
1) When the Rx power cycle's binding is not affected ( TX was constally on )
2) When Tx power cycle's with the new build - Internal Rf  Mode -> Off and i have manually turn it to Multi and select Kyosho 
3) When we are trying to bind, FHSS also works but Only Syncro should work.
4) If i use a Flysky2A protocol receiver and power cycle the TX, it re binds and "Internal RF" - Mode is mantained and RX automatically binds

pascallanger, 2024-01-30:
1. Ok
2. Wy would the internal rf turns off? It has nothing to do with Multi since the radio stores the info so most likely a bug in the MT12 firmware. It's normal if the internal RF modules is off that you have to reselect Kyosho/Sync
3. This is what I would have guessed. The 2 protocols are so closed that only one byte is preventing the Sync RX to bind with FHSS. I think could even modify the FHSS protocol to bind FHSS and Sync RXs without having to add Sync...
4. Does the internal module stays on if you select Kyosho/FHSS? (my expectation is yes)

Please redownload the latest test firmware and test it. The version will be the same but I've done some code cleanup.

Wandereradi, 2024-01-30:
> 4. Does the internal module stays on after power cycle if you select Kyosho/FHSS? (my expectation is yes)   -> Yes it as you 
       mentioned

pascallanger, 2024-01-30:
> > 4. Does the internal module stays on after power cycle if you select Kyosho/FHSS? (my expectation is yes)   -> Yes it as you
> >    mentioned
> >    But  if we have selected Syncro than at power cycle of tx, the internal module is off and also a random protocol is selected.

It proves that it is a MT12 EdgeTX firmware bug. You can try the latest nightly to see if they have fixed it. As a temporary fix, bind in Syncro and go back to FHSS.

Wandereradi, 2024-01-30:
> It proves that it is a MT12 EdgeTX firmware bug. You can try the latest nightly to see if they have fixed it. As a temporary fix, bind in Syncro and go back to FHSS.

I do not see a MT12 EdgeTx firmware nightly build, do you have a link?

pascallanger, 2024-01-30:
https://github.com/EdgeTX/edgetx/releases

pascallanger, 2024-01-30:
If you click the link I sent you above it's the first top entry called nightly... The MT12 fw is in the zip file.
But you can do all that from the companion tool if you install it as I already suggested, in the settings you can ask it to install the latest nightly or latest stable on the radio for you...

Wandereradi, 2024-01-30:
> If you click the link I sent you above it's the first top entry called nightly... The MT12 fw is in the zip file. But you can do all that from the companion tool (not buddy) if you install it (installation files are on the same link as above but from the stable release edgetx-cpn-xxx-v2.9.3.zip) as I already suggested, in the settings you can ask it to install the latest nightly or latest stable on the radio for you...

I installed the Nightly build -> Installed the Syncro build -> Turned on the Internal Module -> Set Kyosho -> Syncro, Binded the receiver, -> power cycled the tx -> worked as expected i.e. RX was bound back on power on.

pascallanger, 2024-01-30:
Have you tried the latest test version?

Wandereradi, 2024-01-30:
> Have you tried the latest test version?

Yeah, it was the you told me where you did some clean up and the version would be same.

Wandereradi, 2024-01-31:
@RichW012 and @WazabiQc Guys could you also assist in testing the Build for KR331 receviers, I have had an accident with my car. 
@pascallanger thank your this quick turnaround, Really appriciate the help. 
Also I will be further testing the code, currently trying to get the trims and dual rate set up on the MT12 ( completely new on EdgeTx)

pascallanger, 2024-01-31:
Please test the latest v1.3.3.47 test version. There is only FHSS and Hype as sub protocols. FHSS will bind both FHSS and Syncro receivers.

Wandereradi, 2024-02-01:
> Please test the latest v1.3.3.47 version available here: https://downloads.multi-module.org/latest-test/ The sub protocols are FHSS and Hype. FHSS will bind and control both FHSS and Syncro receivers.

Tested the 1.3.3.47 build, 
Observation - 
1) Syncro is no longer displayed.
2) Fresh binding is a bit slowe compared to when we had subtype syncro - my understanding is that now Tx is sending 2 types to 
    bind so its taking time.
3) Flysky2A also binds the tx to rx.

I will test more on the weekend or tonight and repot back here.

pascallanger, 2024-02-01:
1. Ok
2. Not really... I haven't touched the total bind time. What you could observe is the RX accepting the bind packets 250ms after you press bind on the TX and not right away.
3. Too bad. But I think we are ok.
4. The bind payload size of Flysky2A is the same as Kyosho/FHSS and they use the same ID. So depending exactly on what the RX is checking it can just accept the bind but won't be controllable.
5. Good

I think I can close this issue now but let me know.

closed 2024-02-05 by pascallanger

Wandereradi, 2024-03-15:
> 1. Ok
> 2. Not really... I haven't touched the total bind time. What you could observe is the RX accepting the bind packets 250ms after you press bind on the TX and not right away.
> 3. Too bad. But I [...]

For #4 from above I am able to control the Kyosho. kR331 even if I have selected the FLYSKY2A protocol. 

Range is all is good - have been able to control my car for around 100mts which was more than enough for me. 

When would this code come to Release is what I wanted to know. 

Also I sent you an email some time back for some help. Would you have time to help on it. 

Also could we have Weight changed to Dual Rate or D rate

[2 more replies]