pascallanger/DIY-Multiprotocol-TX-Module #904: FS-iA6 not binding

carsonpoland, 2023-11-10:
Hi, i have bunch of flysky AFHDSA2 receivers, mostly iA6B and they working fine but im not able to bind FS-iA6 rx which is also afhdsa2 protocol. Is this a bug? My TX is a radiomaster boxer 4in1

carsonpoland, 2023-11-10:
yes i did and i have latest version

fiechr, 2023-12-25:
I just bound a iA6 (the one without the B) successfully a few days ago. This doesn't really help, but at least confirms that this combination should work.

Did you remove the bind plug and restart the receiver after binding?

Quaxbruch, 2024-10-29:
Have the same issue. Bind plug attached, power to receiver, fast flashing LED. Choosing HF Module ext., Multi, Flysky AFHDS2A, choosing free ID, pressing bind, bind window appears. Nothing else happens. Pressing bdin window => disappears
Switching off receiver, unpluging Bind cable, switching on receiver, 
Does not bind => Slow flashing LED. 
ETHOS 1.5.15, Multi firmware 1.3.4.0, IRX4 Lite module.

Worked fine with earlier releases. What is wrong?

Quaxbruch, 2024-11-03:
flipped through older issues, now ETHOS 1.5.18, Multi firmware **1.3.3.33**, IRX4 Lite module. IA6 reciever
Works fine:
1.) Set internal Hf module to off and external to on
2.) select "multimodule"; "flysky AFSHD2A", "PWM+IBUS"
3.) define UNIQUE Modell Id, otherwise all previous binds for this protocol are gone and need rebinding 
(Helpfull to keep list of Model-ID, model, receiver mapping notes
4.) press "Bind" 
5.) Put bind clip to receiver and power on receiver
6.) Pull bind clip 
=> BOUND!
7.) press bind field on TX in order to clear screen (will not go away by itself)

Fertig