pascallanger/DIY-Multiprotocol-TX-Module #906: Cannot switching banks

Passawon, 2023-11-12:
Hello. My module is Jumper JP 4in1 when I turn knob to 15 red LED is flashing 3 times. I press bind switch and cannot change the bank. Here is the [video](https://drive.google.com/drive/folders/1LKuYWh_1yZ6bJ8vArrFNKYrFDCF_7AyE) for the issue. Please help

pascallanger, 2023-11-12:
Have you updated the module to the latest version firmware? Which version I've you used?
What I see on the video is not how the module is supposed to boot with the Multi banking system... Assuming you've done the PPM and banks configuration as indicated on this github, can you upload your _config.h?
Here is how the bank process works: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#protocol-selection-in-ppm-mode

Passawon, 2023-11-12:
> Have you updated the module to the latest version firmware? Which version I've you used? What I see on the video is not how the module is supposed to boot with the Multi banking system... Assuming [...]

Hi Pascallanger, thank you very much for your reply. I cannot connect the module to my PC. I tried the driver from flash-multi 0.6.1 and CP210x USB to UART Bridge VCP Drivers they didn't solve the problem. Is there another way to update the firmware?

pascallanger, 2023-11-15:
You can connect directly on the serial port of the stm32 instead of going through the CP2102. You also should look at the BOOT pin to see if it is pull like it should to put the stm32 in bootloader mode.
And finally, you should head to the flash-multi repo since there are a lot of tips there.

closed 2023-11-15 by pascallanger