pascallanger/DIY-Multiprotocol-TX-Module #909: LOSI protocol has extremely limited range compared to the original transmitter

oss81, 2023-11-20:
I've bound my LOSI micro with my radiomaster zorro and noticed, that the range is extremely limited compared to the original transmitter. With the original transmitter I can drive via fpv to every room of my flat with no problems, with the zorro I get signal losses as soon as drive into another room.

pascallanger, 2023-11-21:
I've just looked at the LOSI protocol which is a bit unfinished with only 1 TX ID and 1 RF frequency.
Since this protocol only uses 1 RF frequency, it is highly subject to noise.
The start of the code I written a long time ago seems to suggest that the original TX was looking for a free channel but in the end I did not implemented that part. This is what happening when I don't have the TX and RX in my hands but working remotely with someone...
Could you check if the range is improving if you turn off FPV?
To finish this protocol I need a TX and RX to be sent to my attention or someone to sponsor the purchase of a set.

oss81, 2023-11-21:
I've just tested it: The range without FPV is exactly the same. Using the original TX I have no problem to go much further away. 

Using the Spectrum Analyzer of my Radiomaster: When turning the original TX on, I see that there is only one RF frequency, but it is at another frequency every time I turn the radio on.

If it helps I could do some remote investigation for you since I have an oscilloscope (PicoScope).

pascallanger, 2023-11-21:
An oscilloscope is not useful here. You need a logic analyzer and solder it to the RF component.

oss81, 2023-11-21:
The oscilloscope has a logic analyzer, soldering wouldn't be a problem either, I just need to know where to probe.

oss81, 2023-11-21:
Just realized that on the receiver (car) there is a CYWUSB6953 which is a microcontroller with built-in radio, so it's not possible to probe the SPI...

pascallanger, 2023-11-22:
It's not the receiver but the TX you want to connect to.

pascallanger, 2023-11-22:
I've added dynamic RF channel, not sure that it improves anything or if it will work each time you will select the protocol: https://downloads.multi-module.org/latest-test/

oss81, 2023-11-22:
I've flashed the mm-stm-serial-aetr... version of your test build to my zorro. Unfortunately it is now missing the Losi protocol. When I first tried to bind it wouldn't work and then I saw that under subtype was stated "invalid protocol", so I tried to select the Losi protocol again but it is missing.

pascallanger, 2023-11-22:
Oh I see you must have been running an old version or you compiled it by yourself since the LOSI protocol is excluded from the automated build due to two things single ID and low flash space.
If you build from the source yourself you can enable this protocol.

oss81, 2023-11-22:
I've used the newest version that Radiomaster itself provides (1.3.3.20). I'll try to compile it myself then. Does this mean that new official releases will not support Losi anymore?

pascallanger, 2023-11-22:
v1.3.3.33 is the latest version... 20 is almost a year old.
Unless we fix the protocol, I don't think I'll reenable it.

oss81, 2023-11-22:
I'll try to compile it myself and will try to dump the "power on" of my radio and send it to you, maybe it will help.

oss81, 2023-11-22:
The new version for the Losi protocol does not bind anymore:
the LED of the RX stops flashing during the bind process, but stays off afterwards.

I've then replaced the code of the Losi protocol with the the last version (leaving all other code on "master") and it binds normally, but still with the limited range.

oss81, 2023-11-22:
I've tried to dump the startup data, I'm not sure if it is correct:
[losi_startup.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13444383/losi_startup.zip)
You will need to download the pico software from here to open the dump file:
https://www.picotech.com/products/picoscope-7-software

oss81, 2023-11-25:
I was able to add the CS signal to the dump as well so the individual packets get detected, I think the dump is correct:
[image]

Here is the complete dump:
[losi_startup_cs.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13462752/losi_startup_cs.zip)

And here just the packets as CSV exported:
[losi_startup.csv](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13462754/losi_startup.csv)

pascallanger, 2023-11-27:
The capture is good but not long enough as there are only the bind packets in the dump... I'm missing the switch to normal packet with the RF channel selection, data code and may be the config that would improve the range.

oss81, 2023-11-27:
check this dump please:
[losi_startup_3s.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13468659/losi_startup_3s.zip)

It contains 3 waveforms, each 1 second long. Can you open it with the pico software or do you need the exported CSV data?

pascallanger, 2023-11-27:
Can you export it in CSV ?

pascallanger, 2023-11-27:
Do you have multiple TXs by any chance ?

oss81, 2023-11-27:
[losi_startup_1.csv](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13468735/losi_startup_1.csv)
[losi_startup_2.csv](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13468736/losi_startup_2.csv)
[losi_startup_3.csv](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13468737/losi_startup_3.csv)

pascallanger, 2023-11-27:
losi_startup_2.csv has the info we need since there is the switch from bind to normal
We need more losi_startup_2.csv or losi_startup_3.csv with different RF channels being used. The DATA_CODE in use depends on the RF frequency. These are really small dumps, what we need to collect are the 80 XX where XX is the RF channel and all the associated A3 XXs where the XXs are the data code.
From losi_startup_3.csv, it will give something like:
CHANNEL|CODE
---|---
45|C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC

oss81, 2023-11-27:
Here are two more, how many do you need? (I'll have to make a small script that extracts the info from the CSV directly)

Channel 23:
F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4
Channel 3B:
C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC

pascallanger, 2023-11-27:
as many as you can so we can figure out the relation between the DATA code and the channels

oss81, 2023-11-27:
is it possible that it is dependent on the TX ID?

pascallanger, 2023-11-27:
No it's independent by the look of it.
If you are sure of your dump then please continue to gather more and extend the table

oss81, 2023-11-27:
Channel: 4D
Code: 4C 97 9D BF B8 3D B5 BE 0C 5D 24 30 9F CA 6D BD
Channel: 17
Code: 7D 2D 49 54 D0 80 40 C1 B6 F2 E6 1B 80 5A 36 B4
Channel: 19
Code: F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4
Channel: 1D
Code: C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC
Channel: 37
Code: F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4
Channel: 41
Code: F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4
Channel: 4F
Code: C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC
Channel: 2D
Code: F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4

now the channels seem to repeat, I'll try more later

oss81, 2023-11-27:
some more

Channel: 43
Code: 4C 97 9D BF B8 3D B5 BE 0C 5D 24 30 9F CA 6D BD
Channel: 0B
Code: 17 FF 9E 21 36 90 C7 82 BC 5D 9A 5B EE 7F 42 EB
Channel: 27
Code: C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC
Channel: 3F
Code: 7D 2D 49 54 D0 80 40 C1 B6 F2 E6 1B 80 5A 36 B4
Channel: 45
Code: C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC
Channel: 3D
Code: 17 FF 9E 21 36 90 C7 82 BC 5D 9A 5B EE 7F 42 EB
Channel: 07
Code: 4C 97 9D BF B8 3D B5 BE 0C 5D 24 30 9F CA 6D BD
Channel: 35
Code: 7D 2D 49 54 D0 80 40 C1 B6 F2 E6 1B 80 5A 36 B4

oss81, 2023-11-27:
I think I got it:
the table is
F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4
17 FF 9E 21 36 90 C7 82 BC 5D 9A 5B EE 7F 42 EB
4C 97 9D BF B8 3D B5 BE 0C 5D 24 30 9F CA 6D BD
7D 2D 49 54 D0 80 40 C1 B6 F2 E6 1B 80 5A 36 B4
C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC

you get the index for the code by:
RF channel modulo 5

pascallanger, 2023-11-27:
RF channels go from 07 to 4F in your dumps. I'm wondering if you could see anything above or bellow? like the lowest would be 05 or even 00? I think 4F is the max value but if you spot anything above let me know.

pascallanger, 2023-11-27:
Try the latest code I've pushed a second ago. It uses your TX ID and variable RF freq using your table.

DATA code is used to encapsulate the packet so without it you shouldn't be able to read the original message.

pascallanger, 2023-11-27:
From what I'm starting to guess the code I posted might work with multiple IDs.

If I look at my first implementation I can see that the TX listens with all these data codes on channel 27 where 27%5=4 :
E1 D6 31 26 5F BD 40 93 DC 68 08 99 97 AE AF 8C
DC 68 08 99 97 AE AF 8C C3 0E 01 16 0E 32 06 BA
**C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC**
E0 83 01 FA AB 3E 8F AC 5C D5 9C B8 46 9C 7D 84
5C D5 9C B8 46 9C 7D 84 F1 C6 FE 5C 9D A5 4F B7
F1 C6 FE 5C 9D A5 4F B7 58 B5 B3 DD 0E 28 F1 B0
58 B5 B3 DD 0E 28 F1 B0 5F 30 3B 56 96 45 F4 A1
5F 30 3B 56 96 45 F4 A1 03 BC 6E 8A EF BD FE F8

As you can see the line starting with C3 is in your table index 4, so it looks like it uses a table of 4 x 8 DATA code table
Note that only the first 8 bytes are used in 64 SDR mode and that the next 8 bytes is in fact the beginning of the next row.

So for me the data codes are TX ID independent, we just need to dump the full 4 x 8 x 8 = 256 bytes table to be able to do exactly like them. After this is really large to store, so I'm not sure we want to go there... But may be once we have the full table we will see a way to build it.

oss81, 2023-11-27:
How did you get different data codes for the same channel? I always get the same data code for the same channel, I've had channel 1D for example multiple times, but the data code was always the same. Does the data code change over time? I've always took the data codes after 3 seconds from startup.

pascallanger, 2023-11-27:
You can see the different data codes in your dumps losi_startup and losi_startup_1. Check at the beginning the TX looks for a free channel (low RSSI) then it tries all eight data codes then pick one and stay with it.
It only selects the same one each time since there are just no other TXs around I guess.

oss81, 2023-11-27:
Ok, I guess I understand it now. It is basically CDMA and ideally, you can have 8 TX on the same channel at a time.
I will dump the TX again until I have all 5 codes at second 3. I will then extract all 8 codes from the first second for each of the 5 dumps. I hope I can do it later today or tomorrow.

oss81, 2023-11-27:
Why did you say 4x8 data codes table, it should be 5x8?

pascallanger, 2023-11-27:
> Why did you say 4x8 data codes table, it should be 5x8?

Yes you are right 5x8x8

Please test the latest code.

oss81, 2023-11-27:
still the same unfortunately.

pascallanger, 2023-11-27:
TX ID 47 52 AE AA -> CRC xor 0xB
TX ID 56 52 22 8A -> CRC xor 0xF

oss81, 2023-11-27:
line 67 has one "LOSI_Check(" too much:
uint16_t val = LOSI_check(LOSI_check(Channel_data[i]<<1);

other than that this version does bind! And the range is great, at least in my flat I don't have any range problems anymore, even if I'm at another floor the link is still up!

pascallanger, 2023-11-28:
Ok that's good but we are not done yet. But I'm hopping you are willing to continue since having multiple IDs is really important...
Assuming yes, can you comment your ID+crc8 and uncomment the other ID+crc8? Does it work like I suspect?

If it works then we need to find out how the xor is calculated. For this, we need to get plenty xor values which is doable with your current setup.
The way to do it is the following:
 - change the ID, 1 bit at a time or an increment or even starts with 0x00000000, compile and upload
 - bind your RX to this ID
 - then with this modification in the code crc8 = RX_num; inserted in the callback function you will be able to cycle the xor from 0 to 15 using the RX number field on the radio. When the RX lights up and you have control, it means that you've found the correct xor.
 - repeat from the beginning with a new ID
You want to first see if all the bytes from the ID are used for the calculation by altering them one by one. Then do increments on 1 byte only and see if the xor increments too and so on.

oss81, 2023-11-28:
unfortunately it does not bind when I change the ID and the CRC8:
`rx_tx_addr[0] = 0x47;
    rx_tx_addr[1] = 0x52;
    rx_tx_addr[2] = 0xAE;
    rx_tx_addr[3] = 0xAA;
    crc8 = 0x0B;`

oss81, 2023-11-28:
Here are all the codes, for each table entry we figured out earlier, there are 9 codes:

**0: F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4**
Code: D0 D2 8E BC 82 2F E3 B4 8C FA 47 9B 83 A5 66 D0
Code: 8C FA 47 9B 83 A5 66 D0 07 BD 9F 26 C8 31 0F B8
Code: 07 BD 9F 26 C8 31 0F B8 EF 03 95 89 B4 71 61 9D
Code: EF 03 95 89 B4 71 61 9D 40 BA 97 D5 86 4F CC D1
Code: 40 BA 97 D5 86 4F CC D1 D7 A1 54 B1 5E 89 AE 86
Code: 03 BC 6E 8A EF BD FE F8 88 17 13 3B 2D BF 06 D6
Code: 88 17 13 3B 2D BF 06 D6 F1 94 30 21 A1 1C 88 A9
Code: F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4
Code: D7 A1 54 B1 5E 89 AE 86 C9 2C 06 93 86 B9 9E D7

**1: 17 FF 9E 21 36 90 C7 82 BC 5D 9A 5B EE 7F 42 EB**
Code: BC 5D 9A 5B EE 7F 42 EB 24 F5 DD F8 7A 77 74 E7
Code: 24 F5 DD F8 7A 77 74 E7 3D 70 7C 94 DC 84 AD 95
Code: 3D 70 7C 94 DC 84 AD 95 1E 6A F0 37 52 7B 11 D4
Code: 1E 6A F0 37 52 7B 11 D4 62 F5 2B AA FC 33 BF AF
Code: 62 F5 2B AA FC 33 BF AF 40 56 32 D9 0F D9 5D 97
Code: 83 F7 A8 2D 7A 44 64 D3 3F 2C 4E AA 71 48 7A C9
Code: 3F 2C 4E AA 71 48 7A C9 17 FF 9E 21 36 90 C7 82
Code: 17 FF 9E 21 36 90 C7 82 BC 5D 9A 5B EE 7F 42 EB
Code: D7 A1 54 B1 5E 89 AE 86 C9 2C 06 93 86 B9 9E D7

**2: 4C 97 9D BF B8 3D B5 BE 0C 5D 24 30 9F CA 6D BD**
Code: 0C 5D 24 30 9F CA 6D BD 50 14 33 DE F1 78 95 AD
Code: 50 14 33 DE F1 78 95 AD 0C 3C FA F9 F0 F2 10 C9
Code: 0C 3C FA F9 F0 F2 10 C9 F4 DA 06 DB BF 4E 6F B3
Code: F4 DA 06 DB BF 4E 6F B3 9E 08 D1 AE 59 5E E8 F0
Code: 9E 08 D1 AE 59 5E E8 F0 C0 90 8F BB 7C 8E 2B 8E
Code: 40 56 32 D9 0F D9 5D 97 8E 4A D0 A9 A7 FF 20 CA
Code: 8E 4A D0 A9 A7 FF 20 CA 4C 97 9D BF B8 3D B5 BE
Code: 4C 97 9D BF B8 3D B5 BE 0C 5D 24 30 9F CA 6D BD
Code: D7 A1 54 B1 5E 89 AE 86 C9 2C 06 93 86 B9 9E D7

**3: 7D 2D 49 54 D0 80 40 C1 B6 F2 E6 1B 80 5A 36 B4**
Code: B6 F2 E6 1B 80 5A 36 B4 42 AE 9C 1C DA 67 05 F6
Code: 42 AE 9C 1C DA 67 05 F6 9B 75 F7 E0 14 8D B5 80
Code: 9B 75 F7 E0 14 8D B5 80 BF 54 98 B9 B7 30 5A 88
Code: BF 54 98 B9 B7 30 5A 88 35 D1 FC 97 23 D4 C9 88
Code: 35 D1 FC 97 23 D4 C9 88 E1 D6 31 26 5F BD 40 93
Code: C0 90 8F BB 7C 8E 2B 8E 80 69 26 80 08 F8 49 E7
Code: 80 69 26 80 08 F8 49 E7 7D 2D 49 54 D0 80 40 C1
Code: 7D 2D 49 54 D0 80 40 C1 B6 F2 E6 1B 80 5A 36 B4
Code: D7 A1 54 B1 5E 89 AE 86 C9 2C 06 93 86 B9 9E D7

**4: C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC**
Code: E0 83 01 FA AB 3E 8F AC 5C D5 9C B8 46 9C 7D 84
Code: 5C D5 9C B8 46 9C 7D 84 F1 C6 FE 5C 9D A5 4F B7
Code: F1 C6 FE 5C 9D A5 4F B7 58 B5 B3 DD 0E 28 F1 B0
Code: 58 B5 B3 DD 0E 28 F1 B0 5F 30 3B 56 96 45 F4 A1
Code: 5F 30 3B 56 96 45 F4 A1 03 BC 6E 8A EF BD FE F8
Code: E1 D6 31 26 5F BD 40 93 DC 68 08 99 97 AE AF 8C
Code: DC 68 08 99 97 AE AF 8C C3 0E 01 16 0E 32 06 BA
Code: C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC
Code: D7 A1 54 B1 5E 89 AE 86 C9 2C 06 93 86 B9 9E D7

oss81, 2023-11-28:
Here are the exported CSVs, naming of the files is:
<first byte of actually used code>_<captured second>.csv

[losi_codes.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/13493384/losi_codes.zip)

column MISO is not correct in these files, since my wire fell off and I can't solder it back on...

oss81, 2023-11-28:
Just saw that the ninth entry is actually always the same, it is the bind code... so there are actually 8 codes per table entry, 5x8 as we already figured out

oss81, 2023-11-28:
using this table (replaced entries of the table we had with the 5th entry of the "complete table")
const uint8_t LOSI_data_code[][16] = {
  { 0x40, 0xBA, 0x97, 0xD5, 0x86, 0x4F, 0xCC, 0xD1, 0xD7, 0xA1, 0x54, 0xB1, 0x5E, 0x89, 0xAE, 0x86 },
  { 0x62, 0xF5, 0x2B, 0xAA, 0xFC, 0x33, 0xBF, 0xAF, 0x40, 0x56, 0x32, 0xD9, 0x0F, 0xD9, 0x5D, 0x97 },
  { 0x9E, 0x08, 0xD1, 0xAE, 0x59, 0x5E, 0xE8, 0xF0, 0xC0, 0x90, 0x8F, 0xBB, 0x7C, 0x8E, 0x2B, 0x8E },
  { 0x35, 0xD1, 0xFC, 0x97, 0x23, 0xD4, 0xC9, 0x88, 0xE1, 0xD6, 0x31, 0x26, 0x5F, 0xBD, 0x40, 0x93 },
  { 0x5F, 0x30, 0x3B, 0x56, 0x96, 0x45, 0xF4, 0xA1, 0x03, 0xBC, 0x6E, 0x8A, 0xEF, 0xBD, 0xFE, 0xF8 },
};

it works with your original TX ID. I've restarted my TX multiple times so I guess I got different channels...

Looks like there is another mapping from the TX ID to an index of the "complete table". Each TX ID can only use 5 codes. If this is true we unfortunately need the complete table for random TX IDs.

pascallanger, 2023-11-29:
Note that crc8 >> 1 seems  to give the entry in the tables:
F>>1 -> 7th entry
B>>1 -> 5th entry

oss81, 2023-11-29:
I've tried adding 
crc8 = RX_num;
at the begin of LOSI_callback() to be able to test our assumptions. I tried it with your original TX ID, so it should work with RX_num==11. Unfortunately this does only work if the RX_num is already correct during binding, why could that be? When I use 11 during binding it works and the link goes down when I change RX_num and comes up again when I set it to 11.

pascallanger, 2023-11-29:
The table seems to be rotated.
The first eight are followed by the the first eight of the next line. But you can see that always for the same index it doesn't work...
So let's rotate the tables to have them like they have it in their memory.

oss81, 2023-11-29:
problem is that for the first block, the bold line "links" to the bind code and the second 8 bytes of the bind code aren't anywhere in out tables...

Code: D0 D2 8E BC 82 2F E3 B4 8C FA 47 9B 83 A5 66 D0
Code: 8C FA 47 9B 83 A5 66 D0 07 BD 9F 26 C8 31 0F B8
Code: 07 BD 9F 26 C8 31 0F B8 EF 03 95 89 B4 71 61 9D
Code: EF 03 95 89 B4 71 61 9D 40 BA 97 D5 86 4F CC D1
Code: 40 BA 97 D5 86 4F CC D1 **D7 A1 54 B1 5E 89 AE 86**
Code: 03 BC 6E 8A EF BD FE F8 88 17 13 3B 2D BF 06 D6
Code: 88 17 13 3B 2D BF 06 D6 F1 94 30 21 A1 1C 88 A9
Code: F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4
Code: D7 A1 54 B1 5E 89 AE 86 C9 2C 06 93 86 B9 9E D7

oss81, 2023-11-29:
maybe that's just because the bind code is the very last entry of the original table...

oss81, 2023-11-29:
That would be the whole table, the bind code is the last entry

00: 83 F7 A8 2D 7A 44 64 D3 3F 2C 4E AA 71 48 7A C9
01: 3F 2C 4E AA 71 48 7A C9 17 FF 9E 21 36 90 C7 82
02: 17 FF 9E 21 36 90 C7 82 BC 5D 9A 5B EE 7F 42 EB
03: BC 5D 9A 5B EE 7F 42 EB 24 F5 DD F8 7A 77 74 E7
04: 24 F5 DD F8 7A 77 74 E7 3D 70 7C 94 DC 84 AD 95
05: 3D 70 7C 94 DC 84 AD 95 1E 6A F0 37 52 7B 11 D4
06: 1E 6A F0 37 52 7B 11 D4 62 F5 2B AA FC 33 BF AF
07: 62 F5 2B AA FC 33 BF AF 40 56 32 D9 0F D9 5D 97
08: 40 56 32 D9 0F D9 5D 97 8E 4A D0 A9 A7 FF 20 CA
09: 8E 4A D0 A9 A7 FF 20 CA 4C 97 9D BF B8 3D B5 BE
10: 4C 97 9D BF B8 3D B5 BE 0C 5D 24 30 9F CA 6D BD
11: 0C 5D 24 30 9F CA 6D BD 50 14 33 DE F1 78 95 AD
12: 50 14 33 DE F1 78 95 AD 0C 3C FA F9 F0 F2 10 C9
13: 0C 3C FA F9 F0 F2 10 C9 F4 DA 06 DB BF 4E 6F B3
14: F4 DA 06 DB BF 4E 6F B3 9E 08 D1 AE 59 5E E8 F0
15: 9E 08 D1 AE 59 5E E8 F0 C0 90 8F BB 7C 8E 2B 8E
16: C0 90 8F BB 7C 8E 2B 8E 80 69 26 80 08 F8 49 E7
17: 80 69 26 80 08 F8 49 E7 7D 2D 49 54 D0 80 40 C1
18: 7D 2D 49 54 D0 80 40 C1 B6 F2 E6 1B 80 5A 36 B4
19: B6 F2 E6 1B 80 5A 36 B4 42 AE 9C 1C DA 67 05 F6
20: 42 AE 9C 1C DA 67 05 F6 9B 75 F7 E0 14 8D B5 80
21: 9B 75 F7 E0 14 8D B5 80 BF 54 98 B9 B7 30 5A 88
22: BF 54 98 B9 B7 30 5A 88 35 D1 FC 97 23 D4 C9 88
23: 35 D1 FC 97 23 D4 C9 88 E1 D6 31 26 5F BD 40 93
24: E1 D6 31 26 5F BD 40 93 DC 68 08 99 97 AE AF 8C
25: DC 68 08 99 97 AE AF 8C C3 0E 01 16 0E 32 06 BA
26: C3 0E 01 16 0E 32 06 BA E0 83 01 FA AB 3E 8F AC
27: E0 83 01 FA AB 3E 8F AC 5C D5 9C B8 46 9C 7D 84
28: 5C D5 9C B8 46 9C 7D 84 F1 C6 FE 5C 9D A5 4F B7
29: F1 C6 FE 5C 9D A5 4F B7 58 B5 B3 DD 0E 28 F1 B0
30: 58 B5 B3 DD 0E 28 F1 B0 5F 30 3B 56 96 45 F4 A1
31: 5F 30 3B 56 96 45 F4 A1 03 BC 6E 8A EF BD FE F8
32: 03 BC 6E 8A EF BD FE F8 88 17 13 3B 2D BF 06 D6
33: 88 17 13 3B 2D BF 06 D6 F1 94 30 21 A1 1C 88 A9
34: F1 94 30 21 A1 1C 88 A9 D0 D2 8E BC 82 2F E3 B4
35: D0 D2 8E BC 82 2F E3 B4 8C FA 47 9B 83 A5 66 D0
36: 8C FA 47 9B 83 A5 66 D0 07 BD 9F 26 C8 31 0F B8
37: 07 BD 9F 26 C8 31 0F B8 EF 03 95 89 B4 71 61 9D
38: EF 03 95 89 B4 71 61 9D 40 BA 97 D5 86 4F CC D1
39: 40 BA 97 D5 86 4F CC D1 D7 A1 54 B1 5E 89 AE 86
40: D7 A1 54 B1 5E 89 AE 86 C9 2C 06 93 86 B9 9E D7

oss81, 2023-11-29:
without the duplication it would look like this:

00: 83 F7 A8 2D 7A 44 64 D3
01: 3F 2C 4E AA 71 48 7A C9
02: 17 FF 9E 21 36 90 C7 82
03: BC 5D 9A 5B EE 7F 42 EB
04: 24 F5 DD F8 7A 77 74 E7
05: 3D 70 7C 94 DC 84 AD 95
06: 1E 6A F0 37 52 7B 11 D4
07: 62 F5 2B AA FC 33 BF AF
08: 40 56 32 D9 0F D9 5D 97
09: 8E 4A D0 A9 A7 FF 20 CA
10: 4C 97 9D BF B8 3D B5 BE
11: 0C 5D 24 30 9F CA 6D BD
12: 50 14 33 DE F1 78 95 AD
13: 0C 3C FA F9 F0 F2 10 C9
14: F4 DA 06 DB BF 4E 6F B3
15: 9E 08 D1 AE 59 5E E8 F0
16: C0 90 8F BB 7C 8E 2B 8E
17: 80 69 26 80 08 F8 49 E7
18: 7D 2D 49 54 D0 80 40 C1
19: B6 F2 E6 1B 80 5A 36 B4
20: 42 AE 9C 1C DA 67 05 F6
21: 9B 75 F7 E0 14 8D B5 80
22: BF 54 98 B9 B7 30 5A 88
23: 35 D1 FC 97 23 D4 C9 88
24: E1 D6 31 26 5F BD 40 93
25: DC 68 08 99 97 AE AF 8C
26: C3 0E 01 16 0E 32 06 BA
27: E0 83 01 FA AB 3E 8F AC
28: 5C D5 9C B8 46 9C 7D 84
29: F1 C6 FE 5C 9D A5 4F B7
30: 58 B5 B3 DD 0E 28 F1 B0
31: 5F 30 3B 56 96 45 F4 A1
32: 03 BC 6E 8A EF BD FE F8
33: 88 17 13 3B 2D BF 06 D6
34: F1 94 30 21 A1 1C 88 A9
35: D0 D2 8E BC 82 2F E3 B4
36: 8C FA 47 9B 83 A5 66 D0
37: 07 BD 9F 26 C8 31 0F B8
38: EF 03 95 89 B4 71 61 9D
39: 40 BA 97 D5 86 4F CC D1
40: D7 A1 54 B1 5E 89 AE 86 
41: C9 2C 06 93 86 B9 9E D7

pascallanger, 2023-11-29:
//(Freq-1)%5=0
83 F7 A8 2D 7A 44 64 D3
3F 2C 4E AA 71 48 7A C9
17 FF 9E 21 36 90 C7 82
BC 5D 9A 5B EE 7F 42 EB
24 F5 DD F8 7A 77 74 E7
3D 70 7C 94 DC 84 AD 95
1E 6A F0 37 52 7B 11 D4
62 F5 2B AA FC 33 BF AF
//(Freq-1)%5=1
40 56 32 D9 0F D9 5D 97
8E 4A D0 A9 A7 FF 20 CA
4C 97 9D BF B8 3D B5 BE
0C 5D 24 30 9F CA 6D BD
50 14 33 DE F1 78 95 AD
0C 3C FA F9 F0 F2 10 C9
F4 DA 06 DB BF 4E 6F B3
9E 08 D1 AE 59 5E E8 F0
//(Freq-1)%5=2
C0 90 8F BB 7C 8E 2B 8E
80 69 26 80 08 F8 49 E7
7D 2D 49 54 D0 80 40 C1
B6 F2 E6 1B 80 5A 36 B4
42 AE 9C 1C DA 67 05 F6
9B 75 F7 E0 14 8D B5 80
BF 54 98 B9 B7 30 5A 88
35 D1 FC 97 23 D4 C9 88
//(Freq-1)%5=3
E1 D6 31 26 5F BD 40 93
DC 68 08 99 97 AE AF 8C
C3 0E 01 16 0E 32 06 BA
E0 83 01 FA AB 3E 8F AC
5C D5 9C B8 46 9C 7D 84
F1 C6 FE 5C 9D A5 4F B7
58 B5 B3 DD 0E 28 F1 B0
5F 30 3B 56 96 45 F4 A1
//(Freq-1)%5=4
03 BC 6E 8A EF BD FE F8
88 17 13 3B 2D BF 06 D6
F1 94 30 21 A1 1C 88 A9
D0 D2 8E BC 82 2F E3 B4
8C FA 47 9B 83 A5 66 D0
07 BD 9F 26 C8 31 0F B8
EF 03 95 89 B4 71 61 9D
40 BA 97 D5 86 4F CC D1
//Bind
D7 A1 54 B1 5E 89 AE 86

TX_ID 56 52 22 8A => crc8 = 0x0F -> xorout, num_ch = 0x02 -> line in (Freq-1)%5
TX_ID 47 52 AE AA => crc8 = 0x0B -> xorout, num_ch = 0x07 -> line in (Freq-1)%5
With crc8 00..0F and num_ch 00..07 both calculated out of the ID

In fact, we don't care, but it could (Freq-6)%5 since the smallest frequency is 7

pascallanger, 2023-11-29:
Could you test the latest code on github? You should be able to get both IDs to work by uncommenting/commenting the sections.

oss81, 2023-11-29:
I've tried it with the first ID, it would bind normally and I could steer the car. However it now seems broken. The RX LED is now always on after startup even if the TX is off and I can't even bring it into bind mode anymore, it does not react to the bind button anymore... Any idea?

pascallanger, 2023-11-29:
bind button broken ?

oss81, 2023-11-29:
I guess not, since the LED is now always on  at startup:
- if the bind button is not connected at all, the LED would be off
- if the bind button is permanently pressed, it would start in bind mode and flash

Maybe they added a bind limit in the RX and when you exceed it it locks up?

pascallanger, 2023-11-29:
That would be really surprising and nothing in the code should have changed. I've just moved stuff around...

oss81, 2023-11-29:
Yes I know, I don't get it either, maybe something other just broke but I can't bind it anymore, I just know that I've bound it quite a lot now and since you always get the TX and RX together and normally don't ever need to bind, it could be a hard limit in the firmware. Maybe I find something online...

oss81, 2023-11-29:
Ooops, I guess I know what the problem is: I've never charged the batteries of the car since testing and now they seem completely flat, I cannot even charge them anymore, I guess the LED always on means just that...

I'll have to try to revive the battery since I don't have a spare one.

[41 more replies]

closed 2023-12-02 by pascallanger