pascallanger/DIY-Multiprotocol-TX-Module #910: J6 protocol or nine eagles protocol stopped binding after v1.3.3.33

Jadnic, 2023-11-22:
I have a few old scale model, that runs the nine eagles J6 protocol, it has been working since version v1.3.3.20 but after updating to v1.3.3.33 all my models stop binding to it. reverting back to v1.3.3.20 is the only way to fly my J6 pro protocol models. so for now i have to have both old and new firmware to fly.

pascallanger, 2023-11-22:
Please try v1.3.3.38 available here : https://downloads.multi-module.org/latest-test/

pascallanger, 2023-12-13:
No answer considering it fixed...

closed 2023-12-13 by pascallanger

Ralf-W, 2024-02-12:
Hello Pascal,
I had the same problem: v.1.3.33 does not bind to my ancient  Solo Pro 180. 

I just flashed v1.3.3.47 and my TX16S does bind again to my ancient  Solo Pro 180 using J6Pro protocol. Problem solved.

Thanks a lot!