pascallanger/DIY-Multiprotocol-TX-Module #912: Traxxas protocol doesn't appear to bind successfully

jakezilla, 2023-12-11:
I've got the new MT12 and I'm unable to get it to bind with the Traxxas 6519 RX. I've made a YouTube video of my attempts. I also tried with my TX16S on 1.3.1.77 and it behaved the same way.

https://youtu.be/HnF-8NEGKnE?si=b4xi0LlS4DzgPRua

Let me know if there is something else I can try. Thank you!

jakezilla, 2023-12-19:
Is there any other information I can provide?

Do you think playing with 
TRAXXAS_FORCE_ID
would help?

pascallanger, 2023-12-22:
I can bind and control the RX I have without any problem with any TX ID so I know what I'm doing is mostly ok.

Since it's not working for you guys, my guess is that the RX is the one holding the keys here. The RX sends data to the TX during the bind and most of them are unknown, with only one RX it's hard to find out anything. A lot of things could be set by these bytes: RF channel or/and a different sop code or/and a different CRC seed or/and...

On my side to be able to go further, I need to write a RX code so I can see how the original TX will react to some of the data bind change and see if I can figure out what needs to be changes and how. This will take me some time, sorry no quick fix here...

pascallanger, 2023-12-22:
Ok I've spent some time on it and I have a working setup. The CRC is using the TXID and RXID, that's why it only works with my RX.
I'm puzzled by 2 things:
1. You can't bind multiple RXs to a single TX. The TX can only communicate with the latest bound RX, is my assumption correct ?
2. The amount of IDs is low with only 2 bytes... I'm wondering if there is something else...

I need to figure out how they handle the RF channel allocation. Right now my original TX is always on sending on channel 5 because it's basically free to use.

pascallanger, 2023-12-22:
Test version v1.3.3.42 available here: https://downloads.multi-module.org/latest-test/
It should allow you to bind and control the Traxxas TQ 6519 RX.
Please report.

legop3, 2024-01-02:
Hi, I was having the same issue, trying the .42 test release now on my TX16S with the 6519 and it is still not working properly. When putting the RX into bind mode, and pressing the bind button on the tx16s, the light on the RX stops flashing red right away, after a few seconds it blinks green once very briefly, then goes back to blinking red. Am I missing a configuration option somewhere? Do I need to set an ID?

pitts-mo, 2024-01-07:
After seeing youtube videos describing Traxxas TQi binding I had also wondered what they were doing.

It seems that both the TX and RX are able to store multiple "bindings" that seem to function more like bluetooth pairings. 
This allows BOTH the TX and RX to lock their link to the first "paired" device that is available (not already linked) after power up.

https://traxxas.com/news/how-to-become-tqi-power-user

legop3, 2024-01-14:
@pitts-mo I believe that traxxas TQ and TQi are two different systems, TQi being the more modern one. The 6519 is TQ, not TQi. Not trying to be rude, just letting you know.

pascallanger, 2024-02-22:
Traxxas TQ should be working now. Please test.

pascallanger, 2024-02-23:
Air and Surface builds are available here: https://downloads.multi-module.org/latest-test/
Please test the SFC fw with Traxxas/TQ and report.

closed 2024-04-03 by pascallanger

Eingehirner, 2024-04-04:
Is this possibly related to an issue I have with a #2218 TQ receiver? I get channels 1, 1', 2, 3, 4 to work on RRTAE but channel 5 remains inactive although I mapped it to a potentiometer. Is this also a protocol "not supported anymore"?

pascallanger, 2024-04-04:
Sorry I'm not sure I understand what you are saying...
I've used a TX6516/6517 with associated 6519 RX to reverse this protocol which only have 3 channels. But I could see somehow a 4th channel in the payload so I implemented 4 channels in the end without being able to test it.
From this compatibility matrix https://traxxas.com/support/Traxxas-TQi-Compatibility-Guide the TX2238/2240 and the associated 2218(X) RX doesn't seem to be able to work together with the TX I had but that doesn't seem to be true.
So what you are saying is that you can bind the 2218 but you get only 4 channels.
When I look at the payload, there is "space" for a total of 6 channels:
CH1 goes to bytes 2/3, CH2 goes to bytes 4/5, CH3 goes to bytes 6/7, CH4 goes to bytes 12/13
As you can see bytes 8/9 and 10/11 are free.
What I can do is to add a channel on 8/9 or 10/11 to see which one is CH5. The order is just strange CH1, CH2, CH3, CH5/6?, CH5/6? and CH4...
Is there a TQ receiver with 6 channels?
Are you able to test if I modify the code?

Eingehirner, 2024-04-04:
**> So what you are saying is that you can bind the 2218 but you get only 4 channels.**

Yes, that is correct. Binding worked like a charm. And I am happy to test it if you modify the code. I can also check if for some reason CH5 is not in the first five outputs but somewhere else - if that is implemented already(?).

I find the order strange anyway. I would have assumed that (different to how Traxxas does it) the order would follow the AETR (or rather swapped, i.e. TRAE) sequence. But it's basically AETR backwards. But as long as it's working... Perhaps it's somewhat historical, seeing that at least in my Summit CH1/2 are "basic" (steering, throttle), CH3 is switching gears and CH4/5 are controlled by ONE switch to drive BOTH differential lockers. Perhaps they had planned to use only one servo for that and it didn't work out. Still, strange.

Do it see this correctly, there is NO way I can have separate control over CH1 and CH1'? They are linked inside RX, not TX?

**> Is there a TQ receiver with 6 channels?**

I'm not aware of one. A quick search didn't bring up one.

pascallanger, 2024-04-04:
Please test: https://downloads.multi-module.org/latest-test/
I'm now sending 6 channels and I've removed any remapping.
So with this new version CH1 should be CH1, CH2->CH2 but above CH3 I have no idea of the order.

Eingehirner, 2024-04-04:
Fantastic! Worked like a charm! Thank you! The channel order is as expected, AETR for Ch1-4. Potentiometer S1 now controls Ch5.