pascallanger/DIY-Multiprotocol-TX-Module #913: Traxxas TQi

androco, 2023-12-11:
Hello

Is it possible, to add traxxas TQi ?

Would be great to use my traxxas cars on Radiomaster MT12-)

Thanks

pascallanger, 2023-12-22:
Yes it could be done as it is based on a cyrf6936 RF chip.
But I need at least a TX and RX.

williefeb19, 2023-12-23:
@pascallanger I am going to be getting one of these radios soon. I am willing to help out is there any way I could capture the data you need and send it to you if no one is willing to send you a TX & RX? I have the equipment to capture SPI and I2C data. I would just need to know what types of communication (just TX on or just RX on or both) you need and do you need data captured from both the TX and RX or just the TX.

Regards

pascallanger, 2023-12-23:
If you can capture the spi bus of the cyrf6936 on the TX that would be perfect.
I would need a bunch of captures: in bind mode with a RX in bind mode, in normal mode steering, throttle, aux...

patman13, 2023-12-23:
Hey! That would be great if the Multiprotocol would support Traxxas TQi! With the MT12 a new world of car will be covered.
Of course you are working on the basics but what I would like to add is to support the following features:
- support of telemetry; the new ESC in the cars do integrated measurements and transmit them to the receiver; this data is send to Transmitter via telemetry protocol. At least having the Battery status (voltage) would be great
- some settings are don in the transmitter; most of them could be done with mixing inside transmitter; some of them have to be sent to the receiver: stability control setting (gyro setting); traction control setting. Both are essential if we want use the Multiprotocol. 

Find some details in the manual: https://traxxas.com/sites/default/files/89086-4-OM-EN-R01.pdf

williefeb19, 2023-12-30:
I pulled my controller apart and found that it is not based on the cyrf6936. I found a CC2640 and CC2592. Would the CC2592 be compatible with the CC2500 that is in the 4in1 module? The model controller I have is the Traxxas TQi 6529 if I am not mistaken. I have included a picture of the RF section.

[image]

pascallanger, 2024-01-03:
They are emulating the cyrf6936 with this component. The cc2500 can't do it as it is limited to 500kbps max where the cyrf uses 1mbps.

pascallanger, 2024-01-03:
I'm looking at the TQ protocol first (other issue opened) and will look into TQi next.

jakezilla, 2024-01-05:
I'm also willing to help. This is what one of my TQi radios has internally.

[image]

I've got 3 or 4 TQi RX/TX pairs, and probably ha half dozen TQ sets. I've got a BusPirate too, or if there is a different SPI sniffer you prefer, I'd be happy to buy one of those.

pascallanger, 2024-01-05:
@jakezilla Great!
Let see what we can do with the hardware you have. Can you do a SPI dump from power on of a TQ radio binding a 6519 RX ?
Have you tried my latest code to bind your numerous 6519 RXs?

williefeb19, 2024-01-06:
The controller I have does not seem to have a separate MCU to be able to sniff the SPI bus. So, I don't know how much help I can be unless there is another way to get the data needed.

jleftbrane, 2024-02-07:
being that this is one of the most popular surface crawler it would be amazing if we could get this working
I have extra receivers and transceivers. 
has anyone sent the data??

SomeBdyElse, 2024-02-24:
@pascallanger I am also interested in this new protocol and I see that you did some work on TRAXXAS related features. Do you have all the TXs and RCs that you need? Would a bounty of 100 € be welcome?

pitts-mo, 2024-03-03:
@jakezilla what model TX and FCC ID is on the TX from your pictures and do you have a 6533 RX to sample with? (6533 appears to be the most advanced TQi RX).

@pascallanger I did attempt Traxxas TQ binding on a 6533 RX using my TX16S and a surface build from earlier this week (https://downloads.multi-module.org/latest-test/). It appears that when the RX enters binding mode if causes the TX to exit binding but the RX LED continues to flash and does not function after power cycle. 

My Traxxas TX is similar to @williefeb19 's and my RX similar to @inventor7777 's shown in #927 . These both appear to use the CC6240 MCU/RF chip so I don't think we would be able to attempt decode from these without $$$ RF equipment.

I have:
Traxxas model 6528B -2 channel TQi TX w/ FCC ID: XVE-TRX1041
Traxxas model 6533 -5 channel TSM RX w/ FCC ID: XVE-TRX1029

inventor7777, 2024-03-06:
I wonder if it is possible to get data about the CC2640 RF chips off of the FCC filings/patents?

jleftbrane, 2024-03-06:
I have RX/tX I'm not currently using it someone have the tech I'm happy to ship if the cost is t crazyThank you Jleftbrane

pascallanger, 2024-03-06:
The cc2640 is used to emulate the cyrf6936. It's an all in one chip.
To work on this profil I need to get my hands on an old TQi TX which still has a cyrf6936.

jleftbrane, 2024-03-06:
Is there a model number I have 3 or 4Thank you Jleftbrane

pascallanger, 2024-03-06:
No idea, you need to open the TX and look at the RX chip.

jleftbrane, 2024-03-06:
Gotcha, I may find time today!Thank you Jleftbrane

pitts-mo, 2024-03-09:
It appears that only TQi TX with FCC ID: XVE-TRX1017 are using the CYRF6936 chip. 

Below is a summary of Traxxas LP (XVE) FCC filings I found related to TQ and TQi TX/RX. 
Filed/Last_Mod,FCC_ID,Device Detail
2023-06-15,XVE-TRX1041,TQi TX 6528B 6529B 6530B 6529A (only integrated MCU/RF?)
2020-02-17,XVE-TRX1029,TQi RX 6533B 5-ch TSM
2020-02-17,XVE-TRX1028,TQi TX 6530B (STM32 and integrated MCU/RF?)
2018-08-08,XVE-TRX1024,TQi TX 6530B (STM32 and integrated MCU/RF?)
2018-08-06,XVE-TRX1025,TQi RX 6533B 5-ch TSM
2018-07-16,XVE-TRX1026,TQ/LaTrak RX 6519B 3046B 3-ch?
2018-07-12,XVE-TRX1023,TQ/LaTrak TX 6517B 6516B 3047B
2016-04-14,**XVE-TRX1017,TQi TX 6530 6530A 6507R 6529A 6528 6528A 6509R (STM32 and CYRF6936?)**
2012-11-16,XVE-SA12163,TQ TX 6516 6517
2011-01-06,XVE-SA09218,TQ RX 2218 5-ch

Warning: FCC filings can be updated and often contain poor quality images and misleading information. This information should be considered incorrect and incomplete. Please verify details independently.

ANiMOSiTYza, 2024-03-09:
Hi,
Is there a tutorial or something that we can follow to help contribute to all these new requests?
I can code (including basic microcontroller stuff like Arduinos), I know how to use an oscilloscope and a logic analyzer, but I wouldn't know how to get started on this RF emulation stuff. Is there a guide so that the community can help?

I have a couple of Traxxas TQi radios. Maybe I can send out one to you as well. I have a Model 6528 from 2016 (And the latest 6530B which seems to not be useful here) but would prefer to hold onto it if its useful to this project 9if there's atutorial I can follow), or alternatively I could ship it out to you, Pascal.

M-i-c-h-a-e-l1, 2024-03-09:
XVE-TRX1024,TQi TX 6530B 

[image]

pitts-mo, 2024-03-09:
Yes @M-i-c-h-a-e-l1 , it would be helpful to heat the solder joints and carefully pry each side clear to remove the shield in order to verify what chips are underneath. Don't hesitate to ask questions.

Entering Grantee Code: "XVE" and Product Code: "-TRX1024" at https://www.fcc.gov/oet/ea/fccid yielded internal images that can not be utilized to properly identify two chips under the can. Best guess is this FCC ID contains a CC MCU/RF and a CC LNA chip which is not ideal to snooping.

pitts-mo, 2024-03-09:
@ANiMOSiTYza
I do not know of a single tutorial but searching the internet you can find some pieces of how someone decoded/determined bits and pieces of other RF protocols. But just trying something usually offers some learning :-).

To help add Traxxas TQi support to MTM:
-verify the FCC ID on your 6528 TX 
-crack it open to verify the RF segment?

-obtain a cheap logic analyzer? so we can talk you through hookup and sampling communication between the STM and CYRF chips. OR Send your 6528 TX to someone else for sampling?

My  generic  "DIYUSER USB Logic Analyzer 24MHz 8 Channel 24M/seconds" Logic Analyzer cost less than <$10 USD shipped from aliexpress.

M-i-c-h-a-e-l1, 2024-03-10:
Updated my post of 09.03.2024 with a picture of the RF section of the radio.

pascallanger, 2024-03-10:
Too recent...

pitts-mo, 2024-03-10:
> Updated my post of 09.03.2024 with a picture of the RF section of the radio.

Nice picture. Thank you @M-i-c-h-a-e-l1 for taking the time to verify. -p

pitts-mo, 2024-03-10:
To help us proceed I have marked up @jakezilla 's image with the intent of using the larger pads at the STM for sampling. I have reasonably high confidence about the items in purple. I think we still need to figure out sample points for SS and IRQ. I can't even guess if XOUT or PACTL are helpful. And the STM pads flagged orange are ones I found interesting and would like to trace next.

[image]

pascallanger, 2024-03-10:
All we need to care about is mosi miso sck ss

pitts-mo, 2024-03-16:
I feel this protocol will have a powerful draw and the potential to encourage many more users to help branch out our hobbies.

Please indicate if you have a Traxxas TQi TX with FCC ID: XVE-TRX1017 and are willing to sample data from it. 
(some of the TQi TX with model numbers 6530, 6530A, 6507R, 6529A, 6528, 6528A, 6509R may have FCC ID: XVE-TRX1017)

If no one indicates they are up for this task in the next ~1-2 months I would like us to identify someone that can commit to shipping a TX/RX set (preferably 6507R >=4 channel TX with FCC ID: XVE-TRX1017 and a 6533 TSM RX) to someone able to perform the needed testing and sample collection.

M-i-c-h-a-e-l1, 2024-03-16:
I am trying to get hold of one of these TRX1017 radios, but without any luck so far ... If I am able to get my hands on one of these radios I will either try to sample the protocol or to send the radio to Pascal.

jleftbrane, 2024-03-16:
What's the shipping address?  Can you ship back after or will in be garbageThank you Jleftbrane

jleftbrane, 2024-03-16:
Thank you Jleftbrane

pascallanger, 2024-03-16:
Shipping will be to France. I can ship the TX+RX back when done.

jleftbrane, 2024-03-16:
What's the address I'll shipThank you Jleftbrane

pascallanger, 2024-03-16:
@jleftbrane Which TX do you have? Are you sure it has a cyrf6936 RF component? I would need a RX too. If all is yes then you can contact me by email pascal _ langer @ yahoo . fr

jleftbrane, 2024-03-16:
Sent pics last emailThank you Jleftbrane

pascallanger, 2024-03-16:
Sorry I can't see any pics from you on github...

jleftbrane, 2024-03-16:
You got Facebook? Thank you Jleftbrane

jleftbrane, 2024-03-16:
Or WhatsApp?Thank you Jleftbrane

onki69, 2024-03-18:
I have a TQI transmitter and receiver (taken from my TRX-4 car) lying around.
If required I can send it to you for testing etc.
Please let me know if you want to use the equipment. I don't need it at the moment.

M-i-c-h-a-e-l1, 2024-03-22:
@onki69 Please check on the rear of your radio if there is a sticker with mentioning XVE-TRX1017. If the number is different from XVE-TRX1017, then the radio can not be used for the purposes.

M-i-c-h-a-e-l1, 2024-03-30:
Dear onki69,

Thank you for checking. Unfortunately your radio cannot be used.

Kind regards
Michael

Eingehirner, 2024-04-03:
Hi all,

I just joined this conversation as I'm very interested in seeing both TQ and TQi fully(?) supported. I have no TX/RX with TRX1017 - but would one with XVE-SA09181 (TQ) help as well? I also have a TQi with another -SAxxxxx label.

Cheers
Simon

Eingehirner, 2024-04-03:
Hi Pascal,

unfortunately my TX is a model 6513-4-5, 6507-8-9. It's apparently older than _any_ of the models in your list above. Would this still help you in any way? I see a 6507**R** in the list but I'm not sure if that's identical. The FCC label says XVE-SA10044 (also not present in your list).

Cheers
Simon

pascallanger, 2024-04-04:
I'm not the author of that list and I have no idea about it. Please open your TX and just look at the chip near the antenna connection. There is a big change that it will indicate cyrf6936. If that's the case then yes this is what I'm looking for.

pitts-mo, 2024-04-04:
I collected the listing of info based on the current FCC filings. But there
are many reasons that can make the info in my list incomplete/incorrect.

Please attempt to verify the chips in your TX and I will update the
list accordingly.

Eingehirner, 2024-04-04:
[image]

@pascallanger Yesss, it has the right chip it seems. Do I see this correctly, I should leave the shield off for you to access the chip(s), not solder it back on? Can you PM me the shipping address I should send it to?

@pitts-mo The TX has the part no. **6513-4-5, 6507-8-9** (I don't know what to make of those two numbers, that's how it's written on the silver type label together with the serial number). The FCC ID on TX foot and also on the PCB is **XVE-SA10044**. The board is a **TRX_328_XMI-1_PCB Rev. 4**. The RX for it has part no. **6518**, FCC ID **XVE-SA10046**.

pascallanger, 2024-04-04:
I don't know how to find your email address so here is mine: pascal _ langer @ yahoo . fr
I would need the TX and a compatible RX to work on this protocol. And yes don't put the cover back, I need to solder on this little pins to spy the SPI bus...

pitts-mo, 2024-04-05:
Sorry, I had excluded XVE-SA10044 and XVE-SA10046 from my original search results because I could not find reference to TQ or TQi within the FCC filing documents.  I have updated my FCC ID list above to include these.

I also now recall chuckling when I first heard of TQi because it seemed like they were trying to piggyback on another company's marketing hype.

Thank you @Eingehirner , 
-p

Eingehirner, 2024-04-12:
@pascallanger Did you manage to find a cyrf6936-based TQi in your vicinity?

lis-uav, 2024-04-21:
I have just checked all 6 of my TQi transmitters, and they are all the TRX1024 ones.  I'm really keen to help out get the TQi protocol supported, so the only thing I can think, is offer a donation to help?  Is that a thing that would help?

inventor7777, 2024-04-23:
I just came back to this thread after a couple months and I'm happy to see all the support and progress! I'm also glad @Eingehirner has the correct version of RX too.

inventor7777, 2024-10-05:
Any updates on TQi support?

lis-uav, 2024-10-13:
I've managed to find one of the right transmitters according to the list posted above.  I'm not 100% sure of what's needed next, does anyone have any step by step instructions?  I have access to oscilloscope, logic analysers, psus etc in the electronics lab where I work.

Alternatively, I'm happy to post it off to someone who knows what they are doing!

pascallanger, 2024-10-13:
Connect a logic analyzer on the SPI bus of the CYRF and post the result of a power on. Not sure which logic analyzer you might have so a link to the software to open your file might be useful.

Knock82, 2025-03-15:
Any News? TQi would be GREAT !!!

WZ9V, 2025-04-27:
Any updates?  Would love to see TQi with telemetry supported.

inventor7777, 2025-05-19:
@Eingehirner sweet! Looking forward to it

Eingehirner, 2025-05-24:
[CYP_Logging.txt](https://github.com/user-attachments/files/20421868/CYP_Logging.txt)
@pascallanger I have collected the data and from a first look it seemed OK (sort of, with my limited knowledge). Can you make something of it? It's collected with sigrok-cli.

[19 more replies]