pascallanger/DIY-Multiprotocol-TX-Module #916: DSM-Auto Does not bind with Lemon-RX G2 Receivers

frankiearzu, 2023-12-15:
If you try to bind one of the new LEmon-RX G2 Receivers (models LM0086 / LM00087) using DSM-Auto, they don't bind.
If you use DSM-X2F it does work.

---- After some research and added more debug messages to the code -------

The cause of "Auto" not working is the MM is not receiving the 10-byte BIND-Response message that will switch "Auto" to the actual sub-protocol.  It does work and gets received in genuine Spektrum TX.

The problem seems timing... we are re-transmitting the "Bind" message for too long (about 3s), and looks like the RX timeouts waiting for a quiet period of time.    Reducing the bind re-transmit timing to about 1.5s solves the problem.

Looking at the code comments, looks like that was the intention, but the code was not doing the "/2".

```
//During binding we will send BIND_COUNT/2 packets
//One packet each 10msec
#define DSM_BIND_COUNT 300

if(IS_BIND_IN_PROGRESS)
	{
		DSM_initialize_bind_phase();		
		phase = DSM_BIND_WRITE;
		bind_counter=DSM_BIND_COUNT;   // Farzu:  adding the DIV 2 seems to work not with all RXs
    debugln("Bind Started");
	}

```

### ORIGINAL TIMING:  3s transmitting Bind, 4.2s waiting for Bind Response

```
------- Lemon-RX, original DSM_BIND_COUNT=300:   3s of Sending BIND message, 4.2s of waiting
00:18:07.203 -> Protocol selected: 6, sub proto 4, rxnum 1, option 12
00:18:07.303 -> Bind Started
00:18:07.303 -> Proto=DSM, nbr_sub=6, Sub=Auto, Opt=7, FS=0, CHMap=1, rfSw=3
00:18:10.314 -> Bind Check
00:18:14.530 -> Bind Read TIMEOUT

------- Spektrum AR637, original DSM_BIND_COUNT=300:   3s of Sending BIND message, 4.2s of waiting
00:22:53.386 -> Protocol selected: 6, sub proto 3, rxnum 1, option 8
00:22:53.486 -> Bind Started
00:22:53.486 -> Proto=DSM, nbr_sub=6, Sub=Auto, Opt=7, FS=0, CHMap=1, rfSw=3
00:22:56.497 -> Bind Check
00:22:56.551 -> Bind 0A 14 3B 1A 01 48 B2 01 02 DF
```

### NEW TIMING:  1.5s transmitting Bind,  4.2s waiting for Bind Response

```
------- Lemon-RX, DSM_BIND_COUNT=300/2 for sending:   1.5s of Sending BIND message, 4.2s of waiting
00:31:29.365 -> Protocol selected: 6, sub proto 4, rxnum 1, option 8
00:31:29.466 -> Bind Started
00:31:29.466 -> Proto=DSM, nbr_sub=6, Sub=Auto, Opt=7, FS=0, CHMap=1, rfSw=3
00:31:30.971 -> Bind Check
00:31:31.018 -> Bind 0A 14 3B 1A 01 07 B2 00 02 9D

------- Spektrum AR637, DSM_BIND_COUNT=300/2 for sending:   1.5s of Sending BIND message, 4.2s of waiting
00:29:20.748 -> Protocol selected: 6, sub proto 3, rxnum 1, option 8
00:29:20.848 -> Bind Started
00:29:20.848 -> Proto=DSM, nbr_sub=6, Sub=Auto, Opt=7, FS=0, CHMap=1, rfSw=3
00:29:22.354 -> Bind Check
00:29:22.401 -> Bind 0A 14 3B 1A 01 48 B2 01 02 DF

------- NO-RX connected, DSM_BIND_COUNT=300/2 for sending:   1.5s of Sending BIND message, 4.2s of waiting
00:33:59.136 -> Protocol selected: 6, sub proto 3, rxnum 1, option 12
00:33:59.236 -> Bind Started
00:33:59.236 -> Proto=DSM, nbr_sub=6, Sub=Auto, Opt=7, FS=0, CHMap=1, rfSw=3
00:34:00.788 -> Bind Check
00:34:04.956 -> Bind Read TIMEOUT
```

pascallanger, 2023-12-15:
I'm afraid that if I've removed the /2, it was because there was issues with some receivers at one point. I'm fine to reduce the timing for the lemon RXs but it might break for some other brand or older receivers...

frankiearzu, 2023-12-16:
Seems that with 1.8s works fine with all RX..  with 1,9s was a 50% Auto success with Lemon-RX.
As a "Get out of Jail", if you don't use Auto, is the original 3s, that seems to bind even when not receiving the Bind-response package.

PR #918 (2023-12-18): Fix for DSM-Auto Bind on Lemon-RX Gen2
Fixes #916

Tested with multiple brands (Spektrum, OrangeRX, Lemon-Rx, Admiral).  Even some old DSM 2 receivers.
Lemon-RX needed a short Bind Send period (180 count, about 1.8s instead of 3s)

All brands works with "Auto" now.   Just in case one needs a LONG bind sending period, when not using "Auto" works the same as before.
closed 2023-12-18 by pascallanger