pascallanger/DIY-Multiprotocol-TX-Module #921: Joysway J4C12R - no outputs on channels 2 and 3

mshagg, 2023-12-18:
Joysway J4C12R receiver, which are included with RTR Joysway boats.  Binds successfully using the Joysway protocol.

However, there is no signal output on pins 2 and 3 on the receiver when outputs are mapped to these channels in EdgeTX.

Channels 1 and 4 work as expected. 

MPM v1.3.3.14 AETR
Radiomaster MT12 EdgeTX "RM Factory" build

pascallanger, 2023-12-18:
Are channels 1 and 4 of the TX map to channels 1 and 4 on the RX?
Is there really no output on the channels 2 and 3? If you plug a servo on them, will they center and hold their positions?

mshagg, 2023-12-18:
> Are channels 1 and 4 of the TX map to channels 1 and 4 on the RX? Is there really no output on the channels 2 and 3? If you plug servos on these 2 channels, will they center and hold their positions? Test the center by moving by hand the servo arm before plugging the servo, check that it is nearly centered when compared to a center value on 1 or 4. Hold means that if you try to move the servo arm by hand while plugged in a channel it will resist and don't let go.

Yes - you need to map outputs to channels 1 and 4 in order to have them output on pins 1 and 4 on the rx.

Anything mapped to channels 2 and 3 in edgetx have no output on the receiver.  There is no signal at all so far as I can tell - an ESC plugged in to these channels will beep continuously, and servos show no sign of holding position.

pascallanger, 2023-12-22:
Really no signal at all on channels 2 and 3, the servo is not holding its position if you move the arm?
If that's really the case then I'm going to be stuck on what to do next... Can you solder an analyzer on the SPI bus of the A7105 on your original TX?