pascallanger/DIY-Multiprotocol-TX-Module #924: DSMR Support

bitogre, 2023-12-29:
Given RadioMaster just releases a ground radio with the 4-in-1 module, is there a plan to support DSMR?  I would assume it would be a sub-protocol of DSM for surface receivers.  I am specifically interested in using the RadioMaster MT12 with a Losi Promoto-MX Motorcycle and my Horizon boats.

bitogre, 2023-12-29:
Found [this video](https://www.youtube.com/watch?v=OfxvGxojPYc&t=918s) that looks like DSMR is supported but that there is an issue with the Losi Promoto-MX Motorcycle receiver.  Is work being done to fix the issue?

pascallanger, 2023-12-29:
DSMR is already supported.
You are referring to a video where people commented that it works for them so most likely a bad bind or something he did... This video is really full of none sense with many things not working but in fact from someone who doesn't know how to use the radio obviously...
What do you exactly want?

bitogre, 2023-12-29:
What I want is the RadioMaster MT12 with the 4-in-1 module (when I get it - ordered but not yet received) to work reliable and not loose binding with the Losi Promoto-MX Motorcycle receiver.  The one video I saw and linked to above does not have the comments of people saying it worked for them.  Maybe that is a different video (or maybe a reply I didn't see).  If you are saying it will work and keep bind with this receiver, then close this issue.  I can re-open if I run into the same issue when I get my MT12.

pascallanger, 2023-12-29:
I do not own such a model and I can't test/fix anything without getting my hands on it. So there is no point of requesting just in case. If you run into the issue then we will see but I will probably need the RX to be shipped to me in France to see what I can do.

bitogre, 2023-12-29:
If I run into the issue, I will be happy to chip in the $60 US cost for you to buy the receiver.  Thanks.

pascallanger, 2024-01-09:
I've received the confirmation that only the SR6300PM DSMR receiver is showing this specific behavior. Without the RX to test it will be hard to work on it.
Some trials:
 - use the latest test version of multi which has modified bind timing
 - launch the forward programming script to see how the RX behaves

bitogre, 2024-01-09:
@pascallanger Ok, if you re-open this issue and tell me what it would cost for you to buy it locally, I will send you the cost so you can buy this receiver and work on this issue.

pascallanger, 2024-01-09:
It loos like it's not only a Multi issue: https://www.arrmaforum.com/threads/having-trouble-keeping-the-binding-between-dx5c-and-sr6300pm-receiver.63720/ . Too bad the guy has not posted anymore... I'm wondering if it could be that some channels have to be set to some specific values to be accepted or have enough range.

pascallanger, 2024-01-09:
Reading through the SR6300PM RX manual it looks like the TX must have enough range on throttle and steering for the calibration to be successful. The calibration happens each time you bind the RX. So my guess is that you must set the "Enable max throw" in the model config and may be even use "Extended limits" at 125% for a full range.

Gaz-B, 2024-02-01:
I have now spent quite some time trying to get a Taranis X9D+ with a JP4in1-SE Module on the latest release to keep bind with a Spektrum SR6300PM receiver in a Promoto. I Tried calibrating and in never gives the final led off and back on again to indicate completion. What seems odd to me is the stock transmitter binds and keeps bind even with calibration. The other thing is all the time the X9D connected it's loosing and recovering telemetry unls the the sticks aren't touched for a few seconds. I've tried 11 and 22ms different numbers of channels. It only binds with the receiver number set to 00. At the moment my best option seems to be remove the teletry voice from the SD and disable the haptic feed back and a hole in the bodywork to access the bind button for every battery change. If i turn the TX off and back on it relinks it's power cycling the RC when it falls over.     🙏

pascallanger, 2024-02-02:
@Gaz-B I think the calibration never completes due to the channel output range.
Can you check the box called "Max throw" which is near where you have selected the protocol DSMR. If it's not enough, in the model check "Extended Limits" and set the outputs end points of the channels you use to -125/+125%. This is explained in the DSM protocol: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#DSM---6
For the telemetry, most of the DSM models send telemetry irregularly with none useful information. This is why a check box has been added on the telemetry page 'Disable telemetry alarms".

Gaz-B, 2024-02-02:
I did try that after reading your previous comments and it didn't help. I measured the stock radio and all channels come set at 100%.the receiver manual says it has to be over 80% to calibrate.

pascallanger, 2024-02-02:
That sounds really strange. If the calibration does not complete it means that a channel somewhere is not at the value it is expecting to be. On the all in one planes board, it's usually the throttle channel which doesn't go down enough, that's where you need to enable extended limits and push it to -125%.
Can you confirm that you've tried both settings turned on at the same time "Max throw" and "Extended limits" with all the outputs set to -125%/+125%?
The problem is that the % is subjective in the DSM world, that's what is annoying. At one point they switched from 1100=-100% 1500=0% 1900=+100% to 1000=-100% 1500=0% 2000=+100%. So basically depending on the radio age you don't get the same output... By default multi is using 1100..1900 which might be below the 80%.

On the original radio, can you see the values of all the channels. Are you setting the exact same thing? For that you want to activate the feature "Disable Ch. map" which would tell the module to send CH1 on the radio to CH1 on the receiver.

Only the RX number 0 works? You've tested all 22 possibilities?

Gaz-B, 2024-02-02:
I will check again  I made sure it was 100 100 as the manual says above 80%.

pascallanger, 2024-02-02:
That's my point 100 100 doesn't mean anything neither 80%...
Please make sure to test with "Max throw" enabled (near where you have selected the protocol DSMR) and in the model config check "Extended Limits" then go to the output page and set all channels to -125 +125.

Gaz-B, 2024-02-02:
setting throttle and steering to 125% didn't help On my X9D that's 860 - 2060 us 100% is 988 - 2012. The annoying alarm has stopped though thank you for that.

pascallanger, 2024-02-02:
Once again the 860 - 2060 µs or whatever % has nothing to do with the current protocol. This is what the radio would output when using PPM nothing else.
Have you set Max Throw and extended limits? Have you set all the channels to output -125/+125%? Have you looked at the position of all the channels on the original TX and reproduced the same? Have you disabled channel mapping to make sure that you are sending the correct channels where you think they should be?

Gaz-B, 2024-02-03:
I've tried again with a new model to make sure none of the mixes i have setup for the bike are affecting it. Max throw and disable channel map ticked I've tried 125 and full max 150 end points. I've tried both 11 and 22 ms. Different channel quantities.  Different receiver Nos anything other than 00 it flashes about every 2 secs. Nothing gets the receiver to calibrate.

bitogre, 2024-02-03:
@pascallanger Did you get my email about the money I sent you with regards to this issue?

Gaz-B, 2024-02-05:
A little bit more testing no idea if it will help. I've tried with the new model both channels set to 125% and trid seeing if the order of channel to enter calibration has changed for the promoto as it's effectively a bespoke firmware but didn't help. Also the stock TX had throttle endpoints at 100/100 and steering at 125/125.

Gaz-B, 2024-02-05:
Could the auto setup that spectrum does when binding to a different spectrum tx be causing it. Seems the receiver tells the TX it's a bike and auto configures everything.

Gaz-B, 2024-02-09:
I can see it's an issue just with the promoto version of the RX. Even with Spektrum TXs they have to be flashed with V3.0 firmware or the bind isn't successful this sets everything to the bike settings and there is no manual calibration must be some hand shake to confirm it's a radio with bike setting to complete bind. Is there anyway to put my multi module on the TX into bind mode without going through the full menu holding the momentary  switch while switching the TX on would be perfect. The gains of running edgeTX outweigh the hassle of rebing for every pack by quite some margin.

pascallanger, 2024-02-28:
To hopefully solve this issue I've finally decided to purchase a full Promoto set. The RX alone is not enough I also needed the TX.

Gaz-B, 2024-02-28:
By fat the cheapest way is jennys rc they sell the tx and rx as a set for $80 brand new. They are a breakers.

pascallanger, 2024-02-28:
@Gaz-B I haven't found this bundle in France... Too late I've placed the order yesterday and it has left the facility already.
You indicated that you've done some settings on the MT12 for the motorcycle. Can you share your model or settings?

Gaz-B, 2024-02-28:
I use the Taranis X9D+ I don't do steering wheels. I will send you what I have set on that a bit later.

Gaz-B, 2024-02-28:
I took the measurements from the receiver I found the RX has to be started and not moved to do this. I used a flight stabiliser system go give me the numbers which I don't think are of any real use other than channel by channel Iset my radio to the exact same numbers.
CH 1 Steering  -75%/0/+75% this is the only channel I altered as stock the servo was binding at full travel a known issue on the bike.
CH 2 Throttle  -77.3%/0/+82.5
CH 3 Mode      -86.3% wheelie, 0% offroad, 80% on road.
CH4 Flywheel This I have my own range as I'm using a flight ESC to give variable gyro speed also the stock one has issues and regularly burns stators out.
Ch 5 Front Brake again my figures have nothing to do with the stock set up this channel is reverse thugh.
CH6 Gain -86.2%/86.2%

There are two ways of putting the RX into bind. Hold bind and switch on or what I prefer power on then press bind after the beeps have stopped this way the RX is perfectly still while calibrating.

The only issue I found running with having to rebind for every battery is in a tumble the steering servo can cause a RX brownout BEC is a but feeble.
I have resolved this by powering the steering servo from my flywheel ESC BEC not an option with the stock one.

What concerns me is even Spektrum TXs won't keep bind with the bike unless on V3.0 firmware.
The stock TX if flashed to an update the app offers over bluetooth stops the TX working with the bike.
Once a radio is on V3.0 most of the basic functionality is lost even servo reverse.
I know a couple of people have fallen foul of this one could reverse the steering on the stock TX but there's no way to reverse the gyro sense. The other changed the brake servo he was using a DX5 and couldn't reverse the channel.

Good luck

Gary

-

Gaz-B, 2024-04-24:
Have you made any progress with this issue ?.

AlphaCharlyX, 2024-12-05:
Does the Losi Promoto work with the 4 in 1 module and the MT12?

bitogre, 2024-12-15:
> Does the Losi Promoto work with the 4 in 1 module and the MT12?

This whole issue is trying to get the Losi Promoto work with the 4 in 1 module and the MT12.  I am still awaiting news of any progress being made by @pascallanger despite me sending him money to buy the RX.

Gaz-B, 2024-12-15:
I think it may be an insurmountable problem. Spektrum seems to have moved the goal posts on this one to keep the bikes locked to their radios on specific firmwares. I would be nice to hear from pascallanger on any progress or failure though.

Gaz-B, 2025-02-08:
Has there been any progress. I'm guessing not but see 1.3.4.0 is now available. I'm also trying to see prior to purchase if binding to the SLT protocol used in the Arrma Groms is an issue or not ?.

steph280, 2025-03-20:
I got a MT12 coming in and would love to be able to use it with my Promoto.  Hope we'll hear from Pascalanger soon!

okevin1954, 2025-03-29:
Gaz-b is your MT12 setup above working ok on the promoto?
Thanks,
Kevin

Gaz-B, 2025-03-29:
I don't have an MT12 but that doesn't make a difference I have a Taranis with the 4in1 module. The current state is I have to bind with every battery which to me is worth it to not have a steerwheel radio. I know more development is planed for this bind issue.

okevin1954, 2025-03-29:
Thanks, your channel descriptions will help a lot. My MT12 gets here monday and I got my 22s sprint car thursday.

Gaz-B, 2025-05-06:
Has there been any progress? as someone on a Promoto group is saying theirs is keeping bind with the MT12 but no real proof or detail given.

pascallanger, 2025-05-06:
Have you tried the latest test build? I've modified the dsmr protocol so worth a try.

Gaz-B, 2025-05-06:
Still seems the same. I see their is now DSM R 1F and DSM 2SFC. They seem to behave exactly the same with the promoto RX.

pascallanger, 2025-05-16:
The telemetry line in that case is taken by the internal module. Therefore the multi module can't tell the radio anymore about protocols that exists or not so the radio uses the protocol list it knows about. DSM telemetry can't be received anymore as well.
Only Radiomaster/jumper/... radios supports truly 2 modules running at the same time from a hardware perspective.

lschunn, 2025-05-31:
> The only issue I found running with having to rebind for every battery is in a tumble the steering servo can cause a RX brownout BEC is a but feeble.

can you explain a bit more about the servo caused receiver burnout issue on BEC, while rebinding each every battery? 
how does it happen?

Gaz-B, 2025-06-17:
Could bind on a switch be added to the DSM-R1F options ?. Putting the RX into bind on this model is super simple by flexing the left body panel after the gyro has finished initialising. So being able to bind by flicking the momentary switch would make the not remembering bind a non issue for me.

pascallanger, 2025-06-17:
You can use bind on channel. That's the purpose of it.
I'm hopping to look at the Losi this week.

Gaz-B, 2025-06-17:
I've just checked some other protocols and it's exactly where I expected it to be just below fine tune. It's not available on any of the DSM protocols at present probably never been useful before.

pascallanger, 2025-06-22:
I haven't had time to look into it.
I can't add back bind on channel since it's used by another feature...
HPI is listed: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-scx24---2

Gaz-B, 2025-06-22:
> I haven't had time to look into it. I can't add back bind on channel since it's used by another feature... HPI is listed: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-scx24---2

Wow 🤣 I'm not surprised we couldn't find it. I would never have expected it to be shared with the SCX24 as that's a Horizon product so would expect that to be DSM-R or SLT2. I'd never heard of Pelican.

pascallanger, 2025-06-24:
> Wow 🤣 I'm not surprised we couldn't find it. I would never have expected it to be shared with the SCX24 as that's a Horizon product so would expect that to be DSM-R or SLT2. I'd never heard of Pelikan. I've been trying to find a way to search the protocol list by key words but can't find a way this would be a very useful tool.

Open the protocol page and then do a text search using your browser : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md
I'm trying to add Manufacturer, TXs and RXs to that page but it's hard to keep up.

Gaz-B, 2025-06-24:
> I've been spending some time on the Promoto's RX but I don't see why it doesn't save the bind information.
> The bind process seems to work fine and terminate normally. I've tried to match the bind packet being sent to be exactly like the DX3 with no luck.
> I've looked at the normal packets that the DX3 sends and I don't see anything special compared to what the current DSMR implementation is doing. There is one difference but I've tried to add it in my code and it doesn't change anything.
> I'll keep trying but I don't know where to really look at right now. The problem is that I don't have access to the SPI bus on the DX3 (too recent with an all in one chip) so I need to listen over the air which is complicated with the cyrf. If someone has an old radio where the cyrf6936 is accessible, I could make that work easily.

Not the most concise video but may be of some help showing other pre Promoto Spektrum radios need to be flashed to V3.00 to keep bind with the bike. I've tried both 11ms and 22ms as in the video He states it has to be 11ms but makes no difference I can power cycle the TX and it reconnects fine but if the RX gets cycled it's needs to be bound. Plan B could you make an alternative file to download with bind on channel with or another DSM option R 1FB or whatever.    https://www.youtube.com/watch?v=NSkg0JtQ5Wc
[https://www.bing.com/th?id=OVP.FH08YbcnYrALl6SMgMwlUQEsDh&pid=Api]<https://www.youtube.com/watch?v=NSkg0JtQ5Wc>
How To Bind Your Losi Promoto MX to DX5r-Dx5Pro-Dx5c-Dx5 Rugged<https://www.youtube.com/watch?v=NSkg0JtQ5Wc>
How to Setup your Losi on a Dx5r Dx5Pro Dx5c and dx5 rugged. If you have any questions feel free to leave a comment #losi #losipromotomx
www.youtube.com

Jetman1, 2026-10-22:
Does in the meantime anybody found a solution for the lost binding when binding the multi-module with the promoto? The bike works great with my new Frsky VX10 running Ethos in combination with the multi-module, only the binding problem ist a little bit boaring.

Jetman1, 2026-11-07:
> Does anyone have LMT 2.0 with SPMSR6400A receiver to try binding with MT12? to confirm will the binding stay? Kevin Talbot made an interesting video to modify the Promoto with LMT 2.0 receiver https://www.youtube.com/watch?v=fGXNfcscda0&t=723s

😵‍💫