pascallanger/DIY-Multiprotocol-TX-Module #930: FlySky ANT Protocol

theCommaLlama, 2024-01-18:
It would be really awesome to have support the FlySky ANT protocol! I'm happy to connect up using an st-link or dumping something from the SPI bus if it would be helpful

fiechr, 2024-01-29:
It seems ANT is based on AFHDS 2A (not AFHDS3), but uses "upgraded hardware":
https://www.flysky-cn.com/enant

jleftbrane, 2024-02-07:
this would be a great protocol to have use of with the MT12 4in1

DonKracho, 2025-01-28:
very interested too in teh FlySky ANT protocol

pascallanger, 2025-01-28:
I would need someone to dump the SPI bus of the RF chip to see what it looks like.

puneet00, 2025-01-28:
How do I dump this from the receiver on my fms fcx24m ?
It would be great to have multi module support for FMS receivers

DonKracho, 2025-01-29:
I've got an unused RE3LY Gen7 car remote. I'll try to open it. May be I find a way to attach cables to the SPI bus of the TX chip to a simple 8CH USB Logic Analyzer (using the salease software). May take a little time.

pascallanger, 2025-01-29:
Saleae is perfect. They should use a A7xxx RF chip, that's where you should hook to the SPI bus.

pascallanger, 2025-02-26:
Any news from a SPI dump?

inventor7777, 2025-03-12:
@DonKracho any news from the SPI dump?

DonKracho, 2025-03-13:
Sorry not yet. To many other projects.

CGPro856, 2025-08-17:
Subscribed. Definitely looking forward to this protocol.

raphaelcoeffic, 2026-11-10:
AFAIK, the main issue is that the Amiccon chip is driven at 4Mbps, which is fairly unique. None of the current chips on the MPM can do that, nor any of the replacement RF MCU I'm looking at right now (CC2652, CC2755).