pascallanger/DIY-Multiprotocol-TX-Module #936: Please provide multi module builds with no channel reordering at all

aheyer, 2024-02-05:
I'm coming from a Devo 10 with DeviationTX and internal 4-in-1. I'm trying to reimplement my models on a Radiomaster Zorro with EdgeTX and internal 4-in-1. When programming an Eachine H8S 3D with Silverware firmware I discovered that channel mapping not only occurs on the first 4 channels but also on later control channels. E.g. channels 5 and 6 in DeviationTX are now channels 6 and 10 in the mixer of EdgeTX.

In Deviation it's my task to provide the correct channel order so I'm very unhappy with the concept of EdgeTX and this implementation of relying on a fixed channel order like AETR or TAER. I'd like a module build that simply outputs EdgeTX channel 1 to protocol channel 1, EdgeTX channel 2 to protocol channel 2 and so on. There should be no mapping/reordering from e.g. EdgeTX TAER order to protocol order AETR.

Or at least please enable the "No channel remapping" option on all protocols!

pascallanger, 2024-02-05:
The channels order is just a convention, nothing else. Deviation and Multi are not using the same channels for the same functions and there is nothing I can do about that. You have to configure the models according to the doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#available-protocol-table-of-contents-listed-alphabetically
This lua script might help you as it names the channels used by the current protocol: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/Lua_scripts#multichannelsupdater

closed 2024-02-05 by pascallanger

aheyer, 2024-02-05:
But since Deviation hasn't a concept of (fixed control) functions it doesn't shift channels between UI and protocol, does it? There's no concept like AETR or TAER so UI mixer channel 1 goes to protocol channel 1 and so on. Whereas you're shifting channels around – my EdgeTX is set to TAER and I have to flash the multi-module build TAER accordingly. On the first try I don't mind that and set the mixers/output channels for the model to AETR order. The same as on my Devo. But the interface between EdgeTX and the internal 4-in-1 relies on TAER so everything was mixed up.

So of course, you can do what I like to get. I don't mind if you don't want to make an additional build but please enable the "No channel remapping" option on all protocols! And I know this LUA script and used it already. But since I'm using a custom firmware on the quad the original channel function naming of Bayang isn't suitable anymore. I can live to adapt to the first 4 control functions TAER/AETR but why does shifting/remapping occurs on channel 5 (GEAR), 6 (AUX1) and higher?

pascallanger, 2024-02-05:
You don't understand... There is no remapping on channels above 4, like there is no remapping on most of the protocols below 4 unless it's indicated (and you can disable it).
You can do the same on OpenTX and EdgeTX as devo and assign channel X to whatever you want.
Channels above 4 are whatever I assigned them while writing the code. Since the module can be used on PPM radios with limited number of channels, the most important features have a low number that's basically the rule.
When you have one bit in a packet for a feature, you don't have a channel associated, you just choose arbitrary whatever you want... This is what I've done and there is no compile option to be added for something purely arbitrary.
You have to understand that the multi code is not the deviation code.