pascallanger/DIY-Multiprotocol-TX-Module #937: Option value is missing on some protocols in EdgeTX

aheyer, 2024-02-05:
I'm binding an Eachine H8S 3D with Silver13 firmware to a Radiomaster Zorro with internal 4-in-1 and EdgeTX 2.9. According to the [multiprotocol manual](https://www.multi-module.org/using-the-module/protocol-details/bayang) there is an option value for the Bayang sub protocol "Standard". In EdgeTX I can't see and modify this option to value 1 to use Silverxxx firmware telemetry. Or is this option (and the manual) outdated because I'm already getting telemetry data even without setting the value?

There were similar problems with other protocols [in EdgeTX](https://github.com/EdgeTX/edgetx/issues/2713). But the missing UI update in EdgeTX should be fixed now?

pascallanger, 2024-02-05:
The documentation I keep updating is located here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#bayang---14
EdgeTX is having a few bugs to display the correct fields that the module request. But OpenTX works flawlessly.
You should open a new issue on their github since on the multi side everything is working as expected.

aheyer, 2024-02-06:
EdgetTX shows me the option labelled as "Telemetry" with the values "Off", "On", "Off+AUX" and "On+AUX". This might be easier for some users but for me somewhat untechnical because not a direct mapping of the multiprotocol documentation.

pascallanger, 2024-02-06:
I've updated the doc with this information.