pascallanger/DIY-Multiprotocol-TX-Module #938: Evaluate Radiomaster RC3S protocol

pwkpete, 2024-02-05:
Per our conversation on rcgroups. Here are the images from the radio to evaluate if it may even be possible to implement the earlier Radiolink ground protocol: https://www.radiolink.com/newsinfo/664430.html

[image]
[image]
[image]

pascallanger, 2024-02-05:
The RF chip is under the glue so we don't know what it is...
The SPI bus is well exposed and can be connected to an analyzer.
Do you want to purchase and connect an analyzer (around $10) to your radio or proceed differently?

pascallanger, 2024-02-05:
A small 8 ch Saleae analyzer (https://www.ebay.fr/itm/262812296176) connected to MOSI, CLK, MISO, CSN and GND.

pwkpete, 2024-02-05:
That's funny, I happen to have one of those right in front of me, though I
have never used it...

I can solder that up, any particular software to use?

I assume, record from power on to bind?

pascallanger, 2024-02-05:
Please use the logic software v1.2.14
I would need different capture files for:
- power on
- power on + bind
- throttle mid->low->mid->high-mid
- steering mid->low->mid->high-mid 
- same for any other channels

pwkpete, 2024-02-06:
[RC4G.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14174599/RC4G.zip)

Can you see if these are what you need, or if I did it correctly?
CSN = Chan3
MISO=Chan4
CLK=Chan2
MOSI=Chan1

pascallanger, 2024-02-06:
Sorry but the sample rate is too low, the files are unusable as they are... You need to increase it quite significantly (*4 or *8).
GND is well connected between the GND of the radio and the GND of the Saleae, right?

pwkpete, 2024-02-06:
Yes ground is connected. Will up sample rate and try again. First time with
this so learning how to use it correctly. Thx!

pwkpete, 2024-02-06:
Here are the files a 8MS/s, let me know how these work. I did not initiate a bind to new RX yet, want to make sure this works so far.
[8MS per s.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14187129/8MS.per.s.zip)

Thx!

pascallanger, 2024-02-07:
Getting there, but still not fast enough... You should see 8 pulses of clock per byte, there are only 5-6 right now.
If you add a SPI analyzer (on the right side) and configure the inputs (MOSI, MISO...) with everything else by default, it should be able to decode which is not the case right now.

pwkpete, 2024-02-07:
Ok, tried again at 24MHz
SPI analyzer finds stuff, but the clock has me confused - I'd expect to see a consistent pulse
[Power On.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14195637/Power.On.zip)

pascallanger, 2024-02-07:
It works this time. I need to find out which component they are using now.

pwkpete, 2024-02-07:
[Scans @ 16MHz.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14195827/Scans.%40.16MHz.zip)
Awesome, I tried the other ones you requested at 16MHz - hope they work as well

pascallanger, 2024-02-07:
It looks like they are using a CC2500. The startup or bind phase looks strange from a first look so I'm wondering if it could be another CC2xxx. I need to spend more time on it.

pascallanger, 2024-02-07:
Now that I've removed the excessive polling that they are doing, I can see that what they are doing at the power up is to find free channels. Once done they start to send data using 2 of these channels every 14.4ms

pascallanger, 2024-02-07:
For "Power and Bind", what are you actually doing on the TX and RX side?
I'm asking since I don't see a difference between bind and no bind... Is there a bind button on the TX? My guess is that there is none and that if you put the RX in bind mode, it just binds to whatever TX is powered at that time.
Do you have multiple TXs?

pwkpete, 2024-02-07:
Power and bind is just turning on the TX and turning on the RX (it is already bound)

I was waiting to see if this worked and then going to do an actual bind - which I have to refresh my memory on how to do it with this setup. I know there is a bind button on the RX. Will look into that in a bit

pascallanger, 2024-02-07:
The TX and RX don't "talk" at least in normal mode, TX->RX only.
Please do a dump of the TX in bind mode. I doubt that the RX being off/on/bind matters but may be I'm wrong...

pwkpete, 2024-02-07:
I didn't see a bind mode in the TX: per the instructions:
[image]

In fact - I switched to a different model powered up and it connected to the RX without issue

pascallanger, 2024-02-07:
What do you mean by "different model"?
Can you select a model number on the TX? If yes I would need a dump of each model numbre.

pwkpete, 2024-02-07:
Yes, I selected a different model on the TX and it had no effect

It bound to the RX automatically - did not have to put the RX in bind mode

pascallanger, 2024-02-07:
Ok different model does not mean that there is "model match", ie the model and the RX have to match

pwkpete, 2024-02-07:
Correct, it would not seem that it is that 'smart'

pascallanger, 2024-02-07:
Also in the "Gyro" dump I see 2 things moving, like 2 channels. Do you have a mixer or were you moving 2 knobs in the same dump?

pwkpete, 2024-02-07:
In the above upload 'Scans 16MHz.zip' those were with the programmed model. EPA and likely some trim may have been not standard.

There is no 'mid' notch on the gyro channel, it was best guess - that is 100/100 EPA no trim
Aux channel 4 was just an on/off button

Would you like me to re-do any - I can do another gyro scan - I may have bumped the steering, it is close to the knob.

pascallanger, 2024-02-07:
It's not steering moving at the same time as gyro as I can differentiate the channels. It's something else that seems to follow the gyro value in some way. Could it be an unconnected 5th channel which follows the Gyro just because it's closed to it?

pwkpete, 2024-02-07:
That is weird, this is only a 4ch TX.

Mixing is off - I watch output from ch1-4 when adjusting gyro(3) and none move.

[trackstar_dspec_ts4g.pdf](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14200520/trackstar_dspec_ts4g.pdf)

This extra output does not happen on other channels? I can re-do the gyro scan if you would like

pwkpete, 2024-02-07:
[Gyro v2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14200693/Gyro.v2.zip)

Here you go, selected new model, did reset, validated no mix, ran test

Sorry about going too fast, was concerned about the size of the data files - not sure if that was warranted or not

pwkpete, 2024-02-08:
> Houston we have a problem. What I see moving in "Gyro v2" is the steering
> channel and nothing else...

I don't know what to say, I was watching the channel 3 graph in the screen
and making sure not to touch anything else. There wheels were definitely
not turning. Hmmm.

I don't think there is a way to remap channels like that. It was a fresh
model configuration.

Will look again though

pascallanger, 2024-02-08:
That's really strange...
Can you redo a capture of all channels on this model which you have just reset?
Plus:
 - On one channel do a capture with an EPA of -120..120
 - Set the failsafe of every channel to a different value: CH1=-50%, CH2=-25%, CH3=+25%, CH4=+50%

pwkpete, 2024-02-08:
Ok we try again!

Reset model

I set failsafe for each channel different, started capture and then turned off rx, hope that achieved what you were needing.

I re-ran all other tests again (EPA 120 for CH1,2)

Hope for better luck this time

[V3.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14213009/V3.zip)

pascallanger, 2024-02-08:
What values have you used for failsafe on each channel ?

pwkpete, 2024-02-08:
Failsafes are set by input position, can not enter a number manually)

ch1 R49
ch2 F31
ch3 -43
ch4 -100 (option -100 or +100)

pwkpete, 2024-02-10:
I think those values you have there make sense for fail safe, no?

Did my last upload look any different?

pascallanger, 2024-02-10:
Yes it's 200 - failsafe value.
I'll try to code the protocol when I have 5 minutes

pwkpete, 2024-02-10:
5 minutes, lol? You are a machine :)

pascallanger, 2024-02-16:
The code is available on this repo. You need to download, select the protocols you need and compile by yourself since there is not enough space in the flash anymore to do a build with all the protocols.
Documentation is here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-rc4g---3
I still need to implement ant ID and find unused RF channels but I would like to make sure it works first.

pwkpete, 2024-02-16:
Ok, thanks!

I will need to learn how to build this again - long ago I did this for an old Atmega module where I needed to enable/disable protocols due to memory but it's been a while.

Will notify when I get the build environment up and can test

pwkpete, 2024-02-17:
I just did a fresh install of the IDE, I don't think I ever had it on this PC and I can't get beyond this error - nothing compiles. Google offers no assistance either. Any ideas, never had this before? (I uninstalled and searched on the machine for any arduino artifact and deleted and then reinstalled - same result)
[image]

pascallanger, 2024-02-17:
It should be straightforward... Install the recommended version of the Arduino ide (1.8.xxx can't fully remember exactly), point to our board, install the board, select the right build and compile.

pwkpete, 2024-02-17:
I've built it in the past, I'm even following the directions this time :).
I downloaded the nightly version of the ide, pointed to the boards but
something is not right. I used to do embedded system development back in
the dark ages, so I'm not exactly a newbie but my experience with this tool
chain and cross compound with gcc is not up to date

pascallanger, 2024-02-17:
Pay attention to the version of the ide, you need to use an old one, the version is in the instructions.

pwkpete, 2024-02-18:
I guess I didn't realize the ide was that tired to the tool chain

pwkpete, 2024-02-18:
Got it. In my spare time, I'll try to understand why there is this affinity
to an older version. (I assume at some point it will lose updates or
support.)
Thx

pascallanger, 2024-02-18:
I know the build is broken for the latest versions. I haven't spent anytime to troubleshoot it so if you can give a hand that would be great 😁
You can have multiple versions of the ide running in parallel if needed.

pwkpete, 2024-02-18:
Ok got it compiled, I see the RC4G code, but not seeing the sub protocol available to select after flashing - also out of the box it compiled and fit, so I suspect I didn't config something correctly.

pascallanger, 2024-02-19:
The board should be set to "Multi X-in-1 STM32F103CB (128KB)". By default with the full package downloaded from here and the file _config.h by default, it should complain that you don't have enough flash space...
"No multi telemetry" means that either the module has not started at all or that serial mode is disabled.
As long as you'll see "No telemetry" you won't see the subprotocol RC4G.

pwkpete, 2024-02-19:
I'll have to mess with it when I get back later this week
I pulled a zip of the master branch, unzipped a new copy, I see #define
ENABLE_SERIAL, I also see your RC4G code, hrmph...
Made no changes, compiled and get:
Sketch uses 120468 bytes (99%) of program storage space. Maximum is 120808
bytes.

pascallanger, 2024-02-23:
Air and Surface builds are available here: https://downloads.multi-module.org/latest-test/
Please test and report.

pwkpete, 2024-02-24:
Got it compiled and it works perfectly!

Awesome work!

pascallanger, 2024-02-24:
Ok good news.
I still need to implement any ID and find unused RF channels but I wanted to make sure it worked first.
The current code uses your ID so your RX should just be controllable without the need to bind it.

pwkpete, 2024-02-24:
So is each tx flashed with a UID?

pascallanger, 2024-02-26:
Please test the latest radiolink/rc4g version which includes any ID and best RFs. It is available in the Surface build here: https://downloads.multi-module.org/latest-test/

pascallanger, 2024-02-26:
Both... If you load the Surface (SFC) firmware it will work on the TX16S, same if you download and compile.

pwkpete, 2024-02-26:
Ok, on the protocol dialog it says ID is unique, is there a way to set the ID? (Can I set it to match my real TX?)

pascallanger, 2024-02-26:
The ID in the new version is set to the one of your TX16S. You could change the ID (using the config lua script) so it match your original TX but this will affect all other protocols.
When the radio is saying "ID is unique", it's talking about the RX number value not being used elsewhere. You can have up to 64 matching RXs per protocol.

pascallanger, 2024-02-26:
Just install the new code and bind the RX to see if it works.

pwkpete, 2024-02-26:
Verified: change RX # lost connectivity, changing back reestablished.
Power cycled RX and TX - connects correctly.

All looks good

pascallanger, 2024-02-26:
Ok I call it done then

closed 2024-02-26 by pascallanger

pwkpete, 2024-02-27:
I'm trying to follow where the TX id comes from:
#define eeprom_read_byte EEPROM.read
#define EEPROM_ID_OFFSET		10

MProtocol_id_master=random_id(EEPROM_ID_OFFSET,false);

random_id(...) calls: eeprom_read_byte((EE_ADDR)(address+10)) = checks for 0xf0 and if so reads from address + 5 to 1 masking and shifting the addreess into id which gets returned

I assume EEPROM.read is coming from the board config and knows the starting addr of the eeprom for the multi module but I'm not following how that is unique/tx (yet)?

Am I reading the code correctly?

pascallanger, 2024-02-27:
id = STM32_UUID[0] ^ STM32_UUID[1] ^ STM32_UUID[2];