pascallanger/DIY-Multiprotocol-TX-Module #939: Option to set transmitter power or What happended to the LBT builds?

aheyer, 2024-02-05:
Coming from a Devo 10 with DeviationTX and internal 4-in-1 module I used to set the transmission power manually for every model. But my new RadioMaster Zorro with EdgeTX and internal 4-in-1 only gives me a single low power option. What about intermediate power levels above some µW and below 100mW? In Europe for example we're now forced to 10mW with protocols like DSM2. As a good citizen I set this PA level (or even lower) in Deviation to my DSM2 models instead of full blown 100mW.

How do I achieve this with EdgeTX and the internal 4-in-1? At the moment I know nothing about transmission power at all! Even the LBT firmware builds for modern protocols are missing since [the merge 3 months ago](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/pull/905). How do you expect us to adhere to local laws?

benlye, 2024-02-05:
There hasn't been a release since we added LBT builds to the automation, so there hasn't been anything to build.

aheyer, 2024-02-06:
Could you do an artificial increment to version 1.3.3.34 to get a new release including LBT? But even with LBT older protocols like DSM2 are unaffected and need power restrictions in some European countries.

Why won't you do that? Is it a hard – or even impossible – task because of the variety of multi-module implementations? You can set output power levels on chips like CC2500 or Cyrf6936. Can't be to hard to use user modifiable values when doing that? Even if you know nothing about the chip-external PA of the multi-module the user has the ability to trim down the power levels to his needs and chances are much better to adhere to local laws.

At least please clarify in the docs what logical power level is used per TX-chip or per protocol.

benlye, 2024-04-06:
LBT builds are available to download since 1.3..4.0 was released.

closed 2024-04-06 by benlye