pascallanger/DIY-Multiprotocol-TX-Module #941: Eazyrc new protocol request

vladovg, 2024-02-07:
Hy Pascal i have one of these 1/18 crawlers and decided to dump it.

XN297LBW  3 wire SPI

[image]
[image]
[image]
[image]

[Eazyrc.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14201923/Eazyrc.zip)

Hope you can make something with it.

pascallanger, 2024-02-19:
Do you have any buttons/features for example lights?

pascallanger, 2024-02-19:
Do you have to put the car in bind mode?

pascallanger, 2024-02-23:
I'm starting the implementation. There are quite some unknowns so let's see how it will work out.
I do not plan to store the car RX ID. The radio will stay on bind waiting for the car to be powered on. If you power cycle the car then you'll need to launch a new bind on the radio (which I don't think you need to do right now).

pascallanger, 2024-02-23:
I've added the protocol under SFC (surface). As I said there are a lot of unknowns and I can't really test it... It's currently using your original TX ID and hopping frequencies, we will have to do further tests.
Please test and report, test builds are located here: https://downloads.multi-module.org/latest-test/
Documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#eazyrc---61

vladovg, 2024-02-23:
Unfortunately it's not binding.

pascallanger, 2024-02-23:
Does the bind stop?

pascallanger, 2024-02-23:
Grrr I have an idea, hold on a few minutes

pascallanger, 2024-02-23:
Test again

vladovg, 2024-02-23:
It's working:) great job tank you.  
Buy the way new file names are too long for edgetx MT12.

pascallanger, 2024-02-23:
Name too long: that's a known issue....
Ok we need to do more tests to enable multiple IDs frequency dump.
Do you know how to compile by yourself?

vladovg, 2024-02-23:
You want me to change TX ID and see if it works?

pascallanger, 2024-02-23:
Try to comment (add // before):
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/af47462ba79eb86e19d8bc6c4685717bd81bd7bf/Multiprotocol/EazyRC_nrf24l01.ino#L19

vladovg, 2024-02-23:
It works thank you again :)

pascallanger, 2024-02-23:
Great, before we can it done and with the modification above, can you try on the radio a couple of RX numbers? That will create different IDs and hopping frequencies.

vladovg, 2024-02-24:
IDs are working:)

pascallanger, 2024-02-24:
I've pushed a final version. Please test a few rx numbers as shown above. Then we call it done.

vladovg, 2024-02-24:
Everything works as described 👌 great job thank you.

closed 2024-02-24 by vladovg